Ultimaker/Cura #6039: alfawise u 20 one

petitfufu, 2019-07-22:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

petitfufu, 2019-07-22:
hello, the printer alfawise u20 one is not reppertorié, so I put the u20 in the place, but on cura, the prints are slipped compared to the plateau, thank you

Ghostkeeper, 2019-07-22:
We don't have an Alfawise U20 to test this with. We can't fix this for you because we don't know what the correct position is of the build plate. Could you help us test this?

Please go into the list of printers and click "Manage printers", then click on the "Machine Settings" button, and try clicking the "Origin at Center" checkmark. Then try printing. Is the print correctly placed then?

Or if you adjust the print volume, is it placed correctly then?

borntofly78, 2020-05-14:
Same problem here... it ignores half of The X axis!!! It starts printing completly out of place! The right corner of the print is in the midle of the printer! Created a profile for the printer using the CR10 as a base printer and it works fine.

borntofly78, 2020-05-14:
But my problem started on the Cura 4.6.0

mahtDFR, 2020-05-14:
The comment above still stands, we don't have the resources to test every third party printer unfortunately. If someone in the community has a working and thoroughly tested profile, they can make a pull request that we'd be happy to review.

borntofly78, 2020-05-14:
The Cr10 profile is working flawlessly here

Ghostkeeper, 2020-05-18:
> Please go into the list of printers and click "Manage printers", then click on the "Machine Settings" button, and try clicking the "Origin at Center" checkmark. Then try printing. Is the print correctly placed then?
> 
> Or if you adjust the print volume, is it placed correctly then?

borntofly78, 2020-05-20:
I had already tried that... But don't worry... I used the Cr10 profile and
it's working great.

Thanks a lot and best regards from Brazil,
Marcello
www.youtube.com/brmakers

mahtDFR, 2020-05-20:
Great! Closing

closed 2020-05-20 by mahtDFR