Ultimaker/Cura #6040: Customizable preview color schemes

DOCaCola, 2019-07-22:
Feature request: Allow customization of preview color schemes.

The current color scheme is not ideal in some scenarios. I will quote a [forum post](https://community.ultimaker.com/topic/27233-change-color-of-preview-color-scheme/) that illustrates the issue:

Is it possible in CURA 4.0 to modify the color in the mode "Preview" => "Line type" => "Color Scheme"?

Currently the Top/Bottom-Layer is represented in a kind of light yellow. The infill is represented in let me say a kind of less light yellow color. As we are all aware about the fact that millions of colors are available I do not understand why this extremly similar colors have been choosen for representation by CURA.

E.g. why not follow the PRUSA Slicer approach?

Cura Approach:
[image]

PRUSA Slicer Approach:﻿﻿

[image]

------------

Allowing user customizable color schemes would mitigate this issue.

nallath, 2019-07-22:
The main reason for it is simply "legacy". It's pretty much how the first versions of Cura did it, so it made sense to keep them. 
That being said you can change these values in the theme. It's not super user friendly to change, but at least there is a way to do it.

Ghostkeeper, 2019-07-22:
That being said, we've changed the default two themes such that infill is now a darker orange for Cura 4.2. That should address the actual issue. Theming should take care of the rest. I don't see the need for the user being able to customise that himself.

closed 2019-07-23 by nallath