Ultimaker/Cura #6041: Cancelled Print; HB continues to Target Temp, Auto Bed Levelling delayed until then

Twopusses, 2019-07-22:
**Application version**
 4.1.0

**Platform**
 WINDOWS 10

**Printer**
 CTC i3

**Reproduction steps**
I cancelled a print as I'd forgotten to Auto Home the bed first, but, having cancelled, CURA doesn't cool down, instead it continues heating until the BED ONLY is at Target Temp and doesn't allow HOMING until I reach HB 60Deg. 

**Actual results**
After Bed Temp Target according to eith Default Temp or GCode setting, the printer can be HOMED.

**Expected results**
 I think both Extruder AND HB Heating should switch off if you Cancel a Print, rather than continue heating the bed.

**Additional information**

mjf55, 2019-07-23:
I believe this to be the firmware blocking new commands until the temperature is reached, IF the gcode that is used is the WAIT version ( i.e. M109 and M190 ).  see this Marlin link for details. http://marlinfw.org/docs/gcode/M190.html

Ghostkeeper, 2019-07-23:
Is this when you are printing via USB cable? We do send commands to turn the heated bed, current nozzle and the fans off:

https://github.com/Ultimaker/Cura/blob/ca48f9171f12081978b6f08e2b8d010635a0f758/plugins/USBPrinting/USBPrinterOutputDevice.py#L356-L358

I think mjf55 is correct. The printer doesn't interrupt the previous command when you cancel. It just continues executing it. There is no way in which Cura can cause it to interrupt the heating command and stop earlier.

closed 2019-07-23 by Ghostkeeper