Ultimaker/Cura #6047: Changed .stl do not reloaded when there are multiples copies

EstudioRoble, 2019-07-23:
**Application version**
4.1
**Platform**
Linux Mint 
Linux I 4.4.0-145-generic #171-Ubuntu SMP Tue Mar 26 12:43:40 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
**Printer**
All
**Reproduction steps**
Load .stl
Multiply model.
Edit .stl (in external software) and save it.
Cura asks nicely to reload. Press Reload.
**Actual results**
Only one of the multiple models is reloaded. The others remain with the old .stl
**Expected results**
Every model is updated.

Ghostkeeper, 2019-07-23:
Yup, this reproduces for me as well. It's not even the original that gets reloaded, just one of the copies.

Ghostkeeper, 2019-07-23:
As a workaround, pressing F5 works fine.

Ghostkeeper, 2019-07-26:
Fixed here: https://github.com/Ultimaker/Uranium/commit/7339c07eca8f7a1fce3e519d10abb536e329fd31 for 4.2.

closed 2019-07-26 by Ghostkeeper