Ultimaker/Cura #6051: cura 4.2 beta; scale down will leave the model floating above the bed

UltiArjan, 2019-07-23:
**Application version**
4.2 Beta

**Platform**
Windows 10

**Printer**
Ultimaker2, with a few modifications for use with Duet3D
I also replicated the behaviour  with the S5 selected, same behavior.

**Reproduction steps**
When you scal a model UP ist alligned with the buildplate as it should, but when scaling a model DOWN (from oversize) it keeps floating above the bed. 
The button "automatically drop to bed"is on

*also tried with a different model, on the S5, it also keeps floating above the bed.

[image]

**Actual results**
model does no drop to the bed, and as a result cura says it can't slice the model, even though it's small enough

**Expected results**
drop on bed

[float-UM2_EMU_Pint_SetUP_improved.3mf.zip](https://github.com/Ultimaker/Cura/files/3423726/float-UM2_EMU_Pint_SetUP_improved.3mf.zip)
[image]

Ghostkeeper, 2019-07-24:
Yeah I could reproduce this. Recently I've had it happen a few times that I load a big model. When scaling the model down so that it fits in the build volume, it doesn't directly drop to the build plate. Only after nudging it slightly again will it finally drop.

Vandrasc, 2019-07-29:
Devs Cura-6686

Vandrasc, 2019-08-07:
Fixed for the next Cura version. Thank you for your feedback!

closed 2019-08-07 by Vandrasc