Ultimaker/Cura #6052: Cura crashes right after startup

jeroenrnl, 2019-07-24:
**Application version**
4.1

**Platform**
Linux 5.2.1

**Printer**
JGAurora A5

**Reproduction steps**
Startup cura

**Actual results**
Cura starts, the splash screen shows, going through all the normal steps, then the application screen shows and after ~2 seconds, it closes

**Expected results**
It doesn't close

**Additional information**
Running cura from a terminal:

```
$ cura
/usr/lib/python3.7/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.7/site-packages/UM/Settings/SettingDefinition.py:46: DeprecationWarning: invalid escape sequence \.
  regex_pattern = '(?<!\.|\w|\d)0+(\d+)'
/usr/lib/python3.7/site-packages/cura/Settings/CuraContainerRegistry.py:76: DeprecationWarning: invalid escape sequence \s
  num_check = re.compile("(.*?)\s*#\d+$").match(new_name)
[MainThread] UM.Resources.__initializeStoragePaths [416]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [427]: Config storage path is /home/jeroen/.config/cura/4.1
[MainThread] UM.Resources.__initializeStoragePaths [435]: Data storage path is /home/jeroen/.local/share/cura/4.1
[MainThread] UM.Resources.__initializeStoragePaths [447]: Cache storage path is /home/jeroen/.cache/cura/4.1
[MainThread] UM.PackageManager.__init__ [52]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager.__init__ [52]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [105]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [115]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [162]: OpenGL renderer type for this OpenGL version: AMD Radeon (TM) R9 380 Series (TONGA, DRM 3.32.0, 5.2.1-arch1-1-ARCH, LLVM 8.0.1)
[MainThread] UM.Qt.QtApplication.initialize [141]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [152]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [122]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [122]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [139]: Loaded user packages management file from /home/jeroen/.local/share/cura/4.1/packages.json
[MainThread] UM.PackageManager._saveManagementData [207]: Package management file /home/jeroen/.local/share/cura/4.1/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [90]: Loading plugin configuration file '/home/jeroen/.config/cura/4.1/plugins.json'
2019-07-24 12:00:56,521 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ConsoleLogger
2019-07-24 12:00:56,561 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: /usr/bin/CuraEngine
2019-07-24 12:00:56,570 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraEngineBackend
2019-07-24 12:00:56,595 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-07-24 12:00:56,620 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-07-24 12:00:56,641 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-07-24 12:00:56,652 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-07-24 12:00:56,664 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-07-24 12:00:56,681 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-07-24 12:00:56,681 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ConsoleLogger was already loaded
2019-07-24 12:00:56,696 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-07-24 12:00:56,708 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLWriter
2019-07-24 12:00:56,715 - WARNING - [MainThread] STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2019-07-24 12:00:56,725 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLReader
2019-07-24 12:00:56,736 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJReader
2019-07-24 12:00:56,746 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJWriter
2019-07-24 12:00:56,761 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-07-24 12:00:56,777 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-07-24 12:00:56,783 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
[... 508 lines cut ...]
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f039e7fc700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x00007f039effd700 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/threading.py", line 427 in acquire
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.7/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Current thread 0x00007f03ebc17680 (most recent call first):
  File "/usr/lib/python3.7/site-packages/UM/Qt/QtApplication.py", line 341 in exec_
  File "/usr/lib/python3.7/site-packages/cura/CuraApplication.py", line 797 in run
  File "/bin/cura", line 136 in <module>
Segmentation fault (core dumped)
```

smartavionics, 2019-07-24:
Hi. I assume you are running from source. Please update your Uranium to the master branch as we have just fixed a crash there. Thanks.

Ghostkeeper, 2019-07-24:
I think you're also using the wrong version of Qt. If you're developing, it's wise to use the same version as what will be in the final build: 5.10. This way you won't accidentally use a feature that wouldn't run with the mainline builds. It also prevents crashes like this that were introduced in more recent versions of Qt, though we do have a workaround for one of those in 5.13 now in Uranium.

jeroenrnl, 2019-07-24:
Hi, I'm running the Arch package version and the Qt version is just the default Arch package too, I'm not developing.

smartavionics, 2019-07-24:
There is a known regression with Qt 5.13 which is what is now being used on Arch. The Uranium fix works around that issue.

jeroenrnl, 2019-07-24:
What is Uranium?

smartavionics, 2019-07-24:
It's provides the application framework that cura is based on.

jeroenrnl, 2019-07-24:
ok, this is what I did:
```
git clone https://github.com/Ultimaker/Uranium
cd Uranium
mkdir build
cd build
cmake ..
make
sudo make install
```
... but this doesn't fix it.
I checked, Uranium was installed in the same directories as where `make install` put them

smartavionics, 2019-07-24:
Then it must be something else, perhaps there's something in the recently closed issue that can help you. Take a look at https://github.com/Ultimaker/Cura/issues/6008.

jeroenrnl, 2019-07-24:
Ok, turned out I hadn't checked all files. The Python site-packages files were installed in `/usr/local/lib` instead of `/usr/lib` where my system keeps them. I moved them there and now it no longer crashes, but I did lose the preview of the print. I tried changing "Force layer view compatibilty modus", but that didn't help.

[image]

krop, 2019-08-13:
I can reproduce this crash since months.
I just updated my git snapshot packages (cura, cura-engine, uranium, arcus...)

With current master branches, I get:

```
# cura
/usr/lib/python3.7/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.7/site-packages/UM/VersionUpgradeManager.py:88: DeprecationWarning: invalid escape sequence \.
  ".*\.lock",       # Don't upgrade the configuration file lock. It's not persistent.
/usr/lib/python3.7/site-packages/cura/UI/ObjectsModel.py:65: DeprecationWarning: invalid escape sequence \(
  self._naming_regex = re.compile("^(.+)\(([0-9]+)\)$")
1:10:58 - cura(13921) - org.kde.kcrash: KCrash::initialize: KCrash disabled through environment.
1:10:58 - cura(13921) - org.kde.kcrash: KCrash::initialize: KCrash disabled through environment.
[MainThread] UM.Resources.__initializeStoragePaths [415]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [426]: Config storage path is /home/krop/.config/cura/master
[MainThread] UM.Resources.__initializeStoragePaths [434]: Data storage path is /home/krop/.local/share/cura/master
[MainThread] UM.Resources.__initializeStoragePaths [446]: Cache storage path is /home/krop/.cache/cura/master
[MainThread] UM.PackageManager.__init__ [52]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [52]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [105]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [115]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [162]: OpenGL renderer type for this OpenGL version: GeForce GTX 660/PCIe/SSE2
[MainThread] UM.Qt.QtApplication.initialize [141]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [152]: Initializing version upgrade manager ...
[MainThread] UM.PackageManager._loadManagementData [134]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [134]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [151]: Loaded user packages management file from /home/krop/.local/share/cura/master/packages.json
[MainThread] UM.PackageManager._saveManagementData [219]: Package management file /home/krop/.local/share/cura/master/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [90]: Loading plugin configuration file '/home/krop/.config/cura/master/plugins.json'
2019-08-14 01:10:58,363 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ConsoleLogger
2019-08-14 01:10:58,391 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: /usr/bin/CuraEngine
2019-08-14 01:10:58,391 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraEngineBackend
2019-08-14 01:10:58,397 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ConsoleLogger was already loaded
2019-08-14 01:10:58,400 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJReader
2019-08-14 01:10:58,406 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJWriter
2019-08-14 01:10:58,410 - WARNING - [MainThread] STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2019-08-14 01:10:58,412 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLReader
2019-08-14 01:10:58,415 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLWriter
2019-08-14 01:10:58,417 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-08-14 01:10:58,421 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-08-14 01:10:58,426 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-08-14 01:10:58,430 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-08-14 01:10:58,435 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-08-14 01:10:58,441 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-08-14 01:10:58,448 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-08-14 01:10:58,451 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-08-14 01:10:58,457 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-08-14 01:10:58,461 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-08-14 01:10:58,466 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimpleView
2019-08-14 01:10:58,467 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [39]: Checking for new version of cura
2019-08-14 01:10:58,489 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin 3MFReader
[... cut ...]
  File "/usr/lib64/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib64/python3.7/threading.py", line 885 in _bootstrap

Current thread 0x00007fa080e1c680 (most recent call first):
  File "/usr/lib/python3.7/site-packages/UM/Qt/QtApplication.py", line 265 in initializeEngine
  File "/usr/lib/python3.7/site-packages/cura/CuraApplication.py", line 859 in runWithGUI
  File "/usr/lib/python3.7/site-packages/cura/CuraApplication.py", line 781 in run
  File "/usr/bin/cura", line 137 in <module>
[1]    13921 abort (core dumped)  cura
```

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

jeroenrnl, 2024-11-21:
I didn't even know this bug was still open. I'm no longer using Cura, but have used it successfully many times after the last comment. I think I switched to a different Arch package to fix it.

closed 2024-11-21 by jeroenrnl