Ultimaker/Cura #6053: Cura 4.1 monitor function blank screen and isn't printing.

Loadie150, 2019-07-24:
**Application version**
4.1

**Platform**
Windows 10 3.4 Mhz quad processor 12Meg RAM

**Printer**
CORE A8 XY Slim

**Reproduction steps**
Periodically the monitor function in Cura has a blank screen after selecting print via USB. Once that happens I have no other option but to close down and start over again.

**Actual results**
Works fine until it happens again. I have no idea why this happens and it happens at different times and with different prints.

Ghostkeeper, 2019-07-25:
Can you provide a log file for us? If the monitor stage is empty I'd expect there to be some QML warnings in there.

Loadie150, 2019-07-25:
How do I get the log file? Where is it?

Vandrasc, 2019-07-26:
From Cura > Help>Show Configuration log>Cura.log or from User>AppData>Roaming>Cura, hope this helps.

Loadie150, 2019-07-26:
Through the Cura app. I have Help>Show Configuration folder>cura.log, got it.  
[cura.log](https://github.com/Ultimaker/Cura/files/3436831/cura.log)

I see quite a few warnings on that log file but don't have a clue on what they mean! I hope this upload works!!

Ghostkeeper, 2019-07-29:
I see those warnings, yeah, like these:

```
2019-07-24 16:11:04,500 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:18: TypeError: Cannot read property of null
2019-07-24 16:11:04,503 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:20: TypeError: Cannot read property of null
2019-07-24 16:11:04,507 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:21: TypeError: Cannot read property of null
2019-07-24 16:11:04,512 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:42: TypeError: Cannot read property of null
2019-07-24 16:11:04,516 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:43: TypeError: Cannot read property of null
2019-07-24 16:11:04,519 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/plugins/USBPrinting/MonitorItem.qml:44: TypeError: Cannot read property of null
2019-07-24 16:11:04,523 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrintSetupTooltip.qml:67: TypeError: Cannot read property of null
2019-07-24 16:11:04,527 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrintSetupTooltip.qml:69: TypeError: Cannot read property of null
2019-07-24 16:11:04,531 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrintSetupTooltip.qml:71: TypeError: Cannot read property of null
2019-07-24 16:11:04,534 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:36: TypeError: Cannot read property of null
2019-07-24 16:11:04,538 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:37: TypeError: Cannot read property of null
2019-07-24 16:11:04,541 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:49: TypeError: Cannot read property of null
2019-07-24 16:11:04,544 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:59: TypeError: Cannot read property of null
2019-07-24 16:11:04,548 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:61: TypeError: Cannot read property of null
2019-07-24 16:11:04,551 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/OutputDeviceHeader.qml:63: TypeError: Cannot read property of null
2019-07-24 16:11:04,554 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/ExtruderBox.qml:17: TypeError: Cannot read property of null
2019-07-24 16:11:04,558 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:29: TypeError: Cannot read property of null
2019-07-24 16:11:04,561 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:30: TypeError: Cannot read property of null
2019-07-24 16:11:04,566 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:40: TypeError: Cannot read property of null
2019-07-24 16:11:04,570 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:73: TypeError: Cannot read property of null
2019-07-24 16:11:04,573 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:135: TypeError: Cannot read property of null
2019-07-24 16:11:04,578 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:246: TypeError: Cannot read property of null
2019-07-24 16:11:04,582 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:247: TypeError: Cannot read property of null
2019-07-24 16:11:04,586 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/PrinterOutput/HeatedBedBox.qml:173: TypeError: Cannot read property of null
[... cut ...]
2019-07-24 16:11:04,693 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.1/resources/qml/MonitorButton.qml:241: TypeError: Cannot read property of null
```

When tracing them though, I find that they all point to something like this:

https://github.com/Ultimaker/Cura/blob/63316fa4e9049043e36a35ff91bc2a9cd5aee465/plugins/USBPrinting/MonitorItem.qml#L18

So it's indicating that it's trying to read a property from `null`. The only property being read on that line is `parent`, so that means that `parent` is `null`? Very strange.

I haven't been able to reproduce the problem yet, so far.

Loadie150, 2019-07-29:
I have absolutely no idea at all! I just have to restart the print, which sucks in some ways!

mantoles, 2019-10-18:
exact same error here on a 25 hour print stopped at 20 hours.  Sucks...

019-10-17 21:09:38,367 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.3/resources/qml/PrinterOutput/MonitorItem.qml:39: TypeError: Cannot read property of null

closed 2020-06-19 by no-response[bot]