Ultimaker/Cura #6055: Screen issues

bruster555, 2019-07-24:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

bruster555, 2019-07-24:
In the monitor section, the extruder and build plate dialogue boxes are covered. In the prepare section, the X arrow (RED) for movement when selected to move, disappears, but only on part of the build plate.

Ghostkeeper, 2019-07-25:
I'm not sure which dialogue boxes you mean. Is this for USB printing, connected via Wifi to an Ultimaker printer, or maybe OctoPrint?

And I've never seen the arrow disappear. It's going behind the selection outline though, but that doesn't seem like what you meant.

closed 2020-06-19 by no-response[bot]