Ultimaker/Cura #6056: Cura 4.1.0 infill is not printing on Ender 3

saxonpaul92, 2019-07-25:
**Application version**
Cura 3.4.1

**Platform**
Windows 10

**Printer**
Creality Ender-3
[ProbeHolder.3mf.zip](https://github.com/Ultimaker/Cura/files/3429920/ProbeHolder.3mf.zip)

**Reproduction steps**
"Normal" print settings --> Infill = "Tri-Xexagon" --> Infill Density = "20"

**Actual results**
If you go to preview you can see that the slicer does not include the infill.
Printer prints the Bottom Layer and sides but, since it does not print the infill, it fails at the top layer.

**Expected results**
Infill should print and be shown in "Preview"

**Additional information**
Reverted back to Cura 3.4.0 with the same settings and preview shows the infill and the print was successful

saxonpaul92, 2019-07-25:
Infill problem

Ghostkeeper, 2019-07-25:
Possibly the infill was disabled from the preview in the legend.

saxonpaul92, 2019-07-25:
No, it is selected however, nothing shows in preview nor in the final print.

saxonpaul92, 2019-07-26:
Update:  The only way to print the infill is to set the "Infill Density" to 100%.  Otherwise, no infill.

Vandrasc, 2019-07-26:
Hi @saxonpaul92 , thank you for sharing the project file with us, that makes it easier for us to see what the problem could be. 

I see that you have Spiralize outer contour enabled, this will transform your print from a solid model into a one wall print. If you disable that setting you will then notice that the infill is being generated in your model.  If that is the case then this is the intended behavior and works as expected, hope this helps.

Ghostkeeper, 2019-07-26:
This is working as expected when spiralize is enabled.

closed 2019-07-26 by Ghostkeeper