Ultimaker/Cura #6058: I can't remove screensaver

TriKriSta, 2019-07-25:
I installed the program (v 4.0.0)

The program showed me screensaver
`Welcome to Ultimaker Cura
Please follow these steps to set up Ultimaker Cura. This will only take a few moments`

But it doesn't disappear and I cannot remove it

Vandrasc, 2019-07-25:
Hi @TriKriSta , do you mean the installation window?  Could you please share a screenshot with us, so we can get a better understanding of what your issue is? I guess you are using Windows? And are you using the 4.0 version or Cura 4.1?

TriKriSta, 2019-07-25:
I tryed to install both versions 4.0 and 4.1.
I'm using Arch Linux.
[image]

fieldOfView, 2019-07-25:
Try the AppImage. That is currently the only supported version on Linux. There are some issues between Cura and Qt 5.13 which is installed on your system. The AppImage doesn’t have those issues.

elbowz, 2020-02-17:
I found here: https://bugs.archlinux.org/task/62133?project=5&string=cura
a possible workaround to start cura without issue:

`env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP cura`

mahtDFR, 2020-02-18:
Closing this for reasons above

closed 2020-02-18 by mahtDFR

JohannCahier, 2020-08-01:
Hey DAWGS..

JohannCahier, 2020-08-01:
That's a fucking showstopper....

fieldOfView, 2020-08-01:
Yo DAWG, try the AppImage. That is currently the only supported version on Linux. There are some issues between Cura and Qt 5.13 which is installed on your system. The AppImage doesn’t have those issues.

JohannCahier, 2020-08-04:
Thanks for the answer. I was meaning : using AppImages IS the showstopper to me. More for political / system maintenance policy reason than technical ones.

I was I bit tired I think, I didn't saw @elbowz answer, that works for me (at least I have the "Get Started" button back, I can browse a few screens... Seems way better).

I keep in mind I may meet instabilities... 

BTW : I'll rather try to built it from sources, before going with an AppImage ! 
I'm surprized there is no recipe for that in the AUR already...

nallath, 2020-08-04:
Dependency issues were the main reason we moved away from .deb images. Conceptually it's a great idea to share dependencies, but in practice, it just doesn't work in all cases (especially if you have a hard dependency on a specific version of a lib).

So it's a bit more data that you're pumping around if you use an appimage, but at least you control what exact configuration is being used (which also decreases the difficulty in finding issues caused by dependency version interplay). 

I'm not aware of any political reasons not to do something with AppImages. Did I miss something?

JohannCahier, 2020-08-04:
Hey @nallath.

Indeed "political" was a bit overkill, take it as irony...
The "system maintenance policy" stated is my previous reply (which lacks a bit of editing), is more accurate.

If you want a "political" aspect of this : I like "green computing", and clearly loading multiple versions of the same lib is a given machine RAM is not "green". 

I don't have any problem with the distribution solution you choose. It's open source, so I can recompile it. 

I'am a bit puzzled I had to land to land here, after the only PKGBUILD if found on the AUR downloaded a binary package...

Maybe it's a call to contribute to the AUR with a "clean" cura-git package.

And maybe i'll change my mind for some reason in the next few days : i'm very new to 3D printing, so i'll learn a lot the next days/weeks.

Don't worry for me,

Many greets to your community and see you soon :)

JohannCahier, 2020-08-04:
@nallath : a .deb distro package wouldn't have solved my problem anyway :)

Thanks for the comment !

Ghostkeeper, 2020-08-05:
We'd welcome fixes necessary to make it work with Qt 5.15, as long as they don't break compatibility with the Qt 5.10 that we ship with the Windows builds, MacOS builds and the AppImage.

KiralyCraft, 2020-12-19:
elbowz's workaround is just fine for me. Running Manjaro KDE with Qt 5.15. Many thanks!

electronoob, 2021-01-05:
worked for me also, edited the shortcut for cura, nice nice. thank you.

983, 2021-05-12:
This issue also occurs on Kubuntu 20.04. The blue `Get started` button simply does not show up. I think this issue should be reopened until `sudo apt install cura` actually installs a working version of cura.

Ghostkeeper, 2021-05-20:
We did fix this issue. Since Cura 4.9 we are using Qt 5.15, so Cura is expected to work with Qt 5.15 now.

Only [the package maintainer didn't update it yet](https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=cura&searchon=names). Again, we don't maintain those packages. It's not under our control. We just provide the AppImage.

983, 2021-05-20:
Great, then we know where to continue. Thanks!