Ultimaker/Cura #6059: [4.1.0] Weird, inefficient and useless travel moves

theWaldschrat, 2019-07-25:
**Application version**
4.1.0

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Reproduction steps**
Open attached project file:
[Glashalter.zip](https://github.com/Ultimaker/Cura/files/3432140/Glashalter.zip)
Slice, then preview, move down to first layer and replay the last about 20 %.

**Actual results**
Watch the print head bounce back and forth instead of a smooth print movement.
Also note the unretracted travel which leaves a nice, solid string on the print bed:
[image]

**Expected results**
Smooth, efficient printing without bouncing back and forth and no unretracted travels outside of the model.

**Additional information**
Not much more to say.

smartavionics, 2019-07-25:
Hello @theWaldschrat , I have sliced your project using my cura build and it doesn't have the problems you mention. That could be because my cura build has fixes for various defficiences in the combing and other features. If you want to try it out, you can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. They can be installed alongside the Ultimaker buids.

[image]

Ghostkeeper, 2019-07-26:
This is fixed in Cura 4.2 along with the fixes we made for gap filling. Please try the beta!

closed 2019-07-26 by Ghostkeeper