Ultimaker/Cura #6060: Dual extruder problem

dongl1, 2019-07-26:
**Application version**
3.6 - 4.2

**Platform**
Win 10 Rearm ,ramps 1.4 Marlin 2.0

**Printer**
Custom printer

**Reproduction steps**
When using auto bed leveling when G29 is added to the start Gcode

**Actual results**
when selecting extruder 2 and assigning it to an object and starting print the extruder heats as it should but it doesn't extrude instead Extruder  1 tries to extrude  even if it is disabled if G29 is removed from the start code it funtions correctly.

**Expected results**
Extruder 2 should operate as normal
[cura.log](https://github.com/Ultimaker/Cura/files/3433724/cura.log)
[cura.zip](https://github.com/Ultimaker/Cura/files/3433725/cura.zip)

**Additional information**

Ghostkeeper, 2019-07-26:
I'm not understanding what your problem is exactly. You're basically saying that you've sliced something for extruder 2, and the g-code is all fine but your custom-built printer is using the wrong extruder to print it? Maybe you need to ask this over at the Marlin repository?

Or perhaps you think the g-code that Cura generates is wrong? What does that all have to do with the G29 in your start g-code?

GregValiant, 2023-09-15:
Can this be closed?

closed 2023-09-29 by github-actions[bot]