Ultimaker/Cura #6062: Cura slicing issue, but same settings work perfect in simplify

mateusbatatais, 2019-07-26:
**Application version**
4.1.0

**Platform**
Windows 10

**Printer**
Custom Hypercube Hevolution

**Reproduction steps**
I have a problem setting up my Cura... Check out my cat print (in gray) made in simplify 3d. Then look at my black print made in Cura(ignore interrupted printing). I used the same settings a friend uses on his printer, just like mine (a Hypercube Hevo we assemble with the same parts and size).

What I want to make clear is that my printer works normally with Simplify, but I would like to use Cura. The problem is that when I print with it, it looks like my printer is defective.

Any suggestion? Everything indicates an overextrusion, but my extruder calibration is fine and I am using a 1.75mm filament, set correctly ...

My cure is 4.1.0, ABS printing, 60mm / s, nozzle 225, table 110.
All identical to my friend's who perfectly with an identical printer.

**Actual results**
Printer looks like overextrusion

**Expected results**
Correct print

**Additional information**
Printer works fine with another slicer, like Simplify 3d
[image]
[image]
[image]
[image]
[image]
[image]

smartavionics, 2019-07-26:
Those photos don't tell us much (and the're far too big for my feeble broadband to cope with, please don't upload any more that size!). What would actually be useful would be if you could save the cura project file (File -> Save) and then zip the .3mf file or rename to .zip and attach that. Thanks.

mateusbatatais, 2019-07-26:
Ok. Sorry for the images. Attached the project zipped. Look at Hevo Normal.
[Untitled.zip](https://github.com/Ultimaker/Cura/files/3437297/Untitled.zip)

smartavionics, 2019-07-26:
Thanks for the file. Unfortunately, it doesn't contain the model. Could you also attach that or another .3mf file with the model.

Sorry for whining about the images but my broadband is currently running at 20 KB/S download speed (that's not a typo). Engineer hopefully will turn up tomorrow and fix it!

mateusbatatais, 2019-07-26:
with the model. Thank you!

[CFFFP_calicat.zip](https://github.com/Ultimaker/Cura/files/3437377/CFFFP_calicat.zip)

mateusbatatais, 2019-07-26:
update. I have another printer, prusa i3x. I just tried the same settings there and it worked. A good print. The strange thing is that I'm making another impression here in my Hevo, sliced by simplify and it's happening normally.

smartavionics, 2019-07-26:
I looked at your project and can't see anything at all obviously wrong. I would recommend enabling the optimize wall printing order but I doubt if that's what the problem is.

Those photos reminded me of a problem I had a while back where seemingly random layers over or under extruded. It turns out it was a faulty thermistor in the hotend.

Ghostkeeper, 2019-07-29:
Two things I'd suggest to change here:
* Outer before inner walls. This normally gives greater dimensional accuracy.
* Enable Jerk Control and turn the jerk down a bit. We could be flexing the belts a bit.

mateusbatatais, 2019-07-29:
Thank you for your help. After a few more tests and attempts, I noticed a small flaw that did not happen in simplify but in Cura ... The extruder is slightly weak and when this problem started to become more evident, I could notice even in other slicers. I squeezed the nozzle and adjusted small items. Problem solved. Thank you again.

closed 2019-07-29 by mateusbatatais