Ultimaker/Cura #6063: Cura does not display "Prepare" button after changing settings on model

squee30000, 2019-07-26:
**Application version**
Cura 3.6.0

**Platform**
Windows 10

**Printer**
Monoprice Ultimate as a custom printer

**Reproduction steps**
<ol>
<li>Opened Cura by opening an STL file as an associated file.</li>
<li>Sliced object.</li>
<li>Changed support pattern.</li>
</ol>

**Actual results**
Button displayed remained "Save to file" and object remained sliced in layer mode

**Expected results**
Button displayed should have changed to "Prepare" and layers should have disappeared

**Additional information**
Bug does not occur when using files opened from within Cura. 
Resetting object position displays "Prepare" button as expected.

Liger0, 2019-07-26:
3.6? lol

Ghostkeeper, 2019-07-29:
You laugh, but this problem existed up until 4.1. We fixed it in version 4.2. Please try the beta (or the stable version that gets released very soon).

closed 2019-07-29 by Ghostkeeper