Ultimaker/Cura #6066: [4.2 beta] Slicing stalls, and PC usage is low

Liger0, 2019-07-26:
**Application version**
4.1 beta

**Platform**
Win 10 x64. Ryzen 5 1600. RX vega 56.

**Printer**
Cr10s

**Reproduction steps**
Slice a complex model

**Actual results**
It just stalls. No advancement after 15+ minutes. GPU is unused, cpu total usage is 10-15% (does cura support multi threading?).

**Expected results**
It should use the available hardware to slice fast

**Additional information**
[image]

Ghostkeeper, 2019-07-29:
Yeah, this reproduces. During the tree support processing it seems to use only one thread.

Block137, 2019-07-29:
The no advancement after 15+ minutes happened to me when the model file size *in total* is above 400 MB. Cura seems to not start slicing at all, progress bar at 0% "Slicing", 0% CPU usage.

At this size of model file, Tree Support takes multiple hours to generate (never waited to see the result).

Ghostkeeper, 2019-07-30:
For that you might want to check your log. If CuraEngine crashed (for instance due to running out of memory) you'd see it in there.

MariMakes, 2022-08-25:
When using the tree supports generation, apparently only one thread is used, and therefore, the slicing time increases a lot. We have an item on our backlog to adress it. For internal reference CURA-6692

closed 2024-12-07 by github-actions[bot]