Ultimaker/Cura #6068: Granular Top/Bottom Pattern setting

Firedrops, 2019-07-27:
**Is your feature request related to a problem? Please describe.**
Current concentric top/bottom pattern settings often make infill>shell layers print over air.

**Describe the solution you'd like**
Let us specify that only the top-most layer is printed in concentric, in-between layers use zigzag/lines etc.

For example, if I specify 3 Top Layers. Let me specify that the Layer 1 and/or 2 above infill are one pattern, e.g. zigzag, and Layer 3 (topmost, exposed to air) another, e.g. concentric.

**Describe alternatives you've considered**
Alternatively, specify that the infill>shell layer concentric pattern be printed from outside-in, so that each line can at least adhere to the outer shell layers.

**Affected users and/or printers**
All FDM printers?

Ghostkeeper, 2019-07-29:
This is already implemented. Set the Top Surface Skin Layers setting to 1 and then change the Top Surface Skin Pattern to Concentric.

closed 2019-07-29 by Ghostkeeper

Firedrops, 2019-07-29:
Oops. Thank you!