Ultimaker/Cura #6072: Better Z-seam visibility in preview.

JohnEdwa, 2019-07-28:
I like to optimize the Z-seam for most of my prints to be as hidden as possible, but it's really quite hard to spot where it actually ends up at - unless I enable coasting, that is, which makes is blatantly obvious:

[image]

But as I have linear/pressure advance tuned properly in my firmware settings, I can't actually use coasting as it will then leave an actual visible gap in my print. (Well, as properly as you can with a bowden system, so I still get a small z-seam blob no matter what I do.)    

Therefore, would it be possible to add an option to render a gap or a blob or something extra visible there, even without coasting enabled, that would show where the Z-seam will end up being better?

Ghostkeeper, 2019-07-29:
The layer view currently doesn't draw anything other than the g-code. When we implement this, we'd have to consider how to detect the seam properly when loading a g-code file from file, for instance.

smartavionics, 2019-07-29:
But the gcode contains comments that tell you what the next section of gcode is printing:
````
;TYPE:WALL-OUTER
````

So can't the g-code reader just indicate the position of the start of that section?

Ghostkeeper, 2019-07-30:
Yeah I was thinking that as well but for some profiles all of the outer walls of a part are drawn before moving on to the next structure, so there will be a seam at the beginning of the first outer wall or the end of the last outer wall, but not for the outer walls it draws in between. And it also fails if you have only one wall (no inner walls).

smartavionics, 2019-07-30:
That's all true but it could notice when there are travel moves within the outer wall (maybe moves > some threshold distance) and consider that the end of the current wall. Also, changes in z would signify the end of the current wall.

Ghostkeeper, 2019-07-30:
Yeah, but that won't work with the Minimum Wall Flow setting.

The most robust technique would be to put `;SEAM` comments in the g-code, I guess.

smartavionics, 2019-07-30:
Good point. Yes, a ;SEAM comment is probably the way to go.

nallath, 2019-07-31:
I think adding more "hints" or helper like markings into the layer view is a pretty good suggestion. I'm trying to think of a few others that we might want to add as well. Does anyone have any further suggestions?

JohnEdwa, 2019-07-31:
The only thing that comes to mind would be highlighting the Post-Processing Modify Gcode plugin "pause/filament change/insert at" etc layer with a transparent plane through the model. Though honestly, it would be mostly because I always forget to disable it before I slice the next model :P

Ghostkeeper, 2019-08-02:
There are quite a lot of things already in the g-code, like the pause commands and temperature commands, that we don't show currently in the print preview.

I can't of anything else currently that is not easily parseable from the g-code.

thesec, 2019-12-08:
+1 on getting the seam to show in preview from me :)

As a workaround, one can temporarily add long coastings to expose it.

BasF0, 2020-09-15:
Hey I started working on this, check https://community.ultimaker.com/topic/33917-skin-starts-z-seam-simulationview-mod/

Ghostkeeper, 2020-09-17:
Nice work. I would've visualised it as a 2D circle in line with how travel moves are shown I think, or as an upwards line similar to how retractions were visualised in Cura 15.04.

BasF0, 2020-09-17:
Yeah I wanted to view the seam positions in 3D so that you can see them from every angle on the surface, but it does get in the way of showing the geometry a little bit, perhaps. Anyway that's why I did add a toggle to show/hide the starts, just as you do with travels, helpers, shell and infill..

I'm not sure what that looked like in Cura 15.04 as I don't own a 3D printer that long, but it sounds to me that that is mostly effective at showing the starts in the visible top layer instead of on the whole model ? Also wondering (I can't tell) if you would still prefer your method over how I've done it.

Also, my implementation currently doesn't work with gcode, not even with gcode generated from cura. It seems that the cura preview can properly differentiate shell/infill/helpers when that gcode was generated from cura, but with a non-cura gcode file it sees everything as a single layer.. Maybe I'll see what I can do with cura-gcode.

thesec, 2020-09-17:
The way it is visualised right now seems perfect to me as it clearly shows the actual seam as it shows (ideally not) on the print.

That said, I would not vote for changing it to a 2D interface which seems not matching the intended visualisation of the seam, not start (or end) points and is more likely to disturb the rendering of the print and perhaps even hiding the seam at critical edges.

Being able to toggle the feature seems therefore a perfect addition.

Ghostkeeper, 2020-09-18:
If you have a stable version, feel free to publish it on https://contribute.ultimaker.com if you'd like more people to see it.

The Cura team has a feature in the backlog, about halfway down, to visualise the Z seam in layer view. If we start working on it and we know that there is a plug-in in the Marketplace with the same functionality, we'd let you know. I don't expect we'll get around to implementing this feature this year unless plans change.

nallath, 2020-09-21:
Why not make a pull request out of it? I think it makes a lot of sense to have it in the simulation view. As far as I can see from the changes made, they all "live" in the simulation view anyway (I also think that would be tricky to get working as an extra plugin!)

Ghostkeeper, 2021-03-15:
This PR is merged for version 4.9 so this feature is implemented then. Thanks!

closed 2021-03-15 by Ghostkeeper