Ultimaker/Cura #6073: master branch issue with z hop speed validity test failing

smartavionics, 2019-07-28:
**Application version**

master

**Platform**

Linux & Windows

**Printer**

Creality Ender 3

**Reproduction steps**

Usual steps, load model, select profile. Then ensure that  Z Hop Speed is <= machine_max_feedrate_z and click slice button!

**Actual results**

Slicing doesn't happen and this error is displayed:

[image]

**Expected results**

Slicing should go ahead as normal

**Additional information**

Note that the Creality definition sets Maximum Speed Z to a low value (5). Because the default value of Speed Z Hop is 10, that will show up red in the Creality profile but even explicitly setting Speed Z Hop to 5 (which makes the red go away because it's now not less than the max allowed) doesn't allow the slicing to be done and you get the above error.

But if you then increase the value of Max Speed Z so that it is >= 10, it can then slice. 

Therefore, it must be comparing Max Speed Z with the default value of Speed Z Hop (i.e. 10) rather than the value set in the profile. Incidentally, that setting is per-extruder, could that be the cause of the problem? Standby... I just tested that hypothesis by making it not per-extruder and then I can slice with both Max Speed Z and Speed Z Hop equal to 5.

Summary: allowing Speed Z Hop to be settable per extruder and giving it a default value that is larger than the value of Max Speed Z will cause the above error to occur.

smartavionics, 2019-07-29:
BTW, I should think this problem will be in 4.2 also!

nallath, 2019-07-29:
I can't reproduce this. If I put the Z_hop speed at 5 it slices as expected.

smartavionics, 2019-07-29:
I've just realised it's slightly more subtle. Try loading a profile that has a z hop speed > max z speed and try and slice while the z hop speed box is red. It will give you the can't slice message (which shows the setting that's bad). Now change that setting to be <= the max z speed (box loses red color) and try and slice again. Now you should see the buggy message.

nallath, 2019-07-29:
I've done the following:
1. Create a profile (for creality-3) with z-hop speed 9
2. See that field is red.
3. Try and slice
4. Fails (expected)
5. Change value to 2
6. Slice again
7. Doesn't fail (expected)

smartavionics, 2019-07-29:
Hmm, I just tried using the CR-10 printer and you're right, that's how it behaves, OK.

So, don't know what to think about this. I'll close the issue and we can always reopen it if required.

nallath, 2019-07-29:
You gave me a bit of a scare (because if this was an issue, we should fix it for 4.2). Thanks for looking into it anyway!

PerChave, 2019-07-30:
Hi,
I have same issue when support is enable. I tried to make a profile and change the z-hop speed value from 9 to 2. It doesn't work...

smartavionics, 2019-07-30:
Reopening this as it obviously is still an issue.

xc-racer, 2019-07-30:
I'm seeing the same issue w/ 4.2.
Windows 10 x 64 (running Win 8 compatibility mode). 
Printer is a CR-10S (if that matters).

Alaric323, 2019-07-30:
Copying what I mentioned in 6085; this only happens with Support enabled for me.   If disabled, zhop speed does not cause a crash (probably because it doesn't zhop without support though)

atmora77, 2019-07-30:
Same issue..... What the hell...

Lunixes, 2019-07-30:
4.2 update makes Cura useless. Cannot print with support: "Unable to slice" when printing with supports.

pfritzd, 2019-07-30:
Same issue as above ...  When I add supports, I am unable to slice due to the Zhop speed

Installed CURA 4.2 on Windows 10 machine and configured to use the Ender 3

smartavionics, 2019-07-30:
People, a workaround for this issue is to install the printer settings plugin from the marketplace and then increase the max z speed value to 10 (or more). It should then be able to slice without barfing.

randellhodges, 2019-07-31:
For me and 4.2, using the printer settings plugin, I get no options under the printer settings. I'm digging thru my configs to see if maybe something didn't make it across when coming from 4.1

diegopradogesto, 2019-07-31:
We are working on it right now. We found the problem and we plan to release a hotfix ASAP. Keep you informed.

Thanks for all the feedback.

emmanuelvende, 2019-07-31:
Same issue with Creality CR-10 profile. When trying to slice with "generate support" capability activated, even with a Z-hop speed forced to reduce to 5 mm/s (default value 10 is not supported the field appears in red and I must force it to 5 or lower), the error message "Unable to slice" is displayed without any further detailed information.

p99oi, 2019-07-31:
Hello, I got the same issue with a Ender-5...

Ghostkeeper, 2019-07-31:
I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/6085. It was fixed by this commit: https://github.com/Ultimaker/Cura/commit/1d7f2e645e3489eedac41fb20f98b9f66c100b28

We're releasing a 4.2.1 build probably tomorrow morning (Geldermalsen time).

Pyrriuss, 2019-07-31:
The only solution for now is to create yourself your supports on the 3d modeling application

Ghostkeeper, 2019-07-31:
You can also increase the Maximum Z Speed to 10 with the Printer Settings plug-in from fieldOfView as a workaround.

closed 2019-07-31 by Ghostkeeper

smartavionics, 2019-07-31:
@Pyrriuss , Ultimaker have said they will release a fixed version tomorrow and I will be releasing a fixed version of my Cura build later today.

smartavionics, 2019-07-31:
@ghostkeeper, are you expecting any more changes to the 4.2 (and master) branches that are related to this issue today?

nallath, 2019-07-31:
Not anymore. We just fixed one thing regarding a profile import that got broken. So the state that 4.2 is in now is expected to become the 4.2.1 release

smartavionics, 2019-07-31:
OK, thanks, I'll make a release from what is there now.