Ultimaker/Cura #6075: Calculated Speed or Temperature for simple speed boost

UltimateSports, 2019-07-29:
the crux
would it be possible to have a fx that would meant that the temp is controled by speed changes or visa-versa?
the story, 
we use a couple of printers for R&D and small batch production, when it comes to quick turn around trials we get a little backlogged,
therefore I have been experimenting with speed/temp increases, but this can be a little hit and miss at times,

Ghostkeeper, 2019-07-30:
We could make a formula, but if we create that formula it would be just as hit-and-miss as when you create it.

We tried to adjust temperature to the flow rate at some point. The result of that was the "Auto Temperature" setting, which is now hidden from the interface:

https://github.com/Ultimaker/Cura/blob/2a7b887eb2b974c21d9ecd95a58a0f43210dbc8c/resources/definitions/fdmprinter.def.json#L6113

It was not a successful experiment. Adjusting the speed indeed means that the temperature should be adjusted to make the material more runny in order to get enough flow, but the relation is not as linear or predictable as we'd like it to be.

GregValiant, 2024-11-25:
I'm cleaning house here.
Is this request still relevant in current Cura versions (5.8.0 and up)?  Can this be closed?

closed 2024-12-09 by github-actions[bot]