Ultimaker/Cura #6077: [Cura 4.1] Dialog/PopUp opening at actual Display Position

volkaoth, 2019-07-30:
Many Prompts or PopUp Dialogs that are not bind to the menu bar poping up at the first starting Position at the Monitor. Not at the actual Displayposition.
Moving Cura over multiple Monitors and searching for opend PopUps on the Cura StartUp Position on a different Monitors is not nice.

**Describe the solution you'd like**
use the actual Display coordinates from Cura as PopUp Position for Dialogs

Ghostkeeper, 2019-07-30:
Hmm, for me it always opens those pop-ups in the centre of the screen. But that may be Ubuntu's doing. Which operating system are you using?

volkaoth, 2019-07-31:
Im using: 
Ubuntu 18.04.2 LTS
Kernel-Version: 4.15.0-55-generic
Window-Manager: Openbox

Ghostkeeper, 2019-08-02:
Hmm, I'm using Ubuntu 19.04 and I can't reproduce this.

We order Qt to bind the pop-up to the window. Maybe it's Openbox that does this differently then?