Ultimaker/Cura #6078: [4.1.0] Last layer does not retract, causes stringing

antonio-fiol, 2019-07-30:
**Application Version**

4.1.0

**Platform**

Windows

**Printer**

Ender 3

**Steps to Reproduce**

Slice with layer height=0.08

**Actual Results**

Resulting gcode runs the last layer with travels without retracting the filament, resulting in stringing.

**Expected results**

Retraction before travel should occur.

**Additional Information**

[CE3_Rueda Molino Integrada - Aspas altas-008.zip](https://github.com/Ultimaker/Cura/files/3445689/CE3_Rueda.Molino.Integrada.-.Aspas.altas-008.zip)

smartavionics, 2019-07-30:
You need to reduce the minimum extrusion distance window setting. I reduced it to 1 and you then get retraction on the top layers.

Ghostkeeper, 2019-07-30:
Yup, that seems to have been it. The number of retractions is being limited to protect the filament against grinding. If you need it to retract more often at the cost of reliability of the print, you can increase the Maximum Retraction Count setting or reduce the Minimum Extrusion Distance Window setting.

See also https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/descriptions/retraction_count_max.md

This is expected behaviour, so we'll close the ticket.

closed 2019-07-30 by Ghostkeeper