Ultimaker/Cura #6079: Polygon.simplify doesn't work as intended

BagelOrb, 2019-07-30:
**Application version**
current master

**Reproduction steps**
collinear_polygon = (57202,12388), (56402,12394), (56393,12394), (56382,13625), (57632,13633), (57621,12385), (57602,12385)

small_poly = (15200,15647), (15201,15647), (15201,15650)

**Actual results**
(57202,12388) is not removed from collinear_polygon
small_poly remains

**Expected results**
(57202,12388) should have been removed from collinear polygon
small_poly is removed altogether

**Additional information**
Polygon.simplify was designed to reduce small segments *and* (nearly) collinear verts.
Currently it doesn't work as intended: collinear line segments are not removed and some small polygons are not removed either.

We should compare with its functioning before it was rewritten a couple of months ago.

This is very dangerous, but it requires some serious investigation to see in what this bug actually results. I think it requires too much work to investigate in which 3D model the problem actually pops up and in which places in code collinearity is assumed not to exist, but I think it is very important to fix this problem.

I know it is difficult to convince the project lead about the importance of this problem. We can have a discussion about it here, but I will need you to be on my side in this one, because it's difficult to reason about this sort of thing; please help me to find the right argument why it is so important to fix such an elementary bug in the geometrical base class.

Ghostkeeper, 2019-07-30:
We have one automated test that tests this: https://github.com/Ultimaker/CuraEngine/blob/4831443d1333706f9dfaed9b2e878335d879afbf/tests/integration/SlicePhaseTest.cpp#L45

This cube is being sliced and must get simplified to 4 vertices in order for the test to succeed. Every layer would otherwise intersect with 8 triangles, 2 for each quad which are coplanar. The test is succeeding so it seems to be doing something right. Also when I slice objects with coplanar faces they all get simplified properly to single straight lines.

I wonder what is going wrong with your test case then.

BagelOrb, 2019-07-30:
It might very well be the case that these two polys are hole-polygons. Maybe that's the difference.

BagelOrb, 2019-07-30:
Awesome!
Does that commit fix both issues?

Ghostkeeper, 2019-07-30:
I've made another test, unit instead of integration level. This raised an edge case where the second vertex was not being checked for colinearity. It might fix your issue if you found that it had one vertex too much but the rest was being simplified properly.

I don't think them being hole-polygons has anything to do with it. This algorithm doesn't work with winding order or points inside polygons, just with line lengths and distances to projections on lines.

Ghostkeeper, 2019-07-30:
The second case is expected behaviour. The very top of the simplify function explicitly skips simplifying it if there are exactly 3 vertices, because simplifying that could make you end up with 2 or 1 vertex in your polygon which was deemed to be not a good idea.

BagelOrb, 2019-07-30:
The algorithm works with volumes of area removed, so it might have not worked for negative volumes. See `area_removed_so_far`.

The second case is not expected behavior in my book.
What the code looks like is never a good argument for saying what the expected behavior is...
The `Polygons::simplify()` function explicitly removes polygons where `PolygonRef::simplify()` ends up with less than 3 points.

What I expect the behavior to be is that if according to the spec the polygon ends up with too few edges then the whole poly should be removed, so all points in the poly should be removed.

I don't think the second case is super important, but it is not according to spec.

Ghostkeeper, 2019-07-30:
Ah, good point. I'll update the spec then and remove the dead code from `Polygons::simplify()`.

BagelOrb, 2019-07-30:
What do you mean?

Those two statements seem opposite. You don't need to update the spec if `PolygonRef::simplify()`  *can* end up with less than 3 points.

BagelOrb, 2019-07-30:
I just ported your code to mine and it seems to do something terribly wrong:

before:
[image]

After:
[image]

BagelOrb, 2019-07-30:
Also: I find it really strange that there are hardcoded values in there for exactly the parameters of the function! Why not just use `allowed_error_distance_squared` instead of 25?

BagelOrb, 2019-07-30:
It currently seems to reduce any square to nothing.

BagelOrb, 2019-07-30:
(Edit: scrap my last couple of uncaffeinated comments. [comments deleted])

BagelOrb, 2019-07-30:
Your fix seems to have fixed my original issue! :+1: 

As for the special case of a small poly with 3 points:
According to the logic of removing something which has small line segments or even according to the logic of removing areas with a height smaller than the projected_error_dist we require the polygon to be dissolved.
I don't see why you want to update the spec instead of the code. I don't see a usecase for wanting polygons of 3 verts not to follow the same logic.

Also note that the simplify algorithm might reduce a 4 vert polygon to two verts, so starting with 3 verts is awkward to make into a special case.

Ghostkeeper, 2019-07-30:
Good point about the 4-vertex polygon. This is not something I can fix in 5 minutes and as you say this is base geometry code so it's risky to change, so I'll put up a note for that to discuss with the team tomorrow.

In general, if there is an algorithm that breaks if two line segments are exactly colinear, I would try to fix that in the algorithm. CuraEngine transforms polygons very often without calling simplify() every time. Clipper doesn't guarantee to remove those extra vertices after an inset or offset operation, or a morphological operation, so the algorithms that run on polygons need to be able to handle that case. Clipper does remove loose vertices and parts of the polygon with 0 width, so in general we assume that these don't occur except if we construct the polygon ourselves from coordinates. That's why this function needs to guarantee to not return 1 or 2 vertices.

BagelOrb, 2019-07-30:
This is super easy to fix right?
Just something like 
```
if (path.size() == 3)
{
if ( getLength() < reduction_length * 2
     || std::min( dist2Line(a, b, c), dist2Line(b,c,a), dist2Line(c,a,b) ) < allowed_error_dist )

{
path.clear();
return;
}
}
```

BagelOrb, 2019-07-30:
This function is made (partly) for removing collinear lines, so it needs to guarantee that.

My algorithm is robust against collinear lines because it calls polygons.simplify before handling the polygons.

Fixing the algorithm in some other way instead of calling simplify will actually make the code needlessly complicated.

closed 2024-12-07 by github-actions[bot]