Ultimaker/Cura #6080: Ironing option stops  middleprint cura 4.0

Erokawa, 2019-07-30:
**Application version**

Cura 4.0
**Platform**

Windows 10 64bits version 1804
**Printer**

Custom printer
**Reproduction steps**

Activating Ironing 
**Actual results**

The printer prints but ramdomly stops and the temperature of the extruder and bed will keep heating
**Expected results**

The printer won't finish the print, it'll stop suddenly
**Additional information**

Ghostkeeper, 2019-07-30:
I don't see anything wrong in the g-code. The ironing parts don't have any extrusion so Cura sees them as travel moves, but there is no pause command or heating command anywhere in between the ironing commands. I don't know why your custom printer is pausing halfway.

Are you printing via USB cable or via local file? Perhaps the printer couldn't keep up with reading the file?

Erokawa, 2019-07-30:
Hi Sakamoto-san, i have to point up that i had another hardware before (MKS GEN 1.4 - TMC 2208 - ANET A8 EXTRUDER) and it happend the same. That time i just reinstalled cura and doing a new profile. It didn't stopped since some days ago when putting on ironing and it happens. I'm printing by tft32 touch panel with SD card.

Ghostkeeper, 2019-07-31:
> Hi Sakamoto-san

:laughing: You know that's the first time someone actually recognised it and pointed it out?

I don't see anything in the g-code that should halt your printer, really. Maybe we can make a more controlled test of a cube with ironing or something?

Erokawa, 2019-07-31:
Definitly not the ironing... for some reason with a simple model i made with Fusion 360 it stops some seconds and continues after, for sure are buffer size problems because maybe i didn't put M502 when i uploaded the firmware. I'll close this because the ironing definitely is not the problem, sorry for the trouble and thanks

closed 2019-07-31 by Erokawa