Ultimaker/Cura #6081: [4.2] Can't increase marketplace tab dimension by dragging

Liger0, 2019-07-30:
**Application version**
4.2

**Platform**
Win 10 x64

**Printer**
It doesn't matter

**Reproduction steps**
Open the marketplace

**Actual results**
The tab can't be enlarged by draggin on the margins and the tab is too much small.

**Expected results**
It should let users enlarge the marketplace tab.

**Additional information**

rburema, 2019-08-08:
I assume you mean the entire marketplace window? Also, it might help to have a screenshot, 'too small' can mean a lot of subtly different things.

Vandrasc, 2019-08-08:
I will add the feature request label as this is currently how the window was implemented thus works as intended. It is a good idea to make it possible to resize the window though, I think we discussed the possibility already. I will discuss bring it with the team.

Vandrasc, 2019-08-09:
This will also be addressed in CURA-6564.