Ultimaker/Cura #6082: [4.1] Nozzle temp not inherit by the material

Liger0, 2019-07-30:
**Application version**
4.2

**Platform**
Win 10 x64

**Printer**
Cr10S

**Reproduction steps**
Change the material with a nozzle temp different than the profile's one.

**Actual results**
The profile won't adjust its temp to the material one, not even with the icon to revert to a calculated default

**Expected results**
The profile should use material nozzle temp like the bed temp does.

**Additional information**

nallath, 2019-07-31:
By the looks of it, your quality changes overrides the default_printing_temperature. If I change the material temperature it changes as it should. 

Could you share your project file that contains that profile so i can verify that that is the case?

Liger0, 2019-07-31:
If you use a base profile, it wil work as intended. If you make a new one based on it changing the temp, it won't. You have no button to reset to calculated/base profile value either.
[nozzle temp.zip](https://github.com/Ultimaker/Cura/files/3451666/nozzle.temp.zip)

Ghostkeeper, 2019-07-31:
I can't reproduce this issue with the reproduction steps above.

If you override the Default Printing Temperature with a custom profile then there will be no fx button because it was never a formula.

Liger0, 2019-07-31:
> 
> 
> I can't reproduce this issue with the reproduction steps above.
> 
> If you override the Default Printing Temperature with a custom profile then there will be no fx button because it was never a formula.

On older releases I could change the profile one and when importing a material it would still be able to use the material one. Now there is no way unless entirely deleting the profile.

Ghostkeeper, 2019-07-31:
Yes, your profile `(4.2 CR10s) PLA 0.4mm 0.12LH` overrides the default printing temperature by setting it to 215. Cura is doing what you asked it to by setting the default printing temperature to 215C.

Ghostkeeper, 2019-07-31:
> On older releases I could change the profile one and when importing a material it would still be able to use the material one. Now there is no way unless entirely deleting the profile.

This was a bug then in the older releases, because the custom profile should ALWAYS override the material profile.

Liger0, 2019-07-31:
> This was a bug then in the older releases, because the custom profile should ALWAYS override the material profile.

So what if I want it to inherit the temp by the material?

tlhintoq, 2019-08-01:
I'm with @Liger0 on this.  Saw and reported the same issue in the past.
The behavior of the profile overriding the material temp make the use of materials moot.  Materials now have no point except to calculate the cost of material used.  Beyond that they are worthless now.

Scenario: I have a profile set up for... printing solid functional parts... it defines 4 walls for strength... coasting volume... and jerk so I don't get ringing in my print.
I want to print this in "Widget Brand Red PLA" - that's 206°
I want to print this is "Gadget Brand Blue PETG" - that's 252°
{...}

Are y'all really saying I have to set up profiles for every Situation times the Materials?
Fine Fast Widget Red PLA
Fine Fast Widget Red PETG
Fine Fast Gadget Blue PLA
Fine Fast Gadget Blue PETG
Fine Fast Zyletch White PLA
Fine Fast Zyletch White PETG
Fine Fast Overture Black PLA
Fine Fast Overture Black PETG
Fine Fast MatterHackers Green PLA
Fine Fast MatterHackers Green PETG
{...}
Repeat with Rough Draft

**Are you kidding me?**

nallath, 2019-08-01:
No. You seem to fail to understand how materials work (and failed to actually read the tool tips). So before getting all mad at me, you might want to consider doing that. But I'm going to try to explain it once more. 

"default printing temperature" is **not** used by the engine. Ever. If you take a look at the ultimaker print profiles, you would see things like:
```
material_bed_temperature_layer_0 = =material_bed_temperature + 5
material_final_print_temperature = =material_print_temperature - 10
material_initial_print_temperature = =material_print_temperature - 5
material_print_temperature = =default_material_print_temperature - 15
material_print_temperature_layer_0 = =material_print_temperature + 3
```
So if you just set the base material temperature to whatever you want, it will work (because the default of those setings is just directly inheriting it from the default temperature.

In short, I'm not saying you should set up profiles for every material. You could if you want to, but if you'd take some time to read  (or experiment) in on how these settings work (or even ask nicely), there are better ways to do it.

tlhintoq, 2019-08-01:
@nallath Who's mad at you?  Nobody even mentioned you or made mention of anything you said.  Chill dude.  Switch to decaf or herbal tea.
-------------
Tried to attach a video - apparently that's not allowed.  Let me try to explain this process.  Whomeever tries, should see the same results. if not let me know and I'll upload the video to youTube or something.

Material A - 220°
Material B - 210°
And a custom profile.

Pick material A and pick your custom profile. Cool, you're at 220
Change to material B. Material become 210 but  Cura is changing the profile as well to "Standard".  WTF? So you get the 210 from the profile and all your custom values are gone.
So you change the profile back to where you had it on your custom profile:
The temp jumps back to 220 instead of honoring the material temp of 210. So the material values don't actually mean anything.

You can say its been like that - or its how a developer thinks it should work - but its not what a **user** expects.  It pretty much ruins the use of materials and I'm back to keeping a spiral bound with all the values I need for each printing profile and material.  Having to reset (or at least check) which values have and have not changed.

nallath, 2019-08-01:
Tying in caps or in bold does signal some sort of anger, especially in the context of "Are you kidding me". Might not have been the intent, but that's generally how it's used. 

You can just select material B with your custom profile. That way you will have 210 and all changes that you made. The reason it jumps back to 220 is because you put that value in your custom profile (instead of putting it in the material. 

The material temperatures do mean a thing, but they can be overridden by a user profile. As I already showed, in some cases this is simply required (eg; printing normal quality PLA might need different temperatures than printing low or high quality PLA). That's also why the (naive) notion that putting a printing temperature in the material profile is simply wrong.

Liger0, 2019-08-01:
Okay so how do I delete the temperature from the profile to use the material one? There isn't a button to delete this value so that the profile temp won't override the material settings.

nallath, 2019-08-01:
That is an issue right now, there is no way to remove settings from the quality_changes profiles. You could check the profiles as stored on your computer and remove the default_material_print_temperature line from them.

tlhintoq, 2019-08-01:
> Tying in caps or in bold does signal some sort of anger, especially in the context of "Are you kidding me". Might not have been the intent, but that's generally how it's used. 

Maybe its a generational thing looking for signs of anger where none exist.  If you get triggered because someone puts something in bold, go pet your therapy dog for half an hour but don't put your issues on the rest of the human race and their use of the English language. You can't really ask millions of other users on the internet to not use bold because you think its indicative of anger.

Bold is an emphasis.  Nothing more.  The beauty of textual communication is the LACK of emotional context.  Even if it was... lets say for sake of conversation and playing devil's advocate.  Why are you taking it as being directed toward _you_?  You had one comment at the start of the thread asking for the OP to share a project so you could test it.  To assume that every comment after that is directed toward you is... presumptuous at best.

Honestly I think you're being more than a little condescending toward other users.  To say someone is naive for putting in a temperature in the material definition is well - smug at best.  How is it naive to use something provided by the software.  The software.  I've been a software engineer for 22 years.  I don't add properties like 'temperature' with the expectation they _not_ be used by the user.  And I don't berate users for actually using the properties and features that they are given.

In other words - If you feel its naive to set a temp in the material definition - Then what is it there for if _not_ to say "I want this material to print at this temp" ?

From a user perspective - the natural expectation of use and order of inheritance is:

- You set a profile for how you expect to print in general: Fast or draft, with support... A general type of print for a genre of model.  A profile for 28mm miniatures... a profile for big parts like shelf brackets... a profile for high detail... a profile for low detail.  That sort of thing.
- Then you apply your material **on top of that**.  There's only like a half dozen values in a material and some of those aren't printing related but display color and cost calculation.  That leaves... what... 3 that **are** meant to affect printing specifically for that material. Temperature. Material density. etc.

The expectation by the user is:

- Apply the 200+ settings from a profile
- Overlay the 6-ish settings from a material

Its not that tough to comprehend how the users expect to use these two settings groups. I ask in all seriousness and without emotion: Do you really not grasp how the two of us are describing our useage expectations? Because if you don't get it... if you don't grok how full-time users of the software are using it in their shops, shifting from 5 machines, 4 materials, a dozen or more colors, and many categories of prints from 10mm small detailed items to 400mm big items.... I'm happy to go into more detail about shop workflow and expectations and even invite you to my shop to see it in person.  I know when I'm making software for a use outside of my experience that seeing it in its native habitat often does wonders.

tlhintoq, 2019-08-01:
> That is an issue right now, there is no way to remove settings from the quality_changes profiles. You could check the profiles as stored on your computer and remove the default_material_print_temperature line from them.

Let me confirm my interpretation of your suggested 'fix'.
You're suggesting that every time someone tweaks a profile in the software they...

1. Close the software
2. Open the profile file in Notepadd++
3. Delete the line you mention
4. Save the profile
5. Re-open Cura
6. Re-load the project
7. And hope the abundance of other issues like random changing of profiles _by cura_ when you change materials doesn't screw up everything else you've been adjusting.

Have I correctly interpreted your expectation of the user to work around this behavior bug?

tlhintoq, 2019-08-01:
I have to say I'm stunned.  Looking at this thread from the perspective of a 2 decade software engineering veteran.

This isn't rocket surgery people.

There are like 3 settings in the material definition that affect printing.  Its pretty common sense the way users are saying they expect them to work:  Apply the profile then overlay the material.

Its the exact same logic used when you have groups and users.  A group can have "edit" permission then a user has that revoked.  Or a folder has read/write, but a nested folder is read-only.  Guys I know you deal (or have dealt) with this same logic of narrow-scoped settings taking precedence over broad-scoped settings in a dozen other situations in your development careers.  Why all the push-back on this?

nallath, 2019-08-01:
If we are on the topic of language, a "Naive approach" in software engineering is the most straightforward way to do it. It has little to do with stupidity, just with the lowest amount of complexity. As I've already posted before, we used to have this strategy. So if it's a sign of stupidity, we were there once as well. 

I explained how your message came across and why my reaction was thus. You gave an explanation as to how you meant it. That you continue feeling the need to defend what you did (and suggest i might need therapy) has more to do with you than with me imho. I'm not telling you what you should or should not do, just how it was initially perceived by me. Do with that information what you will. I can't make you after all ;)

We also didn't add a propery called "temperature" we added a property called _base material temperature_. We also didn't add it with the idea that no-one was going to use it. I'm pretty sure that if you made a hammer and subsequently someone uses it to wack in screws, which after an update doesn't work anymore, you would berate them. I mean, it's not intended to be used for that, was it? Sure, it might have been the manual that was wrong, but that's about it. 

The expectation that there are a few settings from materials overlaying is  **exactly what cura is doing right now**

Lets take the following example:
um3_aa0.4_abs_draft
```
material_print_temperature = =default_material_print_temperature + 10
material_initial_print_temperature = =material_print_temperature - 5
material_final_print_temperature = =material_print_temperature - 10
```
and UM3_aa0.4_ABS_Fast
```
material_print_temperature = =default_material_print_temperature + 5
material_initial_print_temperature = =material_print_temperature - 5
material_final_print_temperature = =material_print_temperature - 10
```
Those quality profiles, even though they override the material settings do just that. The material profile sets the default_material_print_temperature, so the quality profile will ensure that, depending on the quality, the other temperatures get set accordingly.

So i don't understand what your problem is, but as far as I can see it it's simply not there.

nallath, 2019-08-01:
As for the manually editing it, I was trying to provide a work around. Might be me not having 21 years of experience, but i asumed that people do prefer a work around instead of a big fat NOPE.

Liger0, 2019-08-01:
I am unsure how that temperature offset can be set in the profile. I see there a fixed temp can be set, but not a positive or negative offset that affects the material.

nallath, 2019-08-01:
You're right that that is a big issue with the interface right now. The profiles can set it, but the GUI doesn't have a way to do that. It's something that I've been wanting to implement for a very long time now (and would probably fix a lot of the issues that everyone is having right now). But I don't get to decide the priority. 

My ideal case would be that it works a bit like excel, so every setting field gives you the option to type in a formula (or just a value if that's what you want).

nallath, 2019-08-01:
Also: I've removed the value from that profile for you if you don't feel comfortable doing it yourself. 
I tested it already and it doesn't cause any crashes.  
[fixedprofile.zip](https://github.com/Ultimaker/Cura/files/3457008/fixedprofile.zip)

tlhintoq, 2019-08-01:
I like how that sounds.

So let me ask this - and maybe it will clear up a little bit of confusion (some one my part) - that I think stems around the verbiage of the dialog.    Ok the term in the materials box is "Default printing temperature."  From that Cura does things like calculate a final layer temperature that is 10° cooler than the default.  And a first layer temp that is 10° hotter so you get good bed adhesion. Or whatever the calculations are +- off that default temp

Here's my question: Regardless what its called... Why is the material value not overriding the profile value?  That would seem the natural hierarchy.

I go back to the steps I mentioned earlier.
You select a material and the profile changes - maybe fixing that will fix the downstream issues from there.
So then you re-select your profile - But the profile default temp overrides the already selected material default temp - BAM - right there is the flaw in the logic taking place.  To me anyway

nallath, 2019-08-01:
This is going to be a bit of a long story, so hold on a bit. Cura uses a lot of profiles, which are ordered in a stack. Each machine has at least two of these stacks (Extruder stack and a global stack). These stacks each contain a bunch of profiles. The extruder stack is also connected with the global stack.
The extruder stack has "User", "quality_changes", "quality", "material", "variant", "definition_changes" and definition"
The global stack has a "User", "quality_changes", "quality", "variant", "definition_changes" and definition.

So if we want to know the value of a given setting (lets say line width), which is a setting that can be set per extruder, we ask the extruder stack what the value of that setting is. The extruder stack will ask each of the profiles in turn what the value is and return the first answer it gets. If it can't find an answer, it will continue with the global stack. The definition at the bottom of the global stack holds all the default values.

Depending on how you machine is configured, you can also have something called material specific qualities. Previously, the creality was set up super simple. It pretty much used defaults for everything and only changed a handfull settings in the quality profiles (and those didn't change depending on the material at all) and if you want good prints, these should change.

The creality mod made those changes. So instead of having a "generic" normal profile, it introduced the notion of there being a 0.4_pla_normal profile. So it actually became a whole lot smarter about the way it handled things. 

Now, the main reason for it not overriding it is pretty simple; It would mean that the fundamental structure of how these profiles work isn't there anymore. We used to have that system in 2.1, but having exceptions all over the place (eg; Some settings get taken from one profile exclusively, others get taken from another) is a recipe for disaster. Cura 2.1 worked, but the profile system was a clusterfuck to maintain (and totally didn't fucking work for dual extrusion machines, but that's a different story).

So being able to override the base material temperature in the quality changes is probably allowing a user to change a bit more than is good for them. But preventing it might cause a whole lot more weird issues (eg; Being able to do it before, now it doesn't, etc).

Liger0, 2019-08-01:
But seeing the nozzle size doesn't change the profiles as in my other ticket with profiles not switching properly, and materials not being editable becaue they can't be clicked, I don't think the new creality profiles are being done so correctly.

tlhintoq, 2019-08-01:
Great explanation.  I totally get it as we do something similar with truck driver devices and settings:
Company -> Carrier -> Terminal -> Group -> Truck -> Driver

> The creality mod made those changes.

Are you implying that if I define a machine NOT based on a Creality base then a lot of this should go away because it simply doesn't apply?  Or are you talking about some other mod for Creality machines?

> So being able to override the base material temperature in the quality changes is probably allowing a user to change a bit more than is good for them

**THANK YOU** for saying that out loud.  How about a checkbox for "Override material's value with profile value"?  People wanting that can still have it so it doesn't break backwards.  But people expecting a more natural flow can get it.

Thought: You mentioned no way of removing the temp override from the profile once its there.  This is only a bit clunky but as a step towards giving that to the user... if a negative value is entered then remove the property on save.  So if I go from an entered temp of 220 to -1 its kinda clear I want to remove that value.

nallath, 2019-08-01:
> Are you implying that if I define a machine NOT based on a Creality base then a lot of this should go away because it simply doesn't apply? Or are you talking about some other mod for Creality machines?

If you create a printer based on the custom FDM printer, you will pretty much go back to the way the creality printer used to work (eg; with generic profiles instead of specialized ones per material).

> **THANK YOU** for saying that out loud. How about a checkbox for "Override material's value with profile value"? People wanting that can still have it so it doesn't break backwards. But people expecting a more natural flow can get it.

Well, the problem is that i've seen so many workflows from so many different people that claim that their way of working is "The One True Way That Everyone In The Industry Uses" (Which is probably why i've become a bit skeptic towards people claiming that people expect a certain thing). 

I might be able to cobble a plugin together that checks these weird cases and warns the user about it. It's not a real solution, but teaching users these things can be pretty valuable. 

As for removing the actual setting. I think it's better to actually have a bit of interface for that somewhere. Entering a -1 is a bit of an obscure way to do it (eg; How would anyone know that -1 does that and what if a value can be negative? Or what to do for bool / enum values?)

tlhintoq, 2019-08-01:
> As for removing the actual setting. I think it's better to actually have a bit of interface for that somewhere. Entering a -1 is a bit of an obscure way to do it

Totally agree.  As I said its a quick fix for those that are b!tching about it (like me and others) that requires no UI change approval process and the 6 weeks that takes.

> How would anyone know that -1 does that 
The help bubbles that you pointed out everyone should be looking at anyway.

> and what if a value can be negative? Or what to do for bool / enum values?
I'm not going after an entire global mechanism in my conversation.  Just looking at the one complaint that there isn't a way to remove the temp value from the profile so the material value can take its place.  In that case negative values don't exist.

> I might be able to cobble a plugin together that
If you want to take the plugin route... What about an "overlay materials" plugin?  Is it just easier to leave the base app in place; change nothing. And make a plug-in for people that want this behavior to install?  Then you still get backwards compatibility. For the old behavior do things as you've always done. If you want to overlay a material on top of the current configuration select it in the plugin's dropdown.  Or something like that.

nallath, 2019-08-02:
Making a plugin that overlays the materials is theoretically possible, but that's about it. It would easily take me weeks of full-time development to build that and it would be a complete nightmare to maintain. It's a rather fundamental bit in how the structure works for it to be easily changeable.

Liger0, 2019-08-05:
> 
> 
> Also: I've removed the value from that profile for you if you don't feel comfortable doing it yourself.
> I tested it already and it doesn't cause any crashes.
> [fixedprofile.zip](https://github.com/Ultimaker/Cura/files/3457008/fixedprofile.zip)

It doesn't work.
I try to import it, but it even though it says "imported successfully", it doesn't import it.

Liger0, 2019-08-06:
Same for fan speed, it doesn't depend on the material.

nallath, 2019-08-07:
> It doesn't work.
> I try to import it, but it even though it says "imported successfully", it doesn't import it.

Huh. That's weird. I just tried importing the fixedprofile.curaprofile (with a creality CR-10S) and the profile shows up for me under custom profiles.

Liger0, 2019-08-07:
Any idea why removing the temp line would trigger this error? I am trying to mod the profile removing the fan and temp cases but with no success.

Ghostkeeper, 2019-08-07:
@Liger0 which temperature line is that? Nallath's profile has no temperature any more.
You posted a project file up above, so I tried loading that, then removing the default_print_temperature line over in the quality_changes profile that was loaded from the 3MF file, but that works fine. It then defaults to 244 degrees instead of 215.

An error will probably be in the log file. Maybe you can post your log file after getting that error. Or you could post the .curaproject file with reproduce steps here.

Liger0, 2019-08-07:
I mean. I take my initial profile and delete the default_print_temperature line, but the mentioned error occurs. While with nallath's modded profile I get no error but it doesn't import anything.

Liger0, 2019-08-07:
Here you see:
Nallath's profile isn't imported.
Untouched profile is imported.
If I delete the line from untouched profile, it gives an error.

https://youtu.be/yT4w0QdWFW0

tlhintoq, 2019-08-07:
@Liger0 
Just a thought... or two...
Turn on viewing of ligatures so you can see the invisible characters like end-of-line.  Maybe there is a weird condition in there like two end-of-line right after each other.
[image]

What character set is that showing?  Maybe there is an issue with it being saved outside of English.
[image]

Liger0, 2019-08-07:
No encoding at all is set by default to me.

Ghostkeeper, 2019-08-08:
Cura always uses UTF-8 for everything.

tlhintoq, 2019-08-08:
> 
> 
> Cura always uses UTF-8 for everything.

I get that.  I also get that it _should_.  Its 2 seconds for @Liger0 to confirm that her/his non-English PC is doing what is _expected_ and saving in UTF-8 and not converting it to something else to support the culture installed on hi/her PC.

Liger0, 2019-08-08:
I tried UTF-8, it still doesn't import with the same error.

closed 2024-12-07 by github-actions[bot]