Ultimaker/Cura #6083: [4.2] Profile import fails from single to dual extruder machines or inverse

Liger0, 2019-07-30:
**Application version**
4.2

**Platform**
Win 10 x64

**Printer**
Single to dual OR dual single extrusion machines (creality cr-10S a single extrusion, custom or creality cr-10X as dual extruder)

**Reproduction steps**
Create a profile on a single extrusion machines and import it on a dual extrusion machine.
OR
Create a profile on a double extrusion machine and import it on a single extrusion machine.

**Actual results**

The machine defined in profile D:/Profile.curaprofile (fdmprinter) doesn't match with your current machine (creality_base), could not import it.

**Expected results**
It should import successfully

**Additional information**

nallath, 2019-07-31:
This has never been possible (mostly because there has been little need for it. It's a bit tricky to do it, since in the +1 extruder case, you simply don't have settings for the second extruder (and might get some unexpected results if you have another material selected for that extruder than the first one)

Liger0, 2019-07-31:
It worked correctly on previous cura releases. I could import a cr-10s profile into a custom dual extrusion and vice-versa.

nallath, 2019-07-31:
I stand corrected then. I was under the impression that this simply wouldn't work. The upgrader script does seem to be able to handle it.

Liger0, 2019-07-31:
Just to give the profiles to test directly.

Use custom dual extrusion machine (or a creality cr-X) and import my single extrusion profile crated with 4.2: 
The machine defined in profile D:/File personali/3d printing/profili cura/(4.2 CR10s) PLA 0.4mm 0.12LH.curaprofile (creality_base) doesn't match with your current machine (fdmprinter), could not import it.*
[(4.2 CR10s) PLA 0.4mm 0.12LH.zip](https://github.com/Ultimaker/Cura/files/3451717/4.2.CR10s.PLA.0.4mm.0.12LH.zip)

Or

Use creality cr-10S and import my dual extrusion profile created on 4.1
[PLA 0.4mm + PVA 0.5mm 0.16LH brass.zip](https://github.com/Ultimaker/Cura/files/3451724/PLA.0.4mm.%2B.PVA.0.5mm.0.16LH.brass.zip)

closed 2019-07-31 by Ghostkeeper

Ghostkeeper, 2019-07-31:
It seems we had forgotten one printer in the list for upgrading instance profiles, the CR10s. Fixed for 4.2.1.

Ghostkeeper, 2019-08-01:
Ah, sorry. This ticket was not fixed. You still can't import these single-extrusion profiles into a dual extrusion printer because the base printer is not the same. We only allow that if the profile set is the same. That part is working as intended so we won't need to fix that.

What was fixed was importing into the CR10s in general.

Liger0, 2019-08-01:
Is this again a creawsome implementation fault? Since it worked before. :(

Liger0, 2019-08-01:
> Ah, sorry. This ticket was not fixed. You still can't import these single-extrusion profiles into a dual extrusion printer because the base printer is not the same. We only allow that if the profile [...]

I tried to import a cr10s profile into a custom single estrusion printer and it still doesn't work on 4.2.1 :(

Silithas, 2019-08-26:
> It seems we had forgotten one printer in the list for upgrading instance profiles, the CR10s. Fixed for 4.2.1.

Nope, i use 4.2.1 now, it's not able to import a 4.0.0 gcode as profile.

commit 5cf4cd03 (2019-07-31): Fix importing profiles for CR10s

It was somehow missing from this list. An oversight.

Fixes #6083.