Ultimaker/Cura #6084: [4.2] Failure to Slice with supports

lordpreto, 2019-07-30:
Application version
4.2

Platform
Windows 10 x64

Printer
Creality Ender 3

Reproduction steps
Installed Cura 4.2. Created new printer. Default Profiles. Check Generate Support box. Hit Slice button.

Actual results

Error Message: "Unable to slice with the current settings: The Following settings have errors: Z Hop Speed"

If I try to disable Z Hop or lower Z Hop Speed, the error message changes to "Unable to slice with the current settings: The Following settings have errors:"

Expected results
It should slice.

smartavionics, 2019-07-30:
See https://github.com/Ultimaker/Cura/issues/6073 (just reopened).

lordpreto, 2019-07-30:
I will close this one then, thank you

closed 2019-07-30 by lordpreto