Ultimaker/Cura #6086: [4.2] Can't Change Nozzle Size for Creality CR-10S Pro

Alaric323, 2019-07-30:
**Application version**
Latest Version, Cura 4.2

**Platform**
Windows

**Printer**
Creality CR-10S Pro, made using the new merged Creawesome Profile

**Reproduction steps**
Add the printer via "Add Printer", selecting CR-10S Pro from Creality3D Dropdown List.
Attempt to change nozzle size in Settings dropdown menu>Extruder 1 > Nozzle Size > 0.6

**Actual results**
Nozzle Size remains 0.4 mm

**Expected results**
Should switch to 0.6 mm

**Additional information**
Printer had hotend replaced with a MicroSwiss.  Was working fine in Cura 4.1, but when profile was remade using the new merged Creawesome profiles, the option to change nozzle size in the printer settings menu disappeared.

Attempting to force change the profile by adding the line [machine_nozzle_size = 0.6] into the configuration file fails as well (file reverts to original status, even if Cura is closed upon changing config).

nallath, 2019-07-31:
Good find.The work around is to select it from the custom configuration in the middle bar.

This will be fixed for 4.3

trouch, 2019-07-31:
Could be a profile import issue ?
I can properly set nozzle size with a newly added machine.

Alaric323, 2019-07-31:
Unfortunately not the case, cause the profile was created from scratch.

Changing the nozzle size through the mainbar (attached image) DOES work, but the sub menus do not.

[image]

nallath, 2019-08-01:
It's an issue with the menu bar. The main bar does work correctly

nallath, 2019-08-01:
I've fixed it in 3f8a65d9260fdc23c9eae8f876898851b9298ee2

closed 2019-08-01 by nallath

intuity-hans, 2019-08-15:
> Unfortunately not the case, cause the profile was created from scratch.
> 
> Changing the nozzle size through the mainbar (attached image) DOES work, but the sub menus do not.
> 
> [image]

Also, the maximum nozzle diameter is 1.0 mm. I have an E3D Volcano with 1.2mm - any ideas?

Ghostkeeper, 2019-08-16:
The only significant effect of the nozzle diameter is on the line width setting. So if you increase the line width by 20% you'll probably get a semi-decent result.

Alternatively, you can go into the Marketplace and find the "Printer Settings" plug-in from fieldOfView. It allows you to change an underlying nozzle size setting, which then effectively does the same thing: change your line width.

intuity-hans, 2019-08-16:
> The only significant effect of the nozzle diameter is on the line width setting. So if you increase the line width by 20% you'll probably get a semi-decent result.
>
> Alternatively, you can go into [...]

Thanks for the tip! My quick solution was just downgrading to 4.1 (which still has this setting). I have no idea why it was moved from the printer settings to the main screen.

I mean - you don't change nozzles THAT often. And then reducing it to a dropdown-menu without the option to enter custom values just doesn't make sense to me. 🤷‍♂️

Ghostkeeper, 2019-08-16:
4.2 also has this setting in the same way as 4.1. The drop-down is only additional. But in both versions the setting is hidden by default. Maybe you had the Printer Settings plug-in and just need to update it for 4.2.

The drop-down is to select specially-built profiles for that nozzle. It's not just the line width that changes when you change those. It's advisable to start from a profile that is closest to what you're actually using for best results (i.e. the 1.0mm nozzle).

intuity-hans, 2019-08-16:
What exactly do you mean by "hidden by default"? Can it be enabled in the settings somewhere?
I could not find the setting "nozzle diameter" under machine settings in 4.2. In 4.1 it is there (screenshot is 4.1).

[image]

Ghostkeeper, 2019-08-16:
The nozzle size field there is hidden for printers that have selectable nozzles because the nozzle profile you select would override it.

intuity-hans, 2019-08-16:
i don't really understand what you mean by "selectable nozzles". i would assume that people don't change nozzles that often. printheads maybe, but nozzles? 
maybe it is just me, but i have never changed the nozzle (diameter). maybe that's different for some people who change their ultimaker printcores frequently, but i guess for most of the people this doesn't really matter.
nevertheless, thanks for the detailed explanation!

Dulus0, 2019-08-27:
For CR10S PRO (and i assume maybe some other printers aswell), the nozzle diameter is not set in the "machine setting", but instead it is set in filament type menu, see pictures.
[image]

intuity-hans, 2019-08-29:
> For CR10S PRO (and i assume maybe some other printers aswell), the nozzle diameter is not set in the "machine setting", but instead it is set in filament type menu, see pictures.
> [image]

i am aware of that. my problem is that it is limited to 1.0mm. i have a volcano with 1.2mm and i cannot set this nozzle size

Silvatech, 2019-08-29:
one of the many many flaws of the creawsome addon sadly =(. Only workaround would be to add your printer as custom put in everything and then it will let you enter it. Hope they correct this soon. Interesting find. Just currios what are your printing with a 1.2mm?

intuity-hans, 2019-08-29:
actually this has nothing to do with the creawsome mod since i am not using it

i am printing design and mechanical prototypes at work to get a feel if my ideas work. it greatly increases productivity. as a measurement: a benchy takes about 1 hour on my ender with 0.4 V6 and 'good' settings, while it takes 10 minutes on my ender 3 with volcano and 1.4mm nozzle. layer height is about 0.4-0.5 mm. the results look terrible but are fine for trying out things.

Silvatech, 2019-08-29:
You dont understand the new cura they put in the code the creawsome plugin if your using an ender. So the only way not to use the plugin is to manually enter your printer as a custom printer. 
Yeah, even super large I was currios on the looks if they still be bad :P. Thanks for the info on how your using the large nozzle. 3dprinting is just amazin how it speeds up the design process =).

ArtemKobz, 2019-10-06:
Upgraded to 4.3 - still not able to change nozzle size :(

Ghostkeeper, 2019-10-07:
We don't have a profile for the Volcano 1.2mm nozzle. But you can still set it to use the line width of 1.2mm by changing the "Line Width" setting. It's probably best to choose the profile for the nozzle that comes the closest in its design and layout (a nozzle is more than just the diameter of the opening). I'd guess that you'd get the best result by starting from the 1.0mm profile and modifying the line width to 1.2mm, maybe up the temperature a bit.

max-hans, 2019-10-11:
I don‘t need a profile for the volcano nozzle. I was totally fine with the way it worked in an older version where you could just enter the specific nozzle size in the machine settings. Selecting a 1,0mm nozzle and then adjusting the line width seems hacky to me.

I just don‘t get the point why you would offer options for different nozzle over making it a simple number field. This just restricts users with smaller/bigger nozzles without a reason

jornada812, 2019-12-07:
4.4 Version, diameter does not change. Error is ignored

XAKEPEHOK, 2019-12-08:
> 4.4 Version, diameter does not change. Error is ignored

Same problem

DGordeenko, 2020-01-08:
Hello.
Could you try to use ',' (comma) instead of '.' (dot) ?
It help for me in same problem in Cura 4.4.1. Use comma when edit value, then close dialog box. When open it show correct value separated by dot.

phil3412, 2020-01-11:
> Hello.
> Could you try to use ',' (comma) instead of '.' (dot) ?
> It help for me in same problem in Cura 4.4.1. Use comma when edit value, then close dialog box. When open it show correct value separated by dot.

you got a good bypass for this issue
thanks

dpservis, 2020-02-02:
Not sure about Cura 4.4 (upgrade + Octoprint plugin crashes) but on 4.3 go to somrthing like C:\Program Files\Ultimaker Cura 4.3\resources\variants, switch to admin mode, add the variant you want and edit it.