Ultimaker/Cura #6087: [4.2.0] Machine Settings: Max Speed x,y,z change after update from 4.0.0

DivingDuck, 2019-07-30:
This is only a minor issue.
After updating from 4.0.0 to 4.2.0 my machine settings was changed for Maximum speed x, y and z from my values to 2997924580 mm/s. I am using a custom machine setup with own values and I modify these with plugin Printer Settings from fieldOfView

**Application version**
Cura 4.2.0

**Platform**
Windows 10 x64

**Expected results**
Update only if I was asked before so that I'm aware of.  :)
[image]

Ghostkeeper, 2019-07-30:
The maximum speed printer settings are now only used to limit the range that the user can set the print speeds. They are meant to mimic the firmware settings. Since Cura doesn't know what your firmware settings are, we opted to have no limit there any more, so that the user can set their print speed to, say, 501mm/s.

This is the result of the fix for https://github.com/Ultimaker/Cura/issues/5640. It's expected behaviour.

We will not ask all users for every change we make in the profiles. The new release will come with updated profiles.

closed 2019-07-30 by Ghostkeeper

DivingDuck, 2019-07-31:
@Ghostkeeper, 
Right, that is exactly what I had done: limiting the range of this setup. As fare as I understood there should be no update of a user defined setup. How can anyone be sure to have a valid and careful developed profile if I need to review after each update all variables in detail to check if it was "damaged" again?

As fare I understand the request of #5640, he was asking for not to limiting these values and I'm not really sure that the result to overwrite the value to a max value in general was the intension about the initial request.

Just for better understanding, I do not have anything against useful behaviors and for sure I don't want to blame Cura or their developers, but if an user sets a value in a profile, an update should't change any of those values without asking for an user approval (and ideally give an information what will change). That is exactly the kind of behavior a lot of people complain about software updates in the past. I was quite happy with this update as it was the first time I don't need to spend hours to setup my printers again (except this small issue). Please don't stop at the last mile to fail because of overwriting values without recognizing a user what had changed and what he need to review or take attention on. A little improvement with a lot more confidence for all users. This was all about this request.

... and keep on improving Cura. I like this version very much.

Best regards,
DD

Ghostkeeper, 2019-08-01:
I think you're misunderstanding the behaviour a bit here.

> Right, that is exactly what I had done: limiting the range of this setup. As fare as I understood there should be no update of a user defined setup. How can anyone be sure to have a valid and [...]

You hadn't limited the range of the set-up here. Cura's new profiles don't override your old overrides, but if you had no override (which was your case for the Maximum Speed settings) then the new defaults will be used. You left the Maximum Speed alone, never changed them.

The defaults for maximum acceleration also changed, but since you had overridden those changes, your overrides still stand.

> Just for better understanding, I do not have anything against useful behaviors and for sure I don't want to blame Cura or their developers, but if an user sets a value in a profile, an update [...]

We consider the new defaults to be better. If you had changed them away from the defaults then Cura will leave those changes alone, but you had left them on the defaults so Cura will update them to the new defaults. So that is basically doing what you are asking here. Only Cura will never ask the user about it but just always keep the user's changes.

DivingDuck, 2019-08-02:
Thanks for taking the time. You are right, there is maybe a misunderstanding from my side.
I was quite sure that I had modify this values but when I look to the picture it seems that these values wasn't change. Strange.
Anyway, with the update yesterday to 4.2.1 the values are as expected and a first test print in the evening looks good.
Thanks and best regards,
DD