Ultimaker/Cura #6088: [4.2] missing Octoprint feature

iyotee, 2019-07-30:
**Application version**
4.2 x64

**Platform**
Windows 10 x64

**Printer**
Creality Ender 3

**Reproduction steps**
Missing the octoprint feature, even after desinstall, reinstall et reupgrade the plugins from the marketplace

**Actual results**
Missing the instance of octoprint on printers settings.

**Expected results**
Should have Instance, and provide our API to connect from Cura to octoprint

fieldOfView, 2019-07-30:
A version of the OctoPrint Connection plugin compatible with Cura 4.2 is awaiting review. In the mean time you can download it here:
http://files.fieldofview.com/cura/OctoPrintPlugin-v6.0.0-2019-07-18T12_32_22Z.curapackage

Download the file and drop it in a running Cura application.

linhartr22, 2019-07-31:
I tried this too. In the add new network printer dialog it appears to only allow IP address entry. Even when I provided the IP of the PI, it says it couldn't connect.

fieldOfView, 2019-07-31:
> I tried this too. In the add new network printer dialog it appears to only allow IP address entry.

That could be a bug that needs to be reported here:
https://github.com/fieldofview/Cura-OctoPrintPlugin

nallath, 2019-07-31:
As a new version is under review, I'm closing this.

closed 2019-07-31 by nallath

maguilu, 2019-07-31:
FYI: I added the plugin in this thread and it is working fine. No changes to the configuration from 4.1 to 4.2..

fieldOfView, 2019-08-02:
See this post on what's going on:
https://community.ultimaker.com/topic/29068-psa-octoprint-connection-plugin-in-cura-42-and-421/