Ultimaker/Cura #6089: Octoprint not working on Cura 4.2

Mariovali, 2019-07-30:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

elloco054, 2019-07-30:
Yes, same issue here ...
Plug in installed and when I open parameters / printers / manage. There is no new button to connect to Octoprint. Just before with Cura 4.1, all was good ...
:(

iwittkau, 2019-07-30:
I have a fresh install of Cura 4.2.0 and the latest [Octoprint plugin](https://github.com/fieldOfView/Cura-OctoPrintPlugin) and it connected successfully.

**You need to update your Octoprint plugin to the latest available version! Otherwise the plugin itself reports it is not compatible with Cura 4.2.0 and will not be loaded.**

linhartr22, 2019-07-31:
I've uninstalled and reinstalled the Octoprint plugin. Is the one you link to newer than the one on the Marketplace?

dreward, 2019-07-31:
Same problem. Octoprint was working and is no longer. I have uninstalled octoprint plugin and reinstalled to no effect. Not sure if this is relevant but I checked the  plugins folder (which I have never looked in before) and octoprint plugin is not listed but I am guessing it is located elsewhere
All suggestions welcome

fieldOfView, 2019-07-31:
A version of the OctoPrint Connection plugin compatible with Cura 4.2 is awaiting review. In the mean time you can download it here:
http://files.fieldofview.com/cura/OctoPrintPlugin-v6.0.0-2019-07-18T12_32_22Z.curapackage

Download the file and drop it in a running Cura application.

nallath, 2019-07-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/6088

closed 2019-07-31 by nallath