Ultimaker/Cura #6090: [4.2] printing above build plate

400HPMustang, 2019-07-30:
[cura.log](https://github.com/Ultimaker/Cura/files/3447913/cura.log)
[CCR10S5_xyzCalibration_cube.zip](https://github.com/Ultimaker/Cura/files/3447915/CCR10S5_xyzCalibration_cube.zip)

**Application Version**
4.2.0

**Platform**
Win 10 x64

**Printer**
Left Side Creality CR-10 S5 

**Steps to Reproduce**
Load model into slicer
Slice file
Save to card
Insert card in printer
Print from card

**Actual Results**
Gantry raises above build plate, BLTouch attempts to take mesh, mesh fails because it is too high above the build surface. Nozzle moves to front left corner, purge line prints extrudes mid air, and then nozzle moves to center of build plate and extrudes filament in the air as if it were printing normally...except you know in the air.

**Expected results**
The BLTouch takes a mesh, contacting all points on the build plate, the purge line extrudes along the left side of the bed, and the print happens 

**Additional Information**
The same steps repeated in Cura 3.6 go perfectly. Not sure if it's relevant The printer in question is a CR-10 S5, Marlin 1.1.9, and a BLTouch 2.0

Ghostkeeper, 2019-07-30:
I tried to reproduce this. Mind that the 3MF file you supply is not a project file, so it doesn't have your settings. My reproduce steps were:
1. Add CR10s5 printer.
2. Load the calibration cube 3MF file.
3. Slice.
4. Save to file and inspect the g-code.

The resulting g-code starts with this:
```
;FLAVOR:Marlin
;TIME:2161
;Filament used: 1.635m
;Layer height: 0.2
;MINX:368.46
;MINY:368.46
;MINZ:0.2
;MAXX:411.539
;MAXY:411.54
;MAXZ:20
;Generated with Cura_SteamEngine 4.2.0
M140 S50
M105
M190 S50
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

M141 S28
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:100
;LAYER:0
M107
G0 F6000 X371.681 Y372.022 Z0.2
;TYPE:SKIRT
G1 F2700 E0
G1 F1200 X372.374 Y371.357 E0.03514
G1 X373.12 Y370.752 E0.07028
G1 X373.914 Y370.211 E0.10543
```

So it seems to set some acceleration/jerk/speed/flow settings, home with G28, then prime the nozzle 2mm above the build plate, and then start printing the skirt at Z0.2 (initial layer height). Seems normal. What g-code are you getting, and maybe you have a project file that contains any settings/profiles you altered?

Ghostkeeper, 2020-04-08:
Closing this due to lack of information.

closed 2020-04-08 by Ghostkeeper