Ultimaker/Cura #6091: [4.2] New version is "Unable to slice with the current settings.  The following settings have errors.

8BitEngineers, 2019-07-30:
Application version
4.2 x64

Platform
Windows 10 x64

Printer
Creality Ender 3

Reproduction steps
Use the "Super Quality 0.12mm" profile.  Click"generate supports".  Lower z hop to 5mm(if you do not it will list it as an issue)

Actual results
Error message that I talked about in the title pops up.

[image]

Expected results
Cura slices the model with no issue

Lunixes, 2019-07-30:
#6073

Brighouse, 2019-07-30:
Just downloaded the new version and I have the same unable to slice issue

8BitEngineers, 2019-07-30:
I guess Lunixes has narrowed it down to enabling supports.  I noticed this problem but I thought nothing of it, so I guess it checks out.  So we wait for another update then I guess???

Brighouse, 2019-07-30:
[image]

jewjudge90, 2019-07-30:
Windows 7 Ultimate 64, Cura 4.2 X64 here... Printer is defined as an Ender 3... I also get the same Cannot slice due to Z axis speed error when checking "enable supports" for the model... Even if I lower the Z axis speed to say 5mm I re-slice and get "unable to slice due to errors" then no listed errors so don't know what to change to fix it.

AtHeartEngineer, 2019-07-30:
Same issue, OSx 10.14, Cura 4.2, ender 3. It only happens when I enable supports, but it is broken now even when I turn it off. I recreated my printer profile, but I don't see the option to change my zhop speed in the profile (https://github.com/Ultimaker/Cura/issues/6073#issuecomment-515900794) as noted there.

chucktheviking, 2019-07-30:
Windows 10 Pro x64, Cura 4.2 x64 Here.
3D Printer: Ender 3 Pro
Issue: When selecting Generate Support, even on a default profile, I am unable to slice. I receive an error message. 
```
Unable to slice
Unable to slice with the current settings. The following settings have errors: Z Hop Speed
```
Upon changing the Z Hop Speed <6 the red indicator disappears, however when I try to slice I receive the following error message.
```
Unable to slice
Unable to slice with the current settings. The following settings have errors:
```

salgarba, 2019-07-30:
Same issue Windows 10, Ender 3 Pro.

jasminemoreira, 2019-07-30:
Same here, MacOS Mojave:

Unable to slice
Unable to slice with the current settings. The following settings have errors:

atmora77, 2019-07-30:
stop saying the same and tahe an other slicer like MatterControl 
the developper are really poor shit that deploy a think like that. A slicer that is unable to print hahaha

smartavionics, 2019-07-30:
People, a workaround for this issue is to install the printer settings plugin from the marketplace and then increase the max z speed value to 10 (or more). It should then be able to slice without barfing.

atmora77, 2019-07-30:
no car eabout fucking workaround but about solution

smartavionics, 2019-07-30:
Sorry, asshole, I don't have a solution, just the workaround. If you don't care to use it, wait for the devs to provide a solution. That could take anything from 1 day to 2 months.

atmora77, 2019-07-30:
or just stop use this shit

pfritzd, 2019-07-30:
Hey smartavionics

I installed the marketplace plugin and made the changes to the max z speed value (kept it at 10) .. still getting the same issue

smartavionics, 2019-07-30:
> I installed the marketplace plugin and made the changes to the max z speed value (kept it at 10) .. still getting the same issue

And what is the z hop speed in the Speed section set to?

pfritzd, 2019-07-30:
oops ... forgot to include that the z hop speed option no longer shows up as an option to change anything (I've enabled the Show all settings) 

[image]

smartavionics, 2019-07-30:
That's maybe because you don't have z hop enabled?

smartavionics, 2019-07-30:
[image]

pfritzd, 2019-07-30:
hmm ... 

It shows that I have it enabled ... 

[image]

smartavionics, 2019-07-30:
No, those black circled i's mean that those settings are invisible because some other setting (in this case, Z hop when retracted) has disabled them.

pfritzd, 2019-07-30:
I'll blow away my current install and start over and make sure I didn't miss anything.  

Just to make sure I am grabbing the correct plugin ... it's this one
[image]
... correct

smartavionics, 2019-07-30:
Correct.

atmora77, 2019-07-30:
I don't see the printer settings on the sidebar everything is broken holy shit

pfritzd, 2019-07-30:
Cool thanks ... I'll report back shortly

MYepiz, 2019-07-30:
Hello, Ive tried installing the plugin, I restarted Cura, activated z hop, z hop speed set to 10, and then tried setting it to 5, and a bunch of combinations but the same empty error message appears. Turning these settings on and off do nothing for me neither, nor going back to using the CR10S profiles instead of the CR10SPro. WTF is up with this update

peejayw, 2019-07-30:
@smartavionics That worked for me, thanks.

MYepiz, 2019-07-30:
Ah, my bad, I was changing z hop speed values, instead of Z Speed. It slices now as said by @smartavionics Still... this update...

smartavionics, 2019-07-30:
> Still... this update...

Come on, you should be used to it by now. With every Cura release it's a few steps forward and then a few steps back again.

dannyboy76, 2019-07-30:
@smartavionics i downloaded the printer settings plugin and the enabled z hop but im still getting the error message no matter which speed I set it to

smartavionics, 2019-07-30:
So you have both max z speed and the z hop speed the same and >= 10 and it still doesn't slice?

dannyboy76, 2019-07-30:
@smartavionics nevermind. i did the same thing as MYepiz. It slices just fine now. Thank you for your help. Much appreciated

pfritzd, 2019-07-30:
@smartavionics ... I am up and running ... thank you for the help

smartavionics, 2019-07-30:
I'm leaving this thread now. Good luck!

jasminemoreira, 2019-07-30:
Setting the speeds worked for me too. Thanks a lot.

JackS60, 2019-07-31:
using the Ender 3 [Windows 10]  I had the same problem of not being able to slice with supports.  I went to the setting visibility and at the very end is the category  PRINTER SETTINGS I highlighted them all and there is the MAXIMUM SPEED Z that was set at 5 ... I increased it to 10 and now I can slice with supports , no problem... 
[image]
[image]

AtHeartEngineer, 2019-07-31:
> People, a workaround for this issue is to install the printer settings plugin from the marketplace and then increase the max z speed value to 10 (or more). It should then be able to slice without barfing.

This worked for me

GMaidana88, 2019-07-31:
Same Issue when activating supports on Ender 3 with Cura 4.2.0. 
Enabling Printer Settings and setting Max Speed Z to 10 worked for me too.

nubnubbud, 2019-07-31:
I attempted the workaround on 64 bit windows 10, and it allowed me to slice, but supports did not appear in the viewport, and the printhead did not trace any out. I'm not sure why the workaround did not work in my case.

nallath, 2019-07-31:
So there are two issues going on; The error value of the z-hop is  such that it's in an error state by default and there is an issue with the error checker (which has been there for a while but it only now surfaced as an actual issue)
Don't worry, we already have a fix for them, we're testing it as we speak.

diegopradogesto, 2019-07-31:
We are working on it right now. We found the problem and we plan to release a hotfix ASAP. Keep you informed.

Thanks for all the feedback.

Green6166, 2019-07-31:
SMARTAVIONICS YOU LIFE SAVER

trouch, 2019-07-31:
Instead of increasing `machine_max_feedrate_z`, I recommend to lower `speed_z_hop` to 5 to avoid losing steps.
See #6103

MahmoudEzzatAbdul-Rahman, 2019-08-06:
> People, a workaround for this issue is to install the printer settings plugin from the marketplace and then increase the max z speed value to 10 (or more). It should then be able to slice without barfing.

I installed the plugin, still can't get it to work

Ghostkeeper, 2019-09-03:
Fixed in 4.2.1.

closed 2019-09-03 by Ghostkeeper