Ultimaker/Cura #6092: Remove line width multiplier on creality machines

Liger0, 2019-07-30:
**Is your feature request related to a problem? Please describe.**
Yes. On the new Cura 4.2 the line width on the creality machines the line width is not anymore the same as the nozzle size but bigger by a fair amount. Much people ask for help on chats and so on since the creality mod was an unofficial addons and now that it's implemented the number will just increase. With a line width so larger than the nozzle size people report underextruded-alike prints because of it along with disabled extra restart and unused relative extrusion, which I can confirm.  Instead part of people prefer to use a line width *smaller* than nozzle size, which increases lines bond and quality of the print. 
So I don't see why using a line width larger than nozzle sizes. Also, because of it, people that don't want to have a LW larger than the nozzle, have to create a profile for each nozzle.
It also becomes a good mess when someone has to use the custom definition edit plugin to set a nozzle size like 0.45, 0.35 etc for obvious reasons with the newly nozzle-size on the material tab and that non-sense line multiplier.

**Describe the solution you'd like**
Let anyone decide if they want to use a line width equal to nozzle size, bigger or smaller. Go back and set as default a line width equal to nozzle size as a neutral parameter so anyone can experiment.

**Describe alternatives you've considered**
Set a tab that you can choose to set a negative multiplier, positive multiplier, or 100% to automatically calculate the line width based on the profile nozzle size.

**Affected users and/or printers**
Creality printers.

Ghostkeeper, 2019-07-30:
> Let anyone decide if they want to use a line width equal to nozzle size, bigger or smaller.

We do. You can set a value for the line width in the print settings.

We added these profiles because it seemed that the community as a whole found the CreawsomeMod profiles to work out much better than the ones we had before. These are the new defaults, because apparently they do work out better.

Why do they work out better? I don't know because I don't have any Creality printers, but I do have some theories.
* Faster print.
* Improved layer adhesion.
* Improved bed adhesion.
* Fits better through the nozzle after a month or two of wear.
* Slower annealing.
* Better dimensional accuracy after sagging.

Line width is one of those settings that influences basically everything. So as a drop-down for line width we'd need someone to create profiles for the different line widths that adjust the other settings too, similar to the nozzle size. For your suggestion that'd mean 213 new profiles for Creality printers. Many of those will be the same, but it's a momentous task to test all of those.

Liger0, 2019-07-30:
You can, but writing it everytime and creating a profile for each just because line width is wrong is a waste of time. While in the past you changed the nozzle size and the line width was calculated after it.
As there are people that didn't ever want to touch the profile and welcomed that mod, there are people that had only horrible results because of it, so of course those people, even being as much as the ones that liked it, didn't spend a work on the mod, and just requested help to remove it and set good values for the creality machines.
Using a line width multiplier just doesn't make a sense to me and only caused troubles to a lot of people for no real advantage. If people want to increase it, they can manually do it, but that's "fine tuning", it really isn't a starting point for everyone that wants a good base profile that works, if they want to manually decrease it, they can also do it, but IMO the profile should start from a base value the nozzle size which will work out of the box.

Liger0, 2019-07-30:
Also I am unsure why a complete profile was rewritten without the principle optimizations, so what does it optimize exactly?
Relative extrusion is disabled (???)
There is the perimeter overlap enabled (always ruined corners and surfaces to me and others)
Printing the thin walls is disabled (???? It's very useful for most models)
Minimum extrusion width windows is set to 10 (even reported in orange by Cura, why 10??)
Cubic infill. (Why not gyroid????)
Support interface enabled (Why??? It won't give good results with all material and without a precise calibration (which who uses that profile won't have anyway), a proper support without interface layer will work better)
Build plate adhesion to None... self explanhatory... at least use a skirt if it's supposed to be an out-of -the-box profile
No coasting
No extra restart
Steps per millimeter are not even defined.. they should be 80/80/400/100, but instead they are left at 50/50/50/1600. I see the printer settings are changed so I don't get why not defining them at this point as someone reported they will be used in a future release of cura.

nubnubbud, 2019-07-31:
> We do. You can set a value for the line width in the print settings.
>
> We added these profiles because it seemed that the community as a whole found the CreawsomeMod profiles to work out much better [...]

I think we really should have a cr-10(stock) and cr-10(creawsome preset) as two separate "machines". Most notably, the CR-10 is such a big deal because it's so modifiable and open source.

...and this also means the new preset won't work with my cr-10s, because I have an aftermarket hotend. I've tried creawesome before, but it was next to unusable for me, as nice as it promised to be, because my firmware, hotend, bowden tube, extruder, motherboard, and fans were all retrofitted with better parts until it gave me exactly what was in cura.

to me, it seems the creawesome mod (except maybe the layer heights- I've gotten pristine prints at .1mm so I don't know what the big deal is) is a set of presets made to overcome common issues with the printer, but that seems like a band-aid solution.

nallath, 2019-07-31:
> means the new preset won't work with my cr-10s, because I have an aftermarket hotend.
Machine definitions should work with the default setup to the machine. Not any after market upgrades. I understand that it's annoying for you, but you don't really have a CR-10 at the moment. You have a machine that started out as one but is now so heavily modified that it's something else entirely. 

As for the slicing issue, I just pushed a fix for that to the branch, it's being tested as we speak.

nubnubbud, 2019-07-31:
> As for the slicing issue, I just pushed a fix for that to the branch, it's being tested as we speak.

to be fair, the only things that weren't just new parts from the same company were the hotend and fans, but you have a fairer point. Luckily, I calibrated my printer to Cura's default, regardless of machine way back when, so I can just make a custom machine and remake my presets.

nallath, 2019-07-31:
I'd much rather have that creality put in some effort into their own profiles. It's doable for them to set it up in such a way to take these modifications into account. It would take some work on their side, but I think they sell more than enough printers to make it worth their time & effort.

nubnubbud, 2019-07-31:
> I'd much rather have that creality put in some effort into their own profiles. It's doable for them to set it up in such a way to take these modifications into account. It would take some work on their side, but I think they sell more than enough printers to make it worth their time & effort.

I've done some research into them, and it's very uncommon for creality to update anything about their systems once they're made. They're still putting out an old (1-2 full number versions back) version of cure in their cards, and only updated the motherboard when it turned out an issue stopped it from keeping a stable temperature past 218c.

I think it's less about how much they sell, and more avoiding bureacracy and technical issues by just selling a standard printer, with one firmware, and one slicer, all designed to work together. They do put the modifations that are cheap onto their currently produced printers, like bed leveling knobs and filament sensors, but their primary goal is to keep the prints as high quality as possible, with the cheapest parts possible.

as nice as it is, that was 3 years ago and cura left them in the dust, as they refused to update. as well, they're a chinese company, and they don't seem to have an amazing translation service. couple that with china's draconian internet censorship, and they might not even know they can contribute.

nallath, 2019-07-31:
I don't think that the profile we originally had for creality was even made by them. I also don't think that they put a whole lot of effort into optimizing those profiles. It's a fair investment to do so, but you do get a lot better print quality if you do (For instance; Ultimaker has 10 ish people working on just materials & processing alone). But I guess you get what you pay for.

JohnEdwa, 2019-07-31:
+1 for leaving the old Creality profiles in and creating separate Creawesome machines - it would also allow the creawesome dev(s?) to more easily tweak their settings in the future without automatically affecting *all* Creality Cura users, but only those that want the "awesome" tweaks.

Right now, all my profiles are incompatible, and even transferring them manually messed them up because they relied on the old default settings, a lot of which were changed in the background with absolutely no indication, such as enabling support interface.

Liger0, 2019-07-31:
Especially considering they changed for the worse.

nallath, 2019-07-31:
Sorry, but we're not going to change back. This is feedback that should have happend before the final release. It's too little, too late. 

Changing back at this point would cost a huge amount of work and probably incur the same kind of issues as we're having now (and would incur them _again_). You're unfortunately part of a small minority (as far as I can see) that feels this way.

JohnEdwa, 2019-07-31:
I am of the small minority that expects a software version update from 4.1 to 4.2 to not to mess up default profiles and settings in the background with no prior warning nor the possibility of a rollback.  But I see your point, what is done is done and it'll blow over eventually. 
And as far as defaults go, they are most likely better than what Cura had before, just done in a stupid way.

nallath, 2019-07-31:
Well, it does give a warning. It's in the change log and the blog post. One could argue if that's enough or not of course, but it's not like it's a complete suprise.

We considered if this should be merged as it is (the option you suggested was on the table) but we anticipated that a lot of people would simply continue using the old (and pretty shitty) default creality profiles. 

All that being said; I'm hoping to get you on board to actually improve / test these profiles. This is an open source project after all. Just because one person submitted profiles, doesn't mean that they know the best settings or are the only one that get to change it.

torsoreaper, 2019-07-31:
> You're unfortunately part of a small minority (as far as I can see) that feels this way.

Please show us your data for this.  Out of all the Cura / Creality installs, how many people installed Creawesome? How many uninstalled Creawesome?  How do you know that it wasn't just a vocal minority that liked Creawesome?  Again, you make a sweeping generalization to justify adding Creawesome but you have no data to prove your assertion.

Liger0, 2019-07-31:
I am pretty much sure that the most part that installed the creawsome mod uninstalled it because of the non-optimized settings, and the only ones using it were people that never even tried to modify a profile.
From groups and chats it is visible the most part of the community didn't like that mod.

Much settings that are dumb are the ones I posted early in this ticket. It seems that who created the stock profiles doesn't even know what much settings to and how they should be to optimize print quality and give a decent result.

JohnEdwa, 2019-07-31:
I started [a thread on the Ender 3 subreddit](https://www.reddit.com/r/ender3/comments/ck9agf/help_making_the_new_cura_42_creawesome_profiles/) with the idea of gathering some of the suggestions for changes to the current default profiles, so if anyone here happens to use reddit and has ideas on what settings should be tweaked, pop over to there.

FITPrinting, 2019-08-01:
> Sorry, but we're not going to change back. This is feedback that should have happend before the final release. It's too little, too late.
>
> Changing back at this point would cost a huge amount of [...]

So the response translates to "We're not changing it because people that never used it didn't complain about it before we rolled it out and it screwed their profiles"?

Awesome. Great job.

torsoreaper, 2019-08-01:
The irony is that creawesome is generally crap.  Aside from breaking profile support, the guy's settings aren't even that good, he just turned down the speed and set it to print infill before walls.  A bunch of people who didn't understand Cura installed it and said "wow my prints got better".  Now the people who actually know how to use Cura have to suffer.  

Seems reasonable /s

nallath, 2019-08-01:
> Again, you make a sweeping generalization to justify adding Creawesome but you have no data to prove your assertion
You're right. We didn't hold a survey to check this. We probably won't ever do such a thing because it would probably take more time than I can spend on this. 
So there is only one thing that I can listen to; People who contact us (either through github or other means) to ask or request things. Since we only heard people in favor of the new creality profiles and no-one against it, we decided to go ahead with them. 
I'm not trying to prove anything. I also don't need to as far as I'm concerned. I'm simply telling you what the logic was. It's a reason why we did it, not an excuse.

>Especially seeing how support requests related to horrible print quality increased over the community after this messy mod started being a trend.
I've actually mostly seen improved prints on the groups that I'm active on. But maybe we're in different groups.

>Those are lazy people that refuse to learn how to use a software and they shouldn't force a software change because of them, you'll just follow their expectations and they will never learn to properly the software because of it, IMO)
This sounds very elitist. Most people are lazy and that's fine. The people that don't really know how things work should also be able to get good quality prints and should also be facilitated.

> So the response translates to "We're not changing it because people that never used it didn't complain about it before we rolled it out and it screwed their profiles"?
Did i say that? I don't think I did. I understand that you're angry, but there is no need to put words in my mouth. Those profiles were tested. You might not agree with them. That's fine. But complaining about changes in an open source project when you did _absolutely nothing_ to help when the chance was there, is frankly a bit insulting. I am under no obligation to ensure that your workflow stays as you want it. I will try to do so, but in order to do this i need feedback _at a time where i can do something with it_. If you care so much about what happens, why do you only contribute something when the damage is already done? I understand that this is frustrating, but do you also see that this is fucking frustrating for us as well? This isn't mallice on our side.

> Now the people who actually know how to use Cura have to suffer.
Or, you could have done the thing that he did and _actually share improvements over the defaults_. It's really frustrating what is happening now. Someone improved the rubbish quality of the default profiles and now people complain that their settings are better. Really? That's why you are mad? 

> I started a thread on the Ender 3 subreddit  with the idea of gathering some of the suggestions for changes to the  current default profiles, so if anyone here happens to use reddit and  has ideas on what settings should be tweaked, pop over to there.
Thank you! That's exactly what should be done. 

**TL:DR** I understand the frustration, but lashing out now is not helping. You want a better Cura? Then help out make it better. We can't do it if we only get responses after the fact.

torsoreaper, 2019-08-01:
> **TL:DR** I understand the frustration, but lashing out now is not helping. You want a better Cura? Then help out make it better. We can't do it if we only get responses after the fact.

Honestly, we should all take a step back and laugh about this.  It's a bit like the first chapter of A Hitchhiker's Guide to the Galaxy.  I'm saying people should just learn how to use Cura and you're saying people should be on Github reviewing pull requests to see what's coming down the pike.  

It's kind of like when Arthur's house is going to get bulldozed and the workers say "hey the plans were down at city hall for you to review" and he says "I didn't even know they were there for review!  Then the aliens come and say "we're bulldozing earth, if you don't like it, the plans were down at intergalactic city hall, you should have said something".  To be fair, I'm saying "hey, better profile information was available at city hall" and you're saying "hey our github pull requests were available in intergalactic city hall".

Ghostkeeper, 2019-08-01:
> We considered if this should be merged as it is (the option you suggested was on the table) but we anticipated that a lot of people would simply continue using the old (and pretty shitty) default creality profiles.

Actually the consideration was taken mostly because it'd be very confusing for new users to see two Creality printer types, one of which is a CreawsomeMod version that they don't have. Not everyone knows what "CreawsomeMod" is if they were not active in the right communities.

> Please show us your data for this. Out of all the Cura / Creality installs, how many people installed Creawesome? How many uninstalled Creawesome? How do you know that it wasn't just a vocal [...]

We know more than 10.000 people have sent us a bug report about it not working in Cura 4.1. We can also see a significant portion of our 4.0 users sending slice data with Creawsome settings. Of the plain Creality users in 4.0 we don't know how many of those are using Creality because they didn't know about Creawsome and how many tried Creawsome but thought the previous profiles were better. We also looked at the responses of threads on Reddit about it and in Trouch's repository, both of which were generally positive.

Mind that we also have selected community testers now that test with Creality printers and the new profiles were considered a great improvement there, apart from that the printing time also increased.

> Honestly, we should all take a step back and laugh about this. It's a bit like the first chapter of A Hitchhiker's Guide to the Galaxy. I'm saying people should just learn how to use Cura and you're saying people should be on Github reviewing pull requests to see what's coming down the pike.

I don't think everyone should be required to try the beta or keep up with our pull requests from contributors. But we have to rely on the people that *do* to provide us with feedback. Since I don't have the resources to buy every printer out there, we can only rely on what people tell us about the profile changes.

FITPrinting, 2019-08-01:
> Did i say that? I don't think I did. I understand that you're angry, but there is no need to put words in my mouth. Those profiles were tested. You might not agree with them. That's fine. But [...]

>Sorry, but we're not going to change back. **This is feedback that should have happend before the final release. It's too little, too late.**

You stated that people that aren't using the "mod" should have come to github, found that you were trying to push something a small percentage of, admittedly vocal, users found improvements on that would overwrite the settings of those that weren't using it, and complained beforehand about something they didn't know was going to happen.

It would be extremely easy to have a legacy Creality profile that didn't force the "mod" upon those that know how to dial in a printer. I get that printing is spreading to more and more casual users and Cura's interface is moving in the same simplified direction, but these acts of alienating the skilled for to make it easier for the casual user is the wrong way to do it.

Benjelum, 2019-08-01:
it's ok for users to be lazy, but the devs expect these same users to read the change log and blog, insane. Cura has taken it's first major step away from being the obvious choice for free users. "why didn't anyone tell us" is both tone-deaf an unprofessional, as is "it's open source do it yourself" more-so, are you an expert providing a software, or not?

nallath, 2019-08-02:
> it's ok for users to be lazy, but the devs expect these same users to read the change log and blog, insane. Cura has taken it's first major step away from being the obvious choice for free users. [...]

Perhaps. On the other hand, it might also finally get a point across that it's impossible to leave everything with us. It's already a free bonus on our side that we even spend time on this. Ultimaker is getting nothing from us doing that. No Creality user pays us and we get precious little pull requests from that "vast group of expert users". So yeah, my response might be a bit tone-deaf, but the way I see it, it's about as tone-deaf as the responses. So, in this case, we chose to facilitate the largest group of users that we knew about. By saying that we need feedback during the beta is not an excuse or an argument why any of you should shut up, but an explanation as to why we chose to do it like this. 

You can rail and lash out how we are unprofessional, tone-deaf or any of the very colorful hyperboles that I've seen in the past few days. Apart from ensuring that you feel good about yourself, does that help? Does anyone have the idea that it will make anyone change the situation? I very much doubt it.

We've got a good saying about this in dutch: "The best captains are those ashore". So instead of quoting that expert knowledge that you have, you could ya know, contribute some of that back. Then we wouldn't even need to have this discussion, because the profiles would be amazing.

Ghostkeeper, 2019-08-02:
> You stated that people that aren't using the "mod" should have come to github, found that you were trying to push something a small percentage of, admittedly vocal, users found improvements on that [...]

Like I said, we have data that a significant portion of Creality users installed this mod.

Nobody is saying here that everyone should be reading all the updates. We're only saying that we can only rely on feedback from users to know which changes to profiles are good and which aren't. We get good feedback, then we'll change it. We get bad feedback, then we won't. That's all there is to it.

Benjelum, 2019-08-02:
> "why didn't anyone tell us" is both tone-deaf an unprofessional, as is "it's open source do it yourself" 
  
> "The best captains are those ashore". So instead of quoting that expert knowledge that you have, you could ya know, contribute some of that back
  
You seem to be missing my point, Do you need experts to clarify that destroying user data is bad in 100% of cases, and the post-emptive reasoning of 'why didn't YOU read our release notes, blog, changelog, and source?' Is obviously CYA.
  
Here is the expert advice that doesnt seem to occur to you, that users must give you, the expert sea captain: " **dont destroy users data, all updates should be purely additive, and always support legacy functionality** "
  
No go ahead and explain another idiom to me about how users should magically know you'll choose to vaporize thier user data without regard or consideration in a minor version update. You are not the only free slicer, just the only one making these mistakes with userdata. Thanks for destroying my previously pleasant cura memories of turning tweaking and troubleshooting, with a big mod you forced users to opt into and blew away all these hard won settings **without so much as a user prompt?**  *did you forget how to add userprompts?* Or did you want to walk through your sea captain logic on that for us lazy useless landlubbers
  
Thanks for the slicer I guess, I want to appreciate *your* work, but it's hard when your carelessness just destroyed all *my* work within your product.

nallath, 2019-08-02:
Blew them away? Cura doesn't delete any data in an upgrade. The old data is there ready for use when you start up Cura 4.1. You'd be right if Cura would actually destructively upgrade, but this is the exact reason why we ensured that Cura doesn't do that.

My argument regarding the sea captains was aimed at the claims that the profiles in and out on itself were rubish, which is also the one the argument regarding only contributing when it directly affects you (yet still expecting Ultimaker to contribute in all cases). So please don't strawman my arguments into what they are not. 

If you want to influence what happens, contribute. Otherwise, you will simply have to accept the decisions made by those that do (for better or for worse).

FITPrinting, 2019-08-02:
> No Creality user pays us...

You are correct. No _Creality user_ pays **Ultimaker** for Cura. Nor do 101Hero, CdGator, 3DMaker, ABAX, Anet, Anycubic... (get the point yet) ...users. Now why does Ultimaker _pay you_ to "upgrade" and make changes to Cura for all of these printers with no direct payment? The goodness of their hearts? Nah. 

It's because you're in the marketing department. That little "Ultimaker" in front of Cura on every machine that uses it influences the people looking at that screen. It's about brand awareness and having the software out there in front of as many sets of eyes as possible is why you work on it. Do you think Makerbot sells printers to Thingiverse users? 

How many of you do you think Ultimaker would need to employ to maintain and upgrade the software if it was only for Ultimaker printers? 

So because you're in the **marketing department** and because that little Ultimaker name is there you get _indirectly_ paid by Creality users in that Ultimaker thinks it's important enough to employ you to do so. Now, if a marketing effort does not work, if there isn't a return on the investment, if say sweeping changes to please the minority were put into place and caused a mass exodus of the majority how long do you think it would be before accounting starts asking why the _marketing department_ isn't doing their job? 

> So, in this case, we chose to facilitate the largest group of users that we knew about.

Ah yes, the government lobbying version of software updates! "These special interest groups for the oil industry are the only ones that came to my office to tell me that everything is fine, what do you mean the rest of the world is going to drown while I sit in my castle on high?"

> So instead of quoting that expert knowledge that you have, you could ya know, contribute some of that back.

What's the incentive to do so when it'll just be overwritten again when low hanging amateurs start yelling about how X fixed everything they screwed up?

> Like I said, we have data that a significant portion of Creality users installed this mod.

No, you have data that a significant portion of Creality users that are amateur enough to send you data installed it. How many uninstalled it? How many used it without ever changing their settings back? 

> We get good feedback, then we'll change it. We get bad feedback, then we won't.

So you're getting good feedback now that you made bad decisions. What's the verdict?

JohnEdwa, 2019-08-02:
>No, you have data that a significant portion of Creality users that are amateur enough to send you data installed it.

Hmm, good point. A lot of more advanced users will immediately disable sending their data to ultimaker, me included. So the info you get is heavily biased towards the kinds of users that do send it.

nallath, 2019-08-02:
> You are correct. No _Creality user_ pays **Ultimaker** for Cura. Nor do 101Hero, CdGator, 3DMaker, ABAX, Anet, Anycubic... (get the point yet) ...users. Now why does Ultimaker _pay you_ to "upgrade" [...]

That's what is ensuring that we get to do this at all. It's not some binary scale of course. There is putting in some token effort and fully supporting it. As it stands now, that's pretty much how it is. We put in some effort for third party printers, but it's only a little bit. Not as much as I would like and definitely not enough to make it really good. 
 
> How many of you do you think Ultimaker would need to employ to maintain and upgrade the software if it was only for Ultimaker printers?

I think it will be somewhat more, but how much do you think it would cost to actually test all those printers? Yeah, a whole lot more than that befit we are getting. And let's be honest here, before this moment we didn't get a whole lot of feedback to changes we made. So apart from the negative effect of riling up a lot of people, I do see some silver lining; It shows that we can't make this a good slicer for everyone if it's just us doing 95-99% of the development. 

> So because you're in the **marketing department** and because that little Ultimaker name is there you get _indirectly_ paid by Creality users in that Ultimaker thinks it's important enough to employ [...]

Well, as you already stated, we should use more data to make our decisions, so we will have to wait. There are now people claiming that either option in the decision is the wrong/right one.  I'm a bit doubtful as to if this exodus is happening (or will happen). Time will tell that I guess. 

If marketing was the prime reason for us to do it, this change shouldn't come as a surprise. If it's about brand awareness, it's usually about ensuring that it works for the masses. People that buy 300$ 3D printers have long stopped being the target audience for Ultimaker, expert or not.

> Ah yes, the government lobbying version of software updates! "These special interest groups for the oil industry are the only ones that came to my office to tell me that everything is fine, what do you mean the rest of the world is going to drown while I sit in my castle on high?"

Yup. It pains me that it's like that. But I don't have an alternative. I fully realize that that is a weak argument, but it's all I can do. But hey, the office door is open. Feel free to give feedback and make the pull requests. 
 
> > So instead of quoting that expert knowledge that you have, you could ya know, contribute some of that back.
> 
> What's the incentive to do so when it'll just be overwritten again when low hanging amateurs start yelling about how X fixed everything they screwed up?

Well, we're having this discussion now are we not? But yeah, if you don't trust or believe it when any of the devs say that we want your proposed changes, I guess there is little we can do to convince you. My suggestion would be to start with some small things. If you see that we value and implement them, both sides can show that they can be trusted with it. 

If you do decide that we can at least be (somewhat) trusted by that, the main reason for contributing would be influence, recognition, and bugfixes. If you know more about how the software works, it's easier to write high-quality issue reports. The better those are, the easier (and thus, more likely) it is that those get fixed. On a personal level, I also contribute to Cura in my spare time and in all honesty, I'm much more likely to help people out that also contribute something back. 

> > Like I said, we have data that a significant portion of Creality users installed this mod.
> 
> No, you have data that a significant portion of Creality users that are amateur enough to send you data installed it. How many uninstalled it? How many used it without ever changing their settings back?

Yup. But again, you have to use the data that you have. No information source is perfect and the numbers were compelling enough to make the change. I'm not allowed to give you exact data, but I can say that we get well over a million slices a _week_ this way. A few people will disable it, sure. But the data pool is large enough to draw conclusions from I'd say. 

> > We get good feedback, then we'll change it. We get bad feedback, then we won't.
> 
> So you're getting good feedback now that you made bad decisions. What's the verdict?

We get good feedback that people disagree with the decision. But changing the way creality handles the setting back is just going to double the problem (as it will result in the same breaking of profiles in 4.3 again). As that's the main thing that people are complaining about now, I don't think that doing that is the way forward (Two wrongs don't make a right of course). 

Other than that, the best that can be done is to ensure that the added profiles are the best they can be. In that regard, I see it as the silver lining here, because now we're at least having a discussion on how these profiles can be improved (compared to the total silence we got from the majority of expert users). 

This is also putting more priority on our plans to have more nightly & pre-release candidates so issues like this can be caught earlier.

torsoreaper, 2019-08-02:
This has obviously gotten pretty heated and people are very upset.  I have taken the liberty of creating a "regular" ender 3 profile so people can add an "ender 3 classic" that doesn't have creawesome.

https://github.com/torsoreaper/Ender3_classic

Basically all I did was grab the Ender 3 resources from 4.1 and make them compatible with 4.2.  I don't know if there can be some compromise in the future where Cura 4.3 simply has "Creality3D" and "Creality3D Legacy" or "Creality3D Classic" something like that but it seems like a very easy, low effort, way to fix things.

Creating a 4.1 printer profile in 4.2 took me all of 15 minutes, so hopefully that wouldn't really impact the development cycle of Cura in general. 

Anyways, I hope this helps some people and if you have feedback on my classic profile, feel free to reach out to me on that repo.

oofiksoo, 2019-08-22:
Ok, I will not be purchasing an ultimaker printer, and I will no longer be using the ultimaker software. The consistent response of "we made this change, your not happy about it, you should have done something about it, we're open source" is bogus. Not everyone in the world has time to help ultimaker make money and build as a company. Even if you offer an open source product, it is possible that your small group of people made a bad decision. The Creawsome functionality should not have been introduced, and your pride will simply not allow you to admit that this was a rush job. You guys saw a few happy campers within the same cohort, and though this functionality was great. This is a direct failure by the "open source community" working on Cura, and you are being absolutely rude, condescending, and off putting. You are not special because you are offering an open source product, and the users of the product should not be made to feel like we owe you something, we should be happy with what you give us, and we should shut up if were not contributing to your growth. 

As I said before, I will not purchase an ultimaker printer, And I will not be using the Utilemaker Cura platform. I can deal with issues, bugs, and things that need my attention, But I will not support a company so off putting, and whom responds with the remakrs such as yours. It is obvious that this is an issue, it reaches many of your users, and you refuse to take any other position than for us to contribute or shut up.

The functionality is NOT correct, It does not improve the software, and IS AN ISSUE. Try to sweep it away any way you want, but this is a MAJOR FAILURE!

nubnubbud, 2019-08-23:
> Ok, I will not be purchasing an ultimaker printer, and I will no longer be using the ultimaker software. The consistent response of "we made this change, your not happy about it, you should have [...]

please do not blame ultimaker for this, regardless of the end verdict or what happened in testing, Cura is only implicated as their slicer of choice, and while they do own the rights to it, it is an open project, and people like you are free to edit the source code. The creators of ultimaker machines don't, as far as I can tell, have some significant part of their team working on it, compared to people like me who hop in and add a couple features they specifically need.

nallath, 2019-08-23:
> Even if you offer an open source product, it is possible that your small group of people made a bad decision. 

Correct. That's why we ask people to test the things that we can not test (either due to a lack of resources or a lack of priority). This also means that some things are not at an equal quality level (eg; The Ultimaker profiles are of much better quality as a lot of money / time is invested in them).

We also didn't claim that it was a good / bad decision, just that there are a lot of reasons to not switch it back (as it would create even more issues)

> The Creawsome functionality should not have been introduced, and your pride will simply not allow you to admit that this was a rush job. You guys saw a few happy campers within the same cohort, and [...]

I won't admit it being a rush job, because I don't think it was. I do admit that it wasn't tested as much as changes made to Ultimaker printers (but then again, we never made a secret where our priorities lie). As to it being a failure of the OS community, I think that's the case. That's also why i hope that this shows that the quality of these changes is dependant on testing / feedback / etc that we get (even if right now we have path dependency)

> You are being absolutely rude, condescending, and off putting. 

This was not my intention to do so, so if that was the case, I'd like to apologize for that. I know it's not an excuse, but the entire discussion got heated on both sides (and got carried away). With that said, I'd also like to ask to re-read your own message, as it's giving me a lot of the vibes that my messages seem to instill with you as well (especially the copying it in multiple threads does increase the passive aggressive vibe) 

> You are not special because you are offering an open source product, and the users of the product should not be made to feel like we owe you something, we should be happy with what you give us, and we should shut up if were not contributing to your growth.

You don't have to shut up. But do note that the other side is also true. Just because it is open source also doesn't mean that I have to help you either. It's a two-way street. You don't have to be thankful (but it would be nice if you are) and I don't have to help you (but it would be nice if I did).

>  It is obvious that this is an issue, it reaches many of your users, and you refuse to take any other position than for us to contribute or shut up.

Again, we're a company (of which I don't speak on behalf of). Being a company means that we are not an NGO, so our priority has to lie with our customers, which isn't always the exact same as our users. I understand that's a harsh thing to say, but I'm not a fan of sugar coating things. That's also why there is the constant repetition on our side; "We made a decision based on the info and feedback that we had. You want to prevent that in the future? Tell us beforehand". I understand not everyone is able to do so, but it's a bit like voting. You can for any and all reasons choose not to do it, but it then also means that you put your faith in those that do (for better or for worse). 

> The functionality is NOT correct, It does not improve the software, and IS AN ISSUE. Try to sweep it away any way you want, but this is a MAJOR FAILURE!

I think that the community is divided on this. I'm also not sweeping anything away, i'm simply saying how this situation came to be and what can / needs to be done in order to prevent it (because I can not prevent it without open source contributions). Don't mistake inaction on our side for not seeing the issue that is going on, because it's not.

Silvatech, 2019-08-26:
Honestly totally shocked you guys added creawsome the way you did. The profile was terrible they had originally and its worse now. It messed me up on a batch of prints. Finding it was overriding my current profiles i Had. Now granted I should of waited to update, but just really baffled why you would add such code unless you wanted to sabotage Creality printers intentionally because any logic would say this was a bad idea. Really sad, using alternative slicer and frankly I would input, but have never really seen a good way to communicate on the Cura slicer. Hopefully this will open up some ways of communication, but I will have to wait and see. This move is just absurd.

Ghostkeeper, 2019-08-27:
We are honestly doing our best to listen to feedback, really! Back then the feedback was very positive. That's why we added it.

How do you think we should have communicated that these changes have been coming up? What can we do better next time? We don't really have a way to communicate upcoming changes apart from the public beta and this Github repository.

Ghostkeeper, 2019-09-03:
> I don't know why you would think that an Ender would magicially work at a odd force line setting like that.

Well, it works for Ultimaker printers. Our infill is set to 0.5mm line width by default for 0.4mm nozzles. If people claim that it works, I tend to believe them.

> The download page of your website? you have to admit, its very basic. I realize you link out to GitHub, but that's just lazy and unprofessional. Your official website should be communicating these things out.

Yeah I hate this too. Ultimaker's web development team has been refactoring things and in that they forgot to include a place for Cura's betas as well as the old releases. And when we found out it was already like "ah but we don't have time to build that now". So we opted for this solution temporarily.

> I am correct in stating that the "Creawsome" mod, specifically the code changes required, do not affect your ultimaker printers correct?

Correct.

> and with your current base of ultimaker printer users, you would not push a change that negatively affected your legacy machines, correct?

We try our best, indeed. That's only possible because we can test with these printers though, because we DO make changes to Ultimaker's settings all the time. We test them to make sure they are improvements. That's why we ask people who make changes to these 3rd party profiles to test them and ask selected testers among the community to verify them.

Some constructive discussion has been going on in [this](https://github.com/Ultimaker/Cura/issues/6106) thread about the original issue of this ticket. As a result, I've removed the line width multiplier on Creality printers with [this](https://github.com/Ultimaker/Cura/commit/51ed0072db14c0d6d192669d18903d376f6e239c) commit. That fixes this issue, so I'll close this one. If you have further discussions about the Creality profiles and how they were recently changed, I suggest you move that over to #6106.

closed 2019-09-03 by Ghostkeeper