Ultimaker/Cura #6093: [4.2.0] Exclusive Slicing Tolerance loses entire layers

bprzybyl, 2019-07-30:
**Application Version**
4.2.0

**Platform**
Windows 10 64-bit

**Printer**
Custom CoreXY printer

**Steps to Reproduce**
- Open model of known height (I used a 20mm cube)
- Set Slicing Tolerance as "Exclusive"
- Slice Model
- Model no longer fits shadow outline in Z direction in Preview Mode.  Additionally, there will be 2 layers missing.  

This can also be reproduced by creating a model that is 1 layer height and slicing it with the Slicing Tolerance as "Exclusive"

**Actual Results**
A 20 mm cube with 0.2mm layer height and 0.2mm initial layer will only show 98 layers

**Expected results**
A 20mm cube with 0.2mm layer height and 0.20mm initial layer thickness should have 100 layers

**Additional Information**
[image]

[image]

Ghostkeeper, 2019-07-31:
That's basically the point of the setting though. It gives a bit more clearance in the top and bottom side if you use Exclusive.

mbartlett21, 2019-08-01:
@Ghostkeeper 
So should we create the model a slight bit bigger than we need so that it doesn't remove the bottom and top layers?

Ghostkeeper, 2019-08-01:
If you don't need the extra tolerance, I'd suggest to just set Slicing Tolerance back to Middle!

Was there a particular reason why you decided to set it to Exclusive?

mbartlett21, 2019-08-07:
@Ghostkeeper 
When I set it to middle, holes perpendicular to the build plate came out as ovals, while, with it set to exclusive they came out as circles

Ghostkeeper, 2019-08-07:
Is that when actually printing it, or are they already ovals in the print preview?

It's very strange for Cura to generate ovals if the original shape is a circle. Cura has no preference for either the X or the Y direction or anything. Unless the hole was not perpendicular to the build plate, of course, but then an oval would be more accurate.

In places where the 3D model has completely vertical walls, I'd expect the layers to be exactly the same regardless of the Slicing Tolerance setting, because every cross section is the same. Of course at the top and bottom of your model the walls are not vertical any more.

tinkerer2, 2019-09-06:
I don't mean to hijack this thread but there hasn't been much activity lately and I have had a similar issue for quite some time now so perhaps I can help describe it...

From my testing, to measure 20.0mm for the cube in the X and Y, the slicing_tolerance needs to be 'exclusive'.  But to measure 20.0mm is the Z, the slicing_tolerance needs to be 'middle'.  It obviously cannot be both on the same slice.

Ghostkeeper, 2019-09-09:
For a cube, the only difference between "exclusive" and "middle" should be that "exclusive" has one layer fewer. I wouldn't expect any difference in the X/Y dimensions. Perhaps it was randomness? Or was it actually bigger in the g-code?

tinkerer2, 2019-09-09:
I was looking at it again in 4.2.1 this last weekend and things are now different than when I first looked into this.  I should have done due-diligence **before** posting - my apologies.

I agree that X/Y no longer represents https://ultimaker.com/en/blog/52266-whats-new-in-ultimaker-cura-31 and that X/Y always the same now regardless of the slicing tolerance setting.  So my issue has already been resolved.

Regarding your statement on Z though, I was looking through the g-code (0.2 layer height, 0.2 first layer, Cura 4.2.1) and "inclusive" and "middle" do have same and correct layers as you say (Z20.0 on last layer), but "exclusive" seemed to be short two layers.  I don't have my notes here with me right now but as I recall, Z on the last layer was 19.6 not 19.8 which would imply two layers.  I will try and check again when I get home.

Ghostkeeper, 2019-09-09:
Because of how slicing tolerance works, it's expected that "exclusive" would give you fewer layers. After all, it would give you more tolerance.

rachael7, 2020-06-06:
I have observed the same behavior (dropping the first layer when Exclusive is selected) and it just doesn't seem right to me. According to the Settings Guide, "this setting can be used to get more or less clearance between two **_sloped surfaces_**, as can be seen in the attached images." (Emphasis mine) The images show the stair-step approximation of sloped surfaces that FDM printers create and indicates that the Slicing Tolerance setting changes how those are interpreted. This should in no way affect a completely flat surface, such as the bottom face of a cube. Further, even if  you make your layer height such that it doesn't divide evenly into the object height, meaning your part would come out slightly too short because of the rounding error, it still drops the bottom layer, even though there is no need for extra clearance. I think this is an incorrect implementation of the Slicing Tolerance setting and the dropping of the bottom layer is likely an unanticipated side effect.

closed 2020-06-19 by no-response[bot]

tinkerer2, 2020-06-19:
As rachael7 pointed out, in retrospect I think that the problem I had was in the documentation of the feature as it wasn't obvious that the Z would also change.  I was looking for an X/Y sizing solution only.

If I had nominally designed parts (say a large cylinder into a hole with zero clearance), it seemed like exclusive would put the plastic up to, but not over, the STL model edge allowing a tight fit.  Accuracy of Z was not an issue so I didn't want it to change.  Again, I was force fitting this feature as I misunderstood what it was intended for.

I have since taken to doing the calculation myself and using the horizontal expansion/XY scaling to get what I was looking for.  The recent hole horizontal expansion allows a much simpler method of just using the two expansions options.

rachael7, 2020-06-19:
I agree with tinkerer2's comments above, with one exception. As tinkerer2 pointed out, the x/y/z scaling, horizontal expansion, and hole expansion settings already fill the need for general size changes and clearances, leaving just the issue of sloped surfaces that aren't handled. There is a legitimate need there, as illustrated so well in the setting documentation, that isn't addressed by any of the other methods just mentioned. To really make this feature useful, it should be decoupled from planar surfaces and only applied to sloped surfaced, as settings guide indicates. Then it would be both very useful, and free from interference with other size-adjusting settings. As it stands, it is a nearly useless feature, with unpredictable and illogical results.

Ghostkeeper, 2020-06-29:
I'll see what I can do to improve the documentation here. You're not the first to get confused.

The problem with decoupling it from sloped surfaces is that once the model is sliced into layers, everything is a sloped surface. All of Cura's subroutines that deal with surface angle are really just looking at how much difference there is between adjacent layers, and whether that's more than the tangent of some angle times the layer height. It's not able to look at the angle of the original surface because Cura processes a volume which is surrounded by many surfaces. To CuraEngine, the flat surface is just about the same as a surface where the border of the next layer is very far away.