Ultimaker/Cura #6094: [4.2.0] Support Brim generation depends on Buildplate Adhesion settings

GMaidana88, 2019-07-31:
**Application Version**
4.2.0

**Platform**
Win 7 x64

**Printer**
Creality Ender 3

**Steps to Reproduce**
Enable Support Brim with Buildplate Adhesion set to None or Raft.

**Actual Results**
Support Brim is not generated if Buildplate Adhesion is set to None or Raft. If the latter set to Skirt or Brim it's generated correctly.

**Expected results**
Expected Support Brim generation in all cases, regardless of Buildplate Adhesion selection.

**Additional Information**
01: Support Brim: ON, Buildplate Adhesion: None
[image]

02: Support Brim: ON, Buildplate Adhesion: Raft
[image]

03: Support Brim: ON, Buildplate Adhesion: Skirt
[image]

04: Support Brim: ON, Buildplate Adhesion: Brim
[image]

rburema, 2019-08-09:
Devs, see CURA-6700.

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

GMaidana88, 2024-11-22:
Hi Greg,

I just checked on Cura 5.8.0, Win 10:

01) Support Brim: ON, Buildplate Adhesion: None = Works as expected
02) Support Brim: ON, Buildplate Adhesion: Raft = Does not generate brim on first support layer
03) Support Brim: ON, Buildplate Adhesion: Skirt = Works as expected
04) Support Brim: ON, Buildplate Adhesion: Brim = Works as expected

This is a non-issue to me, as I've never encountered a situation where I needed support brim over a raft, but just to be precise about it, this is the actual behaviour:

[image]

Cheers!

GregValiant, 2024-11-22:
I think it's a non-issue too.  The raft supersedes the support brim and so the support brim would not be required in that scenario.
I'll go ahead and close this.

closed 2024-11-22 by GregValiant

GMaidana88, 2024-11-22:
> I think it's a non-issue too. The raft supersedes the support brim and so the support brim would not be required in that scenario. I'll go ahead and close this.

My thoughts exactly, go ahead and close it.

Thanks!