Ultimaker/Cura #6095: [4.2.0] Zig Zag support pattern slices as lines support pattern

cheech1, 2019-07-31:
**Application version**
<4.2.0>

**Platform**
<windows 10 64bit>

**Printer**
<Creality Ender 3 with Printer Settings add-on max Speed Z 10mm/s>

**Reproduction steps**
<selecting support pattern setting Zig Zag>

**Actual results**
<selecting Lines support pattern or Zig Zag support pattern both result in lines support pattern
[image]
>

**Expected results**
<the most bestest zig zag support pattern>

**Additional information**
<Thanks and keep being Awesome>

closed 2019-07-31 by cheech1

Ghostkeeper, 2019-08-02:
From the bottom side it will appear like the grid support pattern since the initial layer support pattern is grid. However if you look from the top side and scroll down the layers a bit you'll discover that it did produce zigzag support.

thelastWallE, 2020-06-14:
So i was searching for this exact thing. I thought maybe someone was issuing this.
It would be nice to have the option or the normal behavior that the initial layer for the support would be the same pattern as the support pattern. Or it should change with the option "bottom pattern initial layer" that we have already in the shell section.
My printer doesn't like to have the filament flow stoping starting and so on.