Ultimaker/Cura #6096: Artillery sidewinder X1

Stone-Ck, 2019-07-31:
HI 
Could you add a new printer? it is Artillery Sidewinder X1.
thanks

nallath, 2019-07-31:
We don't add non-Ultimaker printers. If someone makes a pull request to add it, we will review it.

closed 2019-07-31 by nallath

Ghostkeeper, 2019-08-02:
We can't make a profile for a printer we don't have here at the office. We couldn't test with it.

Stellarspace, 2019-12-18:
> We don't add non-Ultimaker printers. If someone makes a pull request to add it, we will review it.

What if there is an existing configuration that just needs to be updated for the latest version of Cura?

> We can't make a profile for a printer we don't have here at the office. We couldn't test with it.

What do you do to test the 3D printer to create a configuration? Is it guess and check?

Ghostkeeper, 2019-12-19:
> What if there is an existing configuration that just needs to be updated for the latest version of Cura?

We'll add it if someone else tests it.

> What do you do to test the 3D printer to create a configuration? Is it guess and check?

Ultimaker hires a team of process engineers and chemists whose task it is to optimise print quality and productivity. They create a configuration through iterations of test prints, each time observing and reasoning why things are going wrong. They then make adjustments to the profiles based on that and suggest new features for the Cura team to build. So yes, guess and check but also predict and simulate. They are very good at guessing ;)

Stellarspace, 2019-12-19:
> > What if there is an existing configuration that just needs to be updated for the latest version of Cura?
> 
> We'll add it if someone else tests it on an actual printer.
> 
> > What do you do to test the 3D printer to create a configuration? Is it guess and check?
> 
> Ultimaker hires a team of process engineers and chemists whose task it is to optimise print quality and productivity. They create a configuration through iterations of test prints, each time observing and reasoning why things are going wrong. They then make adjustments to the profiles based on that and suggest new features for the Cura team to build. So yes, guess and check but also predict and simulate. They are very good at guessing ;)

I've been using [this](https://3d-nexus.com/resources/file-archives/summary/8-artillery-evnovo/2-artillery-sidewinder-x1-machine-settings-for-cura) configuration, but it has so many customizations to it and only one profile. In the code it looks as if some settings are overriding other settings. I use the default profiles in Cura 4.2.1 with the configuration.

Ghostkeeper, 2019-12-20:
> Are there any rules when it comes to configurations and profiles?

Cura will automatically upgrade any file in its configuration folder to the latest format, and all of the imported files as well (such as that .curaprofile in this zip and things like Cura project files). However we only implement this for the files that Cura can create via the GUI.

Files that can only be installed by copy-pasting things manually into the configuration files will not be upgraded. This includes the definitions, extruders and variants in that Artillery profile archive. You can't normally create these from the Cura interface. You can only have these by modding Cura. Cura won't upgrade these.

Common things that we need to "upgrade" in these files include:
* Remove settings that were deleted.
* Rename settings.
* Change the unit or meaning of a setting (e.g. Prime Tower Wall Thickness became Prime Tower Volume at some point).
* Rename profiles, and then renaming the references to that profile in different profiles.

So that means that a definition file like these may sometimes contain things like settings that we deleted a few versions ago, and Cura won't upgrade them. This is why it may not work to load them. The definitions that are merged into the Cura publication are kept up to date manually by the developers. You'll need to edit them manually as well.

> Do the configurations and profiles have to be made for stock firmware?

Cura doesn't know or care which version of the firmware you're running, so if you're using profiles personally it doesn't matter. However the profiles we merge into the Cura installation are meant to work with the most common printer firmware and make it easy to use, so yes in order for us to merge it we'll ask to test it with stock firmware.

Stellarspace, 2019-12-20:
> Cura will automatically upgrade any file in its configuration folder to the latest format, and all of the imported files as well (such as that .curaprofile in this zip and things like Cura project [...]

How do I know if the settings were deleted or changed in the latest version?

Ghostkeeper, 2019-12-23:
1. Because the setting is gone or changed in the latest version.
2. It'll be in our change log.
3. It'll be implemented in the [version upgrade plug-in](https://github.com/Ultimaker/Cura/blob/master/plugins/VersionUpgrade/VersionUpgrade44to45/VersionUpgrade44to45.py), if you can read Python.

Stellarspace, 2019-12-25:
> 1. Because the setting is gone or changed in the latest version.
> 2. It'll be in our change log.
> 3. It'll be implemented in the [version upgrade plug-in](https://github.com/Ultimaker/Cura/blob/master/plugins/VersionUpgrade/VersionUpgrade44to45/VersionUpgrade44to45.py), if you can read Python.

Does the manufacturer of the 3D printer have to develop a configuration or profile and submit a pull request or can end-users do it? I've tested the profile provided by Artillery with Cura 4.0.0, the speed is too high and they only have a single profile. I've also tested their profile for Slic3r and it's not that different. I know that they are meant to be a base, but if a 3DBenchy doesn't come out correctly using the base, then something is wrong with the profile, right? Artillery only provided the values for the configuration. Is using the default profiles a good start for testing?

Ghostkeeper, 2019-12-30:
> Does the manufacturer of the 3D printer have to develop a configuration or profile and submit a pull request or can end-users do it?

End users can do it. As long as these end users claim that they have tested those profiles and found it to be better in the general case, then we'll generally accept those changes via a pull request. Waiting for the printer manufacturer to supply profiles is not productive.

We can't build the profiles ourselves for you though, since we can't test them on the printer.

> I've also tested their profile for Slic3r and it's not that different.

We don't really consider different slicers to have "the same profile" since many details will be different, as in what the exact order is in which structures are printed, which side they start the line from, that sort of thing.

> Is using the default profiles a good start for testing?

We design the global defaults as you see them in Custom FFF Printer as a baseline for what the developer thinks is a reasonable start for testing. Ultimaker's process engineers fine-tune these settings for Ultimaker's printers so I think that the Ultimaker 3 defaults might be a better place to start from than the global defaults, though you might have some things there that are specific to our printers, like maybe greater accelerations than what cheaper printers can handle.