Ultimaker/Cura #6099: Add post processing script to pause print for alphawise

olivier-boesch, 2019-07-31:
**Is your feature request related to a problem? Please describe.**
the alphawise U20 doesn't recognize the M0 gcode to perform a pause.

**Describe the solution you'd like**
A new script with the M25 gcode (this works).

**Describe alternatives you've considered**
I've written a modified version of the default script : https://drive.google.com/drive/folders/1VDTBfjTMyXb4ArnH09DDlfhRpaMfSOzd?fbclid=IwAR0zfXAu181OiVz2zjPBvi9Ym2yh677pVazcmue4E3SJ-QjubpiVJLkSZTY

or the default script could be modified to let the user enter le right gcode to perform the pause (M0 or M25 or other...)

**Affected users and/or printers**
Alphawise printer owners

Ghostkeeper, 2019-08-02:
So we now have one script for duet that has `M226`, one for Repetier that has `@pause`, one generic one that puts `M83` and one for filament change that puts `M600`. We should merge them into one script, I'd say.

Vandrasc, 2019-08-12:
As discussed internally, we will defer this for now as we do not have time for implementation but we do want to address it in the future.

olivier-boesch, 2019-08-16:
for my own needs, i will continue the dev of this script to make things work good with the U20.
I will be happy to contribute to this script in the future.

fieldOfView, 2020-03-23:
The BQ script uses the M25 gcode. But there's also an attempt to unify all pause at height scripts here: https://github.com/Ultimaker/Cura/pull/7201/files

closed 2023-07-13 by github-actions[bot]