Ultimaker/Cura #6100: [4.2] Unexistant profiles depending on unmanageable materials

Liger0, 2019-07-31:
**Application version**
4.2

**Platform**
Win10 x64

**Printer**
Creality Cr-10S

**Reproduction steps**
Change material.

**Actual results**
With random materials, there will be no profiles. The materials are also unamageable.

**Expected results**
I should have working profiles (even tho the new creality profiles aren't good and are a pain to work with), and manageable materials

**Additional information**
https://youtu.be/uDgGOQsFy8g

nallath, 2019-07-31:
You have working profiles. Just not a quality profile, since they depend on the material. Since you're using custom-defined materials (that, as a result, don't have a matching quality) it shows you a warning to notify you of that. You can still make changes, but you won't have that much "support" regarding what settings are known to work.

Liger0, 2019-07-31:
How can I make changes it there is absolutely no profile at all? On previous versions the profile existed on all the materials, now only on random ones. And they all are custom.

nallath, 2019-07-31:
There is no _quality_ profile. That doesn't mean that a _quality_changes_ profile can't be made. Quality profiles are the ones that can not be changed by the user. Quality_changes are the profiles that you make. So if you make a change, you put in the "user" profile (eg; unstored profiles). If you store those changes ("create profile from custom settings") it puts it in the quality_changes.

All settings in user override those in quality_changes. All settings in quality_changes override in quality & so on.

Liger0, 2019-07-31:
Guess what. Now the quality_profiles are showing with all the material without any warning (why shouldn't stock profiles show up anyway for a material? Why did only some materials have this bug which did hide profiles even ALL of them being custom?)
All I had to do is to switch machine to a custom one, and then switch back to the cr-10S, now all the materials show the stock crappy profiles.
So I do believe this being a bug that I never had on older versions, as all the same profiles showed on all materials, being custom or not.

Liger0, 2019-07-31:
Btw, even tho the material-profile bug solved by itself somehow, I still am unable to manage materials.

nallath, 2019-07-31:
Because some materials probably have a type that have a matching generic profile. If you for instance select Ultimaker red PLA, which doesn't even have a specific profile for Ultimaker machines, it will select the quality profile that maches with generic_pla. 

So when you selected PLA, it did have a specific quality profile. When you selected PETG, it also has a specific profile. PVA and TPU don't have a profile.

Liger0, 2019-07-31:
But, this is 4.2... it randomly shows it or not.
Even in older releases I never had any hidden profile with any custom material
[image]

Liger0, 2019-07-31:
It also isn't updating the profiles if I change nozzle size. 1mm nozzle and stil 0.12mm to 0.28mm layer height, or any nozzle size whatever.

Liger0, 2019-07-31:
Changing machine didn't solve it at this time:
https://youtu.be/vFkUiZ8pVbc

nallath, 2019-07-31:
Could you share your config folder? I just tried creating a custom Esun PVA profile and I get the behavior I just explained to you.

Liger0, 2019-07-31:
With old cura versions I could just create *any* material and the base profiles would never disappear. That's how I except the profiles to work, whatever it could actually be considered a wrong or correct behaviour. 
Because that way I can just use a basic profile selecting the related layer height. The profile shouldn't be influenced by the material except for nozzle temp (wich in 4.2 doesn't have any button to set to inherit from the material either way as per in my other ticket), bed temp and retraction.
And with the current implementation, the nozzle size doesn't change the base profiles anyway as in my last video sometimes, and those times the material also aren't recognized as "custom", which is clearly a bug.
But even so, fixing this buy wouldn't resolve that much except for the nozze/profile correlation.
I'd just delete the known material/profile correlation altogether and only improve the parameters inherit.
[cura.zip](https://github.com/Ultimaker/Cura/files/3453312/cura.zip)

nallath, 2019-08-01:
We used to be in the same boat as you (thinking that material only influenced bed, retraction and fan speed). This is what we did for the UM2.

But a a few thousand hours of testing later, we realised that this is simply not the case. There are a whole slew of settings that both depend on the material as on the quality.

Liger0, 2019-08-01:
Still I can't manage materials on 4.2 because they are unclickable, and changing the nozzle sizes doesn't change the profiles which are locked. So I still end up with a 0.12mm layer height on a 1mm nozzle.
And, even if it worked, I am unsure why in old cura releases the profiles were visible even on custom material and now (assuming the bug was solved, which isn't the case) there aren't base profiles that are shared on custom materials.

Liger0, 2019-08-01:
Did a clean reinstall for 4.2.1. I still can't click on any materials.

Ghostkeeper, 2019-08-02:
The big scary change that you're seeing here is that your printer now has different quality profiles for different materials. This is necessary to improve printing quality beyond a very basic level. For instance I wouldn't want to print nylon on 0.06mm layer height. It just won't work. To allow a printer profile set to say this, the default profiles are not used any more. Otherwise there will be a nylon profile for 0.06mm anyway.

In your case, your printer profiles are essentially saying that printing PC is not supported at any layer height. It colours it yellow then. If you want to build support for it, you have to create a profile. If you want it to have a different layer height, you can change the layer height setting.

I don't understand what you mean that you can't click on any materials though. I tried it and I can click on the material drop-down to select a different material. I can click on materials in the material manager as well. But I suggest that you create a separate bug report for it so that we can process this.

This issue is expected behaviour.

closed 2019-08-02 by Ghostkeeper

Liger0, 2019-08-02:
Expected behaviour? Apart from the material that can't be modified (which I'll make a new ticket), as I said in 4.2 it is expected to change basic profiles when changing nozzle size, because for a 1mm nozzle the layer height should be different, while for a 0.4 nozzle they should be from 0.12mm to 0.28mm. 
But the bug I'm trying to report in all the possible ways, is that no matter the nozzle, could it be 0.2, 0.4 or 1mm, the basic profiles don't adapt, and are always from 0.12 to 0.28mm.

Ghostkeeper, 2019-08-05:
Yeah it should adapt the basic profiles if you change from a 0.4mm nozzle to a 1.0mm nozzle. And as far as I can see it does. My reproduce steps are:

1. Add Creality CR10 printer.
2. Observe that the selected profile is "Standard Quality 0.2mm".
3. Change the nozzle profile to 1.0mm.
4. Observe that the selected profile is "Draft Quality 0.32mm".

That is what you are describing: When the nozzle changes, it changes the quality profile since the larger nozzle size only supports larger layer heights.

Your original bug report is about changing materials. In particular, changing materials to something that has no profiles, so here are the reproduction steps then:

1. Add Creality CR10 printer.
2. Change material to Generic Nylon.
3. Observe that the quality profile is set to Not Supported.
4. Change the nozzle profile to 1.0mm.
5. Observe that the quality profile is still Not Supported.

While the quality profile is not supported, there is no information on which layer heights are available. As far as Cura knows, no layer heights are available. But since every setting must have some value, it picks 0.1mm as default layer height.