Ultimaker/Cura #6104: 4.2 upgrade deletes Monoprice Ultimate printer

donaldnaegele, 2019-07-31:
4.2 upgrade deletes Monoprice Ultimate printer.

Can you add it back?

Don

smartavionics, 2019-07-31:
It's caused by a bug in the new way of presenting the printers by manufacturer. Unfortunately, the manufacturer string in the definition for the Ultimate is "monoprice" but the other monoprice printers use "Monoprice". As a workaround, if you can find the file called monoprice_ultimate.def.json in the files installed with Cura, edit it and change the manufacturer name. Hope this helps until the devs can fix the bug.

donaldnaegele, 2019-07-31:
I can't find that file on my Mac.

Don

smartavionics, 2019-07-31:
Sorry, I don't have a working Mac installation to look at so I can't help.

Here's an idea, try unziping the attached project file and opening it with Cura and it should install the printer.

In fact, loading any previous project that uses the same printer type should create the printer.

[empty.zip](https://github.com/Ultimaker/Cura/files/3453784/empty.zip)

donaldnaegele, 2019-07-31:
That worked! Thank you so much.

Don

Deneteus, 2019-08-01:
I just use the Wanhao D6 profile since its the same printer. I also have the Ultimate.

closed 2019-08-05 by Ghostkeeper

Ghostkeeper, 2019-08-05:
I fixed the manufacturer of this printer. Not sure why the lowercase `monoprice` section would disappear. The only place there where we use case-insensitivity is in the sorting key...

commit ce9bd3b2 (2019-08-05): Fix manufacturer of Monoprice Ultimate

Fixes #6104.