Ultimaker/Cura #6106: Discussion and suggestions about changes for the default Creawesome Creality printer profiles.

JohnEdwa, 2019-07-31:
The new Creawesome profiles have some odd choices, some missed Cura features and in my opinion, could be better. After [a very small Reddit discussion](https://www.reddit.com/r/ender3/comments/ck9agf/help_making_the_new_cura_42_creawesome_profiles/), I compiled my proposed changes and would like to get some feedback and as a possible result, have updated and better default printer profiles included into Cura.

For reference, these are all changes for the "[creality/base/base_global_standard.inst.cfg](https://github.com/Ultimaker/Cura/blob/master/resources/quality/creality/base/base_global_standard.inst.cfg)" using the printer "[creality_ender3.def](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_ender3.def.json)" that itself is based on the "[creality_base.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_base.def.json)" definition.

The comments are mostly mine, and might (will) have factual errors and misunderstood functionality as I'm not a Cura expert.
Cura Profile: [creawesome_ender3_change_suggestions_v1.zip](https://github.com/Ultimaker/Cura/files/3453974/creawesome_ender3_change_suggestions_v1.zip)

---

#Suggested in the discussion but I don't know about them
* Print Thin Walls should be enabled.
* Minimum Extrusion Distance Window: 10 is to large of a value according to /u/Liger_Phoenix
* Line Width should be equal to Nozzle size, according to /u/Liger_Phoenix

---

#Shell    

* Outer Wall Wipe Distance: 0 -> 0.2mm // this is supposed to reduce the Z-seam, so why not use it?
* Filter Out Tiny Gaps: False -> True // the tooltip just says "Filter out tiny gaps to reduce blobs on the ouside of the model". Sound fine to me
* Seam Corner Preference: None -> Smart Hiding // Hiding works rather well and looks good.
* Z Seam X: 117.5 -> 0 // Using the Relative Z seam instead
* Z Seam Relative: False -> True // see above

#Material
* Retraction Extra Prime Amount: 0 -> 0.064 // see Enable Coasting at the bottom - should be a nozzle-based function and not a hard value.

#Infill
* Infill: Cubic -> Gyroid // /u/Liger_Phoenix: "I find it works everywhere and everytime... ...the gyroid is the optimal starting point for any stock machine."
* Infill Overlap Percentage: 30% -> 15% // 30% can cause issues in narrow spots without much benefit
* Infill Before Walls: False -> True // Much better overhangs and the infill showing up is never going to be an issue with the default 3 walls.
    
#Travel    
* Combing: Infill Only -> Not in Skin // Faster priting, but seems to be practically the same quality
* Z-Hop When Retracted: True -> False // Easily causes uneven layers as the leadscrew can bind and the nut has backlash. It's also not really needed for anything if you aren't overextruding and avoid printed parts is enabled.

#Support
* Support Type: Buildplate Only -> Everywhere // Supports should always default to everywhere for newbies. Supports = I want the print to succeed.
* Minimum Support Area: 10 -> 5 // Helps support smaller details
* Use Towers: False -> True // Helps reduce the generation of tiny thin supports with no chance of staying up

#Build Plate Adhesion
* Build Plate Adhesion Type: None -> Skirt // Skirt is always good for priming the nozzle and checking first layer squish
* Skirt Line Count: 4 -> 2 // ...but you don't need 4 lines for it.

#Special Modes
* Relative Extrusion: False -> True // /u/Liger_Phoenix: "Relative extrusion can be enabled and is always recommended on Marlin."

#Experimental
* Slicing Tolerance: Middle -> Inclusive // /u/Tarasque_1024: "anything else might drop details on some models"
* Enable Coasting: False -> True // /u/Liger_Phoenix: "it will work always fine and make seams and underextrusion disappear.

nallath, 2019-08-01:
Coasting on a direct drive printer? Are you sure about that? I've only heard complaints about coasting being enabled by default (the main reason for it is that it does work for Ultimaker printers).

Liger0, 2019-08-01:
Creality machines are all bowden.

FITPrinting, 2019-08-01:
> Coasting on a direct drive printer? Are you sure about that? I've only heard complaints about coasting being enabled by default (the main reason for it is that it does work for Ultimaker printers).

> Creality machines are all bowden.
> I'd also disable "perimeter overlap", it always gave me bad results.

Between this thread and the other where "nallath" is defending the forced adoption and overwriting of saved profiles, it seems obvious that they don't know anything about the machines that are being affected. Opt out (not that there's a way to opt in now) is never the proper procedure and I wonder how badly the code has been written if the only way to update it was to force it to overwrite where it wasn't welcomed.

nallath, 2019-08-01:
> Between this thread and the other where "nallath" is defending the forced adoption and overwriting of saved profiles, it seems obvious that they don't know anything about the machines that are being affected. 
You're right. I have absolutely no idea how ender printers work. I also never pretended to do so. 

So we simply rely on what others tell us in that regard. As I've stated before, although this feedback is welcome, it's a bit on the late side to change it.

JohnEdwa, 2019-08-01:
They probably don't, Cura is from Ultimaker designed for the Ultimaker printers, everything else is a bonus supported by the community. That means us.

In any case, let's concentrate on fixing and making them better than bicker about who did what wrong, it doesn't help anything.

Liger0, 2019-08-01:
Only who used the creawsome mod knew that was going to be incorporated to Cura because they followed the creator, its group, posts, etc, there is no way other people would have known it.
So of course we had really no way to complain about it before it was done, since none of us was asked and no plublic announcment was ever made, and none of us ever focused on what was happening to that useless mod. That just seemed a different alternative for lazy users (read: not new entry, because new entries can just learn like everyone else), so that shouldn't have affected so much people which had everything ready and optimized profiles that may even use it for work and made a reliable profile, which become unreliable with this new mod without them being aware of it in any way.

By the way, I think who made this mod wanted to help but doesn't really have a clue on what is better for creality printers, so I hope someone will make the changes suggested here.

nallath, 2019-08-02:
You could have found out by means of beta's, forums and github. The sad truth here is that we simply have to give priority to people that do involve themselves in the entire process of a Cura release. 

The argument regarding that those profiles are bad is re-iterated over and over again, but I've talked to numerous people that claim that the profiles are a vast improvement over what was already there. So who are we to believe then? People who take the time to try and improve Cura or those that we only hear if their workflow is affected? I try to facilitate both, but the earlier we get feedback, the easier (or at all possible) it is.

Even if I were to agree that the change was bad now, I would still not change it back, because it would simply be doubling down on the issues in the next release. I think it's a much better idea to do what @JohnEdwa suggests; Use the profiles as they are and improve on them further.

JohnEdwa, 2019-08-02:
[Issue #6134](https://github.com/Ultimaker/Cura/issues/6134) is the first one to come from the current stock Creawesome settings. Supports are limited to "Touching Buildplate" so they aren't being generated as people expect using recommended settings.

This should be changed immediately to "Everywhere" even if no other changes are made.

Liger0, 2019-08-02:
If that gets to everywhere, I'd disable the towers tho, because towers are bugged and are created randomly even if they don't have to support anything. This would ruin the surfaces if they are printed on the model.

JohnEdwa, 2019-08-02:
The main reason I like using Support Towers is that without them Cura likes to generate [supports like these](https://user-images.githubusercontent.com/5417183/62401180-bb0fcd00-b58a-11e9-859e-97b0b1d234f7.PNG) and they have absolutely zero chance of staying attached to the build plate. Well, now that [the prime tower got it after over two years of being in the "top 50 important features" list ](https://github.com/Ultimaker/Cura/issues/1950) maybe they can add them for the supports as well.

---

The old XY distance from the fdmprinter default was 0.7mm, Creawesome has it as ``wall_line_width_0 * 2``, meaning with the 0.4mm nozzle \* 1.1 * 2 -> 0.88mm.
Should it be one line width from the wall instead or would that be too close and risk fusing... Maybe 1.5?

Liger0, 2019-08-02:
If we want the supports to be created from the model overhangs, 0.8 is way too high. 0.3 to 0.5 should be for a 0.4 nozzle.

JohnEdwa, 2019-08-02:
Would seem logical to then set them to be based on the nozzle size.

* Support X/Y Distance: wall_line_width_0 * 2 -> machine_nozzle_size
* Minimum Support X/Y Distance: wall_line_width_0 -> machine_nozzle_size / 2 (which would be the same value it used to be with the old fdmprinter setup)

huginen, 2019-08-03:
I should be a good idea to let the user choose either to use Creawsome or a standard profile, my cr-10s prints great with a slightly modified standard profile, and not any better prints with Creawsome

Every printer is not the same

Liger0, 2019-08-03:
The *only* setting that makes sense on the creawsome mod is the 0.04 layer height interval on adapative layer.

JohnEdwa, 2019-08-03:
I think it's too conservative, 0.02mm or even the old stock 0.01mm should be perfectly doable if your leadscrew and stepper are aligned. The Creality printers have 1/16th microstepping on the Z-axis with the 0.04mm lead so it can move in 0.0025mm increments, and those would be 1/2 and 1/4th steps.   
It's not like a printer that homes, uses babystepping or even mesh bed calibration will ever actually be doing perfectly aligned full steps when changing layers.

I think the Creawesome dev jumped on the "Magic Numbers" bandwagon slightly too hard.

Liger0, 2019-08-03:
The 0.04 increments actually help in accuracy.

JohnEdwa, 2019-08-03:
If it was actually moving with full steps I would absolutely agree, going from a full step 0.04mm to a full step 0.08mm has 100% the torque and it will hit that step position exactly. But it's not. You home your printer and the stepper stops at a 6/16th microstepped position because that's where your end switch is. Now when you start printing with those 0.04mm incremental layers, you are asking it to move from 0.00mm + 6/16th microstep to 0.04mm + 6/16th. Then to 0.08mm + 6/16th. Then 0.12mm + 6/16th and so on.

It makes very little difference for it to move to that 0.04mm + 6/16th or to 0.04mm + 12/16th, it has to move to a microstepped position anyway.

Pimentoso, 2019-08-04:
I might suggest some too.

- Overhang wall angle = 50
- Overhang wall speed = 66% # helps having good overhangs at higher speeds
- Optimize wall printing order = true # helps reducing unnecessary travel

JohnEdwa, 2019-08-04:
Oh neat, hadn't even noticed those overhang ones exist.

Another one in the experimentals is "Conical Supports". When you set it to a small negative angle they work how I want Support Towers to work like without the buggy behaviour of generating supports that don't do anything randomly. That is, it makes sure [the support is a certain minimum size at the bottom.](https://user-images.githubusercontent.com/5417183/62423892-bd2f7400-b6cf-11e9-9bf4-49fb460ee73d.png)
I still gotta test how it works in more situations though.

Liger0, 2019-08-05:
Also default first layer height to layer height instead of them being kept separated values.

Ghostkeeper, 2019-08-07:
> Another one in the experimentals is "Conical Supports". When you set it to a small negative angle they work how I want Support Towers to work like without the buggy behaviour of generating supports [...]

I really like conical support myself and use it quite often. We have to re-evaluate to put it outside of experimental. As far as I know the last bug that was in there recently was fixed in 4.2: That it could expand to outside of the build volume.
Conical support is very effective when combined with supporting to build plate only, because the support can crawl around your object then similar to tree support.

Liger0, 2019-08-09:
I also noticed it always zhops between different models even tho zhop is disabled. I don't know if this actually is intended behaviour.

oofiksoo, 2019-08-19:
does anyone feel like this "upgrade" ruined your prints? it appears as though simply upgrading cura has caused my brim lines to be separate. It doesn't appear as though I am able to achieve the proper "squish" any more no mater how I adjust the build plate or adjust the z probe offset. This issue became prevalent after the upgrade, as I use my printer daily, I noticed the change immediately, but chalked it up to me needing to fine tune based on the new functionality. I do not see a way for me to "fine tune" this to the level I had it tuned prior to the upgrade. To put it simply, I do not see a benefit in this change, and believe this should have been optional, or should be further exposed in the configuration allowing me to adjust the "improved" values. basically, I feel like I am dead in the water right now. Nothing I do brings my quality back. I will be compiling a ver 4.1 release and not upgrading going forward.

Liger0, 2019-08-19:
Because the line width is now set to 1.1*nozzle size, so it risks to cause underextrusio, while before you risked overextrusion. Either set manually the line width each time or calibrate the flow.

Liger0, 2019-08-21:
Let's show you what the creugly mod implementatio nhas caused to users:
https://www.reddit.com/r/CR10/comments/ctlspe/left_print_is_older_cura_right_print_is_newest/

In a few words, only issues.
The only way I can succeed printing now is enabling firmware retraction, but that causes a lot of stringing.
Also, 30% of infill overlap causes issues with material on the corners, I lowered it to 20.

Then another stupid thing is skirt gets disabled when enabling support.

oofiksoo, 2019-08-22:
I can fully confirm my issues are resolved fully by reverting to a fresh copy of cura 4.1. I lost my customizations because I wanted to ensure there was no leftovers, and did a full uninstall reinstall. Even without customized profile, my prints are of better quality than 4.2 prints. I feel confident that there are issues with this functionality that should not have made it to production, without further testing. I do not believe the issues are limited to configurable changes within cura 4.2.

Liger0, 2019-08-22:
Edit: now the skirt disabled when support enabled has been fixed by ghostkeeper. Still I have issues with retraction causing clogs with 4.2.1 unless I use firmware retraction.
There are a lot of people reporting how shit the creugly implementation was, yet just another one:
https://www.reddit.com/r/3Dprinting/comments/ckq73i/thanks_cuts_for_forcing_the_creawesomemod_upon_us/

oofiksoo, 2019-08-22:
I feel so strongly about this issue, and the way that Ultimaker is handling the issue, that I am compelled to copy/paste my message left on a seperate thread. I challenge all readers of this post to make your voice heard as well. Although I personally will no longer have a vested interest in the ultimaker cura development process, For your own sake, Please challenge the pride and ego of the development resources responding to our posts, and request this functionality be corrected.

" Ok, I will not be purchasing an ultimaker printer, and I will no longer be using the ultimaker software. The consistent response of "we made this change, your not happy about it, you should have done something about it, we're open source" is bogus. Not everyone in the world has time to help ultimaker make money and build as a company. Even if you offer an open source product, it is possible that your small group of people made a bad decision. The Creawsome functionality should not have been introduced, and your pride will simply not allow you to admit that this was a rush job. You guys saw a few happy campers within the same cohort, and though this functionality was great. This is a direct failure by the "open source community" working on Cura, and you are being absolutely rude, condescending, and off putting. You are not special because you are offering an open source product, and the users of the product should not be made to feel like we owe you something, we should be happy with what you give us, and we should shut up if were not contributing to your growth.
As I said before, I will not purchase an ultimaker printer, And I will not be using the Ultimaker Cura platform. I can deal with issues, bugs, and things that need my attention, But I will not support a company so off putting, and whom responds with the remarks such as yours. It is obvious that this is an issue, it reaches many of your users, and you refuse to take any other position than for us to contribute or shut up.

The functionality is NOT correct, It does not improve the software, and IS AN ISSUE. Try to sweep it away any way you want, but this is a MAJOR FAILURE! "

Liger0, 2019-08-23:
Since with the creugly implementation, using the creality machine definitions has become impossible, I will have to use and set from 0 a generic fdm machine from now on. Thanks for making life so hard to the community making your software unusable...

hessius, 2019-08-23:
(Maybe the general complaints about how unfair, wrong or plain evil people seem to feel the well intentioned decisions of the people making free software (and mods of that software) for us can be kept to reddit / facebook etc. Or at the very least another thread as this was intended to actually try to remedy the situation. Everyone knows that you are dissatisfied, the point has been made. Several times. Maybe we in the creality community can start trying to fix things?)

For me I’ve found that the retraction set as default for this profile has been insufficient to prevent stringing, I’ve seen the suggestion of enabling coasting and have yet to try it, maybe retraction distance should be upped to 6.0 anyway?

Ghostkeeper, 2019-08-23:
I feel that we have a bit more caution now for changing settings suggested by individual users, partially due to the feedback we've gotten about the recent Creality setting changes. And there are a few suggestions even in this thread that contradict each other. But to make any improvements here we need some more force of action. I understand that making pull requests is complex if you're not used to the workflow, so how about this: *If two people tested an actual print with the same setting change and found it improved, we'll make the change.* Does that sound okay?

Things that multiple people have confirmed and should already be applied if people agree with this workflow:
* Support Type: Everywhere (by JohnEdwa and Liger0)
* Build Plate Adhesion Type: Skirt (by JohnEdwa and Liger0)

Currently open proposed changes in this topic tested by only one person afaik:
* Line width: =machine_nozzle_size (by Liger0)
* Seam Corner Preference: Smart Hiding (by JohnEdwa)
* Retraction Extra Prime Amount: =machine_nozzle_size * 16 / 100 (by Liger0)
* Support X/Y Distance: =machine_nozzle_size (by Liger0, to accompany the new Support Type value)
* Minimum Support X/Y Distance: =machine_nozzle_size / 2 (by Liger0)
* Support Interface: disable (by Liger0)
* Optimise Wall Printing Order: enable (by Pimentoso)
* Overhang Wall Angle: 50 (by Pimentoso)
* Overhang Wall Speed: 66% (by Pimentoso)
* Infill Pattern: Gyroid (by Liger0)
* Infill Overlap Percentage: 20% (by Liger0)
* Infill Before Walls: True (by JohnEdwa)
* Combing: Only in Skin (by JohnEdwa) vs. None (by Liger0), may depend on material
* Minimum Support Area: 5 (by JohnEdwa)
* Skirt Line Count: 2 (by JohnEdwa)
* Skirt Minimum Length: 250mm (by Liger0)
* Relative Extrusion: enable (by Liger0)
* Conical Support: enable (by JohnEdwa)
* Conical Support Angle: -2 (by JohnEdwa)
* Conical Support Minimum Width: 10 (by JohnEdwa)

Currently open proposed changes that were just intuition and seem to not be tested at all yet:
* Outer Wall Wipe Distance: 0.2mm
* Print Thin Walls: enable
* Filter Out Tiny Gaps: enable
* Z Seam Relative: enable
* Z Hop: disable
* Support Interface Resolution: =layer_height
* Slicing Tolerance: Middle (personally I doubt that this will have the desired effect though)

One proposed change here is to set the Z Seam X to 0 because of using relative mode. We're probably going to do that across default (to make it auto-adjust based on relative). We've had some feedback from other people about this as well.

There is also a proposed change to `machine_head_polygon` to remove it entirely, but that can cause the print head to collide with previously printed models so I wouldn't advise it. If the print head needs to be corrected though that can be done.

Could you guys test each other's proposals?

tilllt, 2019-08-23:
In my use case with creawsome and/or cura 4.2.1 the practical problems I came across was IMHO 
1. part fan in material settings for PETG should be off.

Liger0, 2019-08-23:
I don't advice creating a profile driven fan speed, that has to be completely used from the material. For instance, to work with petg I NEED to use fan at 30%.

For the combing, yeah, not in skin can still work, depending on material.
For the infill, as I read people using octoprint may have issues with gyroid but it is not for a stock machine. Maybe the box dialogue could put the advice that it could oversaturate and cause stutter if not using an sd card?

tilllt, 2019-08-23:
I did fairly intensive testing with PETG and the default fan settings will lead to bad prints. I would bet that the results with fan off are - for the majority of people - better than with fan on. Fan @ 30% depends on your fan, your duct etc so yes a certain setting I doesn't make sense. But off makes more sense than the default like it is now. Just try it.

Liger0, 2019-08-23:
I can't actually try it because with cura 4.2.1 all my prints fail with filament being grind unfortunately.
But I agree with petg having the stock fan OFF could be better, just don't force a profile rpm.
For instance I use a 5015 fan which blows pretty hard.

JohnEdwa, 2019-08-23:
>If two people tested an actual print with the same setting change and found it improved, we'll make the change. Does that sound okay?

How soon is the next beta/release? My idea is to compile the changes we agree to here and then post them to Reddit for people to test and give more feedback without officiall pushing them into Cura.  It would also give more accurate results as Creality printers are often heavily modified, so what works for the stock printer or my printer might not be what most people would use, nor a good default setting.

And the more we change, the more people will complain things changed *again*. The list I made is not an "all of these should be changed" one, but just suggestions of ones that could be. And I definitely don't try to push myself as an expert in any way - I don't want my head to be the next one on the chopping block.

---

>There is also a proposed change to machine_head_polygon to remove it entirely, but that can cause the print head to collide with previously printed models so I wouldn't advise it. If the print head needs to be corrected though that can be done.

Because there are two polygons, ``machine_head_with_fans_polygon`` and ``machine_head_polygon``. The one [without will also collide with the outside of the build plate stopping you printing in the corners](https://user-images.githubusercontent.com/5417183/62412823-41222700-b610-11e9-9e3d-ed07267fc408.PNG), and therefore you can hardly use the "one at a time" mode at all as you will run out of space even with tiny things. Either that is bugged in in Cura, or it isn't supposed to be used like this.   
If you remove it leaving just the other polygon it will work as long as the grey area doesn't hit another part, but they can overlap as there isn't any reason it can't move in the same empty spot while printing.

---

I would categorize the change roughly in order of importance as follows (and I repeat, these are my opinions):

**Bugfixes/rollbacks:**
Most are also values from the old definition.

* Support Type: Everywhere (by JohnEdwa and Liger0)
* Build Plate Adhesion Type: Skirt (by JohnEdwa and Liger0)
* Skirt Line Count: 2 (by JohnEdwa)
* Skirt Minimum Length: 250mm (by Liger0)
* These are Issue #6147 related, so actual values might change?:
  *  Support X/Y Distance: =machine_nozzle_size (by Liger0, to accompany the new Support Type value)
  *  Minimum Support X/Y Distance: =machine_nozzle_size / 2 (by Liger0)
  *  Minimum Support Area: 5 (by JohnEdwa)

**General Improvements:**

* Infill Before Walls: True (by JohnEdwa)
* Optimise Wall Printing Order: enable (by Pimentoso) 
* Z Seam Relative: enable
* Z Seam X: 0
This is what the definitions were trying to do with the seam before the relative was added as a setting - put the seam at the back of the model.
* Filter Out Tiny Gaps: enable
I've always used this but I don't know how/if it works.

**"Controversial"** (for a lack of a better term, everyone has their own preferences and opinions, so testing and gathering more feedback on these would be required before implementation):

* Support Interface: disable (by Liger0)
And JohnEdwa - this creates issues with the supports fusing into the model for many people but I'm not sure if disabling it would require tweaking other support settings as well. As a default for not properly calibrated printers I think they should be disabled.
* Support Interface Resolution: =layer_height
* Line width: =machine_nozzle_size (by Liger0)
I use nozzle_size * 1.05 following [Slic3r suggestions.](https://manual.slic3r.org/advanced/flow-math), but without "Print Thin Walls" this won't print a wall that is exactly 0.4mm. However, that will also print super tiny things, usually resulting in strange artefacts.
* Print Thin Walls: enabled
See above.
* Outer Wall Wipe Distance: 0.2mm
* Z Hop: disable
I don't think Z-hop works with a Creality printer because the Z-axis is far from perfect and will bind, resulting in uneven layer height
* Infill Pattern: Gyroid (by Liger0)
Gyroid is slightly overkill imho, but it's pretty and works well, so I don't see why not.
* Infill Overlap Percentage: 20% (by Liger0) 
* Combing: Only in Skin (by JohnEdwa) vs. None (by Liger0), may depend on material
[None causes surface scarring](https://content.invisioncic.com/ultimake/monthly_2017_09/IMG_20170913_150049.jpg.cad418a8bdbe35c3078b181be34ca05d.jpg), but it is slightly faster. IMHO quality > quantity.

**Experimental** (and therefore maybe not suitable as defaults? Also mostly untested)

* Relative Extrusion: enable (by Liger0)
* Overhang Wall Angle: 50 (by Pimentoso)
* Overhang Wall Speed: 66% (by Pimentoso)
* Conical Support: enable (by JohnEdwa)
* Conical Support Angle: -2 (by JohnEdwa)
* Conical Support Minimum Width: 10 (by JohnEdwa)

Things I don't know enough about:
* Fan off for PETG (by Tilllt)
* Slicing Tolerance: Middle
* Retraction Extra Prime Amount: =machine_nozzle_size * 16 / 100 (by Liger0)
* Outer Wall Wipe Distance: 0.2mm

Liger0, 2019-08-23:
But even before we do these changes (as I repeat, Outer Wall Wipe Distance should be 0.2mm only for 0.4mm nozzle size and use half the nozzle as default, just like coasting as they use the same principle pretty much).

Ghostkeeper, 2019-08-23:
> How soon is the next beta/release?

In about 2 weeks is the next beta, I believe.

Liger0, 2019-08-23:
I also don't suggest using the option to "enable infill before walls", because that caused underextrusion to me on the points where the infill touched the walls, so the infill was showing and even worse it was not because of material excess but material missing.

Liger0, 2019-08-24:
I also suggest to disable the externall wall overlap compensation, because that ruins all the sharp corners making them round.
The internal one is fine, because it is not reall visible.

Routout, 2019-08-24:
Cannot really understand why Gyroid is listed as maybe default. It shakes a shit out of the ender..

Liger0, 2019-08-24:
Then you probably still use the stock crap fire hazard firmware.

JohnEdwa, 2019-08-24:
Well, if that happens with the stock firmware then gyroid shouldn't be the default for the profiles. It's not really needed most of the time anyway, grid or even lines works perfectly well enough as tested by [Stefan from CNC kitchen.](https://i.stack.imgur.com/NDysz.png) (in Simplify3D, Rectilinear is the same as Curas' Lines).   
And the problem with Cubic is that it has a chance of leaving very large gaps that result in bad top layers.

Liger0, 2019-08-24:
Stock firmware will randomly freeze and make the printer catch fire anyway. It also will give bad result even without gyroid because of the absurd jerk and acceleration.
Creality printers should not be even powered on before upgrading firmware.

Liger0, 2019-08-25:
Btw, grid is still a good pattern, so having it as default is still fine to me.
@Ghostkeeper would it be possible to add in the floating box of the infill that the gyroid could oversaturate the serial connection, from what I hear? I use sd card so I can't confirm

Ghostkeeper, 2019-08-25:
> @Ghostkeeper would it be possible to add in the floating box of the infill that the gyroid could oversaturate the serial connection, from what I hear?

It's possible but I don't think it's a good idea to add too much details about each individual pattern in that text. I've written about the problem in my [article about Cura's infill patterns](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill_pattern.md) in the Settings Guide though.

Liger0, 2019-08-26:
> It's possible but I don't think it's a good idea to add too much details about each individual pattern in that text. I've written about the problem in my [article about Cura's infill [...]

Is that guide actually directly accessible from within Cura?

Ghostkeeper, 2019-08-26:
Yeah it's in the Marketplace. You need to install the plug-in first. That article there is not on the Marketplace yet though. My updated plug-in is still under review for the past 2 weeks because the guy that reviews these is on vacation right now.

Liger0, 2019-08-27:
A big still not considered by the most people reason the new profile gives way worse print quality than the old one is, other than the stupid retraction and all other settings that the 
Outer Wall Wipe Distance
is off instead of half the nozzle size.
Also the stupid line width causing under extrusion, which should be equal instead to nozzle size and then calibrate flow to fix overextrusion.
Relative extrusion works fine, it just doesn't work in Cura reader (but works in other slicers readers).
Disabling the wall overlap compensation for the external wall is also a very important thing as I said.

JohnEdwa, 2019-08-27:
@kazooless 
>Z Seam Relative: enable NO
I would prefer Z Seam: Sharpest Corner

Z Seam Relative just makes the Z seam coordinates be related to each individual model and not the build plate. The current setting is trying to push the seam at the very back of the model, which will simply work better with multiple objects if you use the realative option.
[Pictures tell thousand words, so 9001 hours in MSpaint.](https://user-images.githubusercontent.com/5417183/63791302-e7a5d300-c903-11e9-86b4-84d05c8dfea4.png)

And the issue with Sharpest Corner is that doesn't work on objects that don't have clear corners and it will just randomly start wherever and cause the seam to be very visible, very likely at the front surface and ruin the visual quality of the print.
User Specified combined with Smart Hiding tries to put it in a corner at the back of the object, and if there are no corners, it puts it at the back in a straight line.

Liger0, 2019-08-27:
Let's not forget the first layer height = layer height that I proposed earlier. Sometimes when changing the layer height I forget to change first layer height and it's so bad.

Also, this is what I have with "infill before walls", even with 4 perimeters. I don't suggest to enable this.
https://i.imgur.com/upt9nvI.jpg

Liger0, 2019-08-27:
I have that always with that option On, that is the reason I have it off. The number of walls doen't really matter.

JohnEdwa, 2019-08-28:
At 5 walls there is absolutely zero possibility that the infill would actually show through as that is 2mm of wall, and that artifact has to be the result of something different.  

And reason you should do infill before walls is that it gives overhangs and bridges something to attach themselves into.

tilllt, 2019-08-28:
I understand that this discussion is about proposal for new cura default settings for the ender3... Wouldnt it make sense to have a seperate repo for the refining of the settings and when people agree that the settings are good, just issue a PR and merge them with the then actual cura?

Liger0, 2019-08-28:
this is generic for creality printers. Also let's not forget to add if you people also agree to default zhop height to layer height even having it off by default.

Liger0, 2019-08-28:
> Enable Coasting NO!!! A lot of us update our firmware and turn on Linear Advance. Coasting is recommended to be turned off for that.

I agree but Linear Advance is for expert users that are dialing in the printing a lot and know what they do. If you enable linear advance in the firmware and dial it correctly, you will know to disable coasting in the profile. 
But coasting gives (well, will give since the blob issue will be fixed for 4.3) a huge benefit for all machines with stock and not very advanced functions.

Liger0, 2019-08-28:
> 
> 
> > Also the stupid line width causing under extrusion, which should be equal instead to nozzle size and then calibrate flow to fix overextrusion.
> 
> @Liger0, regarding the default line width, I'm fine either way. I have some profiles using the nozzle size and some using 120%. I tend to use the 120% for parts & 100% for artistic stuff. I have my e-steps and flow dialed in really well and find I can get great walls (close to perfect measurements) with either setting.
> 
> For your combing comment, I don't follow what you mean by the mm's.

After proper calibration they both can work well.
The point is that using a line width > nozzle you are in a case that will cause udnerextrusion to people and this isn't good for a starting point profile since people, especially the new entries, will first do their first prints to test their hardware with stock profile and learn from it.
But I see people are going crazy about it because of underextrusion issues caused by the larger line width, and it's not a good thing.
Using line width=nozzle size will always work on the other side, it may just cause a bit of overextrusion that won't affect the success of the print and the quality and can be calibrated later with flow or a user fixed line width easily.

For the combing, I mean that I had better results using a "max combing distance without retraction" of 15mm instead of 30mm.

Ghostkeeper, 2019-09-02:
> The question first. When CreawesomMod first came out the information about it said it was more than just a custom profile; we were told there were "under the hood" changes made. Looking at one of the original documents it says part of the changes made were "* Deeply defined machine & * Fixed default feedrate, accel and jerk hidden settings." So the question is, if all the visible settings were made to be identical to the earlier versions settings, would the resulting gcode be the same as the earlier version?

The original CreawsomeMod had some interface elements also overwritten with its own. This broke in Cura 4.1 though because we'd made new interface elements that referred to the overwritten ones. We didn't take these changes into Cura 4.2. CrawsomeMod also changed the printer definitions for default feedrate, acceleration and jerk settings. Those settings were taken along to Cura 4.2. You can still see them if you load fieldOfView's Machine Settings plug-in, but they are normally hidden. They **can** influence the g-code, but most of them will only do so in a very minor way. For instance, the maximum feedrate influences time estimates which causes a different time estimate to be put in the g-code and may cause pre-heat commands for extruder switches to be placed earlier or later because it estimates the print to take longer or shorter.

> I understand that this discussion is about proposal for new cura default settings for the ender3... Wouldnt it make sense to have a seperate repo for the refining of the settings and when people agree that the settings are good, just issue a PR and merge them with the then actual cura?

We've asked people to submit PRs to change these settings but that seems to be difficult for some people (forking, committing, etc. and people don't monitor the PRs). So I thought that this thread would be simpler for them.

Ghostkeeper, 2019-09-02:
I've updated my list here: https://github.com/Ultimaker/Cura/issues/6106#issuecomment-524241165

Tomorrow I'll apply the top entries. I've tried to filter out only comments that were actually tested by people.

Liger0, 2019-09-02:
Nice. Also take note of the zhop to default to layer height if it gets manually enabled.

[42 more replies]

closed 2020-07-30 by Ghostkeeper