Ultimaker/Cura #6107: 4.2 fails to import 4.1 curaprofiles

OldGaf, 2019-08-01:
This was reported in the beta and was closed as the fix was to go into the stable ver.
It still is broken.

see:
4.2 Beta fails to import 4.1 curaprofiles #6033

**Application version**
4.2

**Platform**
Win7

**Printer**
CR-10s

**Reproduction steps**
Installed 4.2 and saw no profiles for my CR-10s, but all the Select Mini profiles are fine.

**Actual results**
No CR-10s profiles loaded on install. Manual import also fails with the following error:
"The machine defined in profile E:/Downloads/CR-10-Mech.curaprofile (fdmprinter) doesn't match with your current machine (creality_base), could not import it."

**Expected results**
Expected profile to be available on install. Expected manual import of profiles to work.

**Additional information**

b4545, 2019-08-01:
Any known workarounds?

diegopradogesto, 2019-08-01:
Duplicate of #6083 . We will release a hotfix (Cura 4.2.1) today that fixes this issue (and others).

mihxew, 2019-08-02:
Same problem on Cura 4.2.1!!!

jvcop, 2019-08-05:
I can confirm this problem on Cura 4.2.1. (Windows 10)

Ghostkeeper, 2019-08-07:
The problem here is that Creality printers now have profiles specific to themselves. Cura doesn't know which profiles to put in the "Creality" category and which to put in the "fdmprinter" (profile-less printers) category.

We're looking into finding a way around this. I've heard @nallath had some ideas?

nallath, 2019-08-07:
Yeah. The original reason we initially set the filtering of quality_changes there was to prevent people from accidentally importing profiles for the wrong machine. But I'd say that in this case, it's doing more damage than it's trying to prevent.

Vandrasc, 2019-08-12:
For our internal reference, devs see CURA-6713.

polerix, 2019-08-28:
i eed  omputer with workig keys or  eterl usb keybord this lptp is the pitts hee retypig the proile is rough

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

OldGaf, 2024-11-24:
Feel free to close

closed 2024-11-24 by GregValiant