Ultimaker/Cura #6108: [4.2] Coasting causes unecessary priming

edgary777, 2019-08-01:
Currently using Cura 4.2, but this has been showing since 4.0 and probably earlier.

I'm currently using windows 10, but it can also be replicated using the linux version (Arch, custom install)

I'm using a Tevo Tornado with a custom Cura profile.

When using coasting, whether retraction is enabled or not, cura primes a small amount of filament after the coasting event, I think that it is about the same material that would be extruded if coasting wasn't activated, causing a small but very obvious blob to appear after the coasting, defeating the purpose of the feature entirely.

Here's a sample from the attached Gcode showing a filament priming of 0.04032 after coasting:

```
G1 X155.507 Y126.998 E3229.72109
G0 F2880 X155.426 Y127.152
G0 X155.393 Y127.201
G0 X155.163 Y127.56
G0 X154.905 Y127.917
G0 X154.815 Y128.01
G0 X154.619 Y128.239
G0 X154.343 Y128.532
G0 X154.225 Y128.629
G0 F12000 X154.534 Y128.996
;TYPE:WALL-OUTER
G1 F3600 E3229.76141
```
This sample comprises lines 112612 to 112623 of the attached gcode file

[pcb_vise_v2.gcode.zip](https://github.com/Ultimaker/Cura/files/3454794/pcb_vise_v2.gcode.zip)
[pcb_vise_v2.3mf.zip](https://github.com/Ultimaker/Cura/files/3454795/pcb_vise_v2.3mf.zip)

smartavionics, 2019-08-01:
Hello @edgary777. Yes, looking at the coasting code, it does exactly what you suggest, i.e. the filament that would have been extruded if coasting wasn't enabled is, in fact, extruded following the travel move.

Personally, I never use coasting. Even if it didn't do that extra prime I still wouldn't use it.

tlhintoq, 2019-08-01:
Well... Many of us do use this feature.  Properly used it does wonders to improve print quality and reduce stringing; especially with PETG.  As @edgary777 points out this bug defeats the very point of its use.

smartavionics, 2019-08-01:
> Properly used it does wonders to improve print quality and reduce stringing; especially with PETG.

Hmm, interesting. I print a lot of PETG and don't get stringing even without coasting.

> As @edgary777 points out this bug defeats the very point of its use.

I agree, but that's how it has been for years.

Liger0, 2019-08-01:
Oh this explains why I had bubbles printing with cura using coasting.
+1 for this behaviour to change.

Block137, 2019-08-04:
Will fixing this cause under extrusion to other users?

Liger0, 2019-08-04:
Isn't it what extra restart is for?

Block137, 2019-08-04:
Then setting that to negative number should be a workaround.

Do you mean retraction extra prime amount? Coasting does not necessarily retract, if there are multiple walls, coasting occurs at the end of every walls. With this fix, it basically stops extruding for a moment before moving to next wall.

edgary777, 2019-08-04:
@Block137 This wouldn’t cause under extrusion for anyone, it is a bug because it is “priming” without doing a retraction.

For some reason, when coasting is activated, all the material that isn’t extruded at the end of each wall is then extruded at a single point in the very end of the wall, creating a big plastic blob.

pseudex, 2019-08-07:
Finally someone pointed out what causes these extra primes. Looking now for hours, why this happens.
I also created a small python script to remove those primes. I will go then without coasting.
Thank you very much.
If someone wants the (very basic) script: [filterPrime.py](https://gitea.kaiserflur.de/pseudex/CuraAddons/src/branch/master/filterPrime.py)

Ghostkeeper, 2019-08-07:
Yeah, I don't see a reason for that priming to be there. It's being done explicitly here though:

https://github.com/Ultimaker/CuraEngine/blob/526925a7d8cf51c491566e93e789aec1665bed65/src/LayerPlan.cpp#L1847

This was introduced when the implementation of Coasting was finished, here: https://github.com/Ultimaker/CuraEngine/commit/565757793833d5412e229a5ddc2472a2a5b58db0 But it doesn't say why. Maybe the idea is that this would prevent underextrusion due to having underpressure in your nozzle chamber, but indeed that kind of defeats the purpose of coasting.

edgary777, 2019-08-07:
@Ghostkeeper Could we at least get an option to disable coasting priming?

If it was explicitly done, and there’s the possibility it might be useful for someone, giving us the option to turn it off would be a good compromise.

Ghostkeeper, 2019-08-08:
I'm going to propose removing the prime entirely. It defeats the purpose of coasting. But we'll do some test prints of course if we do that.

edgary777, 2019-08-10:
https://github.com/Ultimaker/CuraEngine/issues/497

I just found this on the CuraEngine issues list, from 2 years ago.

I don't get it at all, does it means that this was done as a sort of test and then they just gave up on coasting afterwards?

Why keep the feature so long if they had already given up on it?

Ghostkeeper, 2019-08-11:
Yes, most new print features start as an experiment like that. That's why the setting is still in experimental. Still, some people seem to like coasting and use it for all of their prints, so we're not about to remove it. But it's not in the place that we'd like either. And so it stays in Experimental until we have any need to improve upon it, which is never. Or at least not any time soon.

Opportunistic improvements can still be made though. If we find that removing this one line of code makes the result that much more usable, that'd be fine.

BagelOrb's comment says that he meant it to prevent creating underextrusion in the next line after the coasting. While coasting, the nozzle oozes the material that was in the nozzle chamber until the chamber is emptied. When printing the next line, it takes a while for the material to start flowing again because the chamber needs to fill up with material again. Now this happens very fast and the nozzle chamber is probably not entirely emptied by far, but the effect is still there. I agree though that we could apply an extra prime after coasting that is a little bit less than all of the volume that was missed. Or maybe it just works to leave out the prime altogether but reduce the coasting volume drastically.

closed 2019-08-19 by LipuFei

Ghostkeeper, 2019-08-19:
We've removed this prime now. When printing with Ultimaker's feeders behind the Bowden tube there is very little change due to the prime. The prime was always very short anyway.