Ultimaker/Cura #6109: [4.2.0] Selected profiles do not remain selected after restarting

FITPrinting, 2019-08-01:
**Application version**
4.2.0

**Platform**
Win 10, 64 bit

**Printer**
CR10S5

**Reproduction steps**
1. Open Cura.
2. Select a custom profile.
3. Slice, print, do whatever.
4. Close Cura.
5. Reopen Cura, find profile selected in Step 2, no longer selected, has returned to a default profile.

**Actual results**
See Step 5.

**Expected results**
Selected profile should remain

Vandrasc, 2019-08-01:
Hi @FITPrinting , I am not able to reproduce, could you share the project file with us, containing the custom profile? Thank you!

FITPrinting, 2019-08-01:
Attached is the 0.2 layer height custom profile, that is completely wrong after the forced adoption of the cre_awful_ mod and a .mht Microsoft Problem Step Recorder file that shows the screenshots of each step from opening Cura the first time, selecting the desired profile, closing Cura, reopening it, and finding the profile no longer is selected. 
[Cura 4.2.0 Selected Profile.zip](https://github.com/Ultimaker/Cura/files/3457367/Cura.4.2.0.Selected.Profile.zip)

FITPrinting, 2019-08-01:
Confirmed that the behavior continues in 4.2.1

nallath, 2019-08-02:
Could you actually share a project file? My guess is that it's not the profile that you created that is causing the issue.

FITPrinting, 2019-08-02:
There is not project file as I haven't attempted to print anything since it doesn't work by default. As can be seen in the screenshots the behavior occurs without a single model on the platform.

Vandrasc, 2019-08-02:
Hi @FITPrinting , the project is not only storing the model and the customized settings but also your custom profile, material, everything you have currently in use. 

I did try with the attached profile but after I restart the application, the profile is still the active one. I did the following:

1. Open 4.2.1 and add CR-10S5 
2. Go to Profile manager and import the attached profile
3. Load a model and slice
4. Restart Cura --> the imported profile is still selected. 

Did I missed or misunderstood any steps? Do you have any other cura user folder then the 4.2.1?

FITPrinting, 2019-08-02:
Closing issue, I've downgraded back to 4.0 to get my correct settings back and am exporting them to move to PrusaSlicer for non-destructive updates.

closed 2019-08-02 by FITPrinting