Ultimaker/Cura #6111: [Feature Request] Cura 4.2 - Printer Profile Import

Deneteus, 2019-08-01:
**Is your feature request related to a problem?**

I was testing the Linux version of Cura 4.2 and realized that there is no Printer Profile Import button/feature and the Prusa MK3/MK3S does not have profiles within the printer list.

**Describe the solution you'd like**

_I would like for there to be a way to import settings so I don't have to manually transfer settings like the start and end gcode._ 

**Describe alternatives you've considered**

 _I was going to manually see if I could import the profiles but I am using the Appimage version of Cura and the documentation fails to specify where anything lives._

**Affected users and/or printers**

_Anyone that has a printer where a profile does not exist yet in the current version. 
Examples: Hatchbox Alpha, Prusa MK3, Anycubic, Zyltech Hypercube 300_

nallath, 2019-08-01:
There is a printer import button. You can find it in preferences->profiles.

In the top of the screen there is a button called "import"

Deneteus, 2019-08-01:
On Cura 4.2 via Linux under Profiles is where you go to import the print settings for infill speeds and layer height. This is not the same as the printer settings for build plate,  printhead, start and end gcode, etc.

Ghostkeeper, 2019-08-07:
I'm not able to reproduce this either. I don't have the printer definitions that you mention, except Anycubic which does have custom profiles. But if I add a printer that doesn't have profiles there are always the fallback profiles: Extra Fine, Fine, etc. through to Extra Coarse Quality. And there is also always an import button:

[image]

ksihota, 2019-08-10:
If I understand what the OP is saying, they are actually talking about Printers and not Profiles.

Currently you can create your own Printer setup and have it display in Cura's Printer list by creating a json file for the printer and extruder and place them in the Cura folder in the Users' directory. They go under the folders 'definitions' and 'extruders'. The extruders file is referenced in the specified printer definition file. The actual locations for Win, Linux and MacOS are different. For Win its under /AppData/Roaming/
These will allow you to create new printers that include your start and end Gcode as well as other machine settings you specify in the json file.
Information relating to creating these files can be found here:
https://github.com/Ultimaker/Cura/wiki/Definition-Files-Explained

I would assume what the OP wants is a way to create a Custom FFF Printer in Cura, name it and then be able to export that information to a file that would essentially be the two files that go into the definitions and extruders folders.

I think these are examples of what the OP is referring to:

> Prusa MK3/MK3S does not have profiles

[image]

> no Printer Profile Import button/feature

_(can't import machine info into existing machine)_
[image]

Vandrasc, 2019-08-14:
Hi @Deneteus , after discussing with Cura team, it was decided not to implement this, sorry. 
There are 2 solutions though: you can either use a CFFF printer and customize from there or create your own machine definition. Hope this helps. 

Thank you for the feedback!

closed 2019-08-14 by Vandrasc

tr1p0p, 2020-01-09:
Oh my... It's SO CLOSE MINDED to decide NOT implement a way to import printers config. It's like so dumb and opposite to the values of 3D print community..

Because, yeah, NOBODY custom is own printer and have to create a FFF or XML files to modify and easily import/export around other software/computers.

Why update the soft just for LOSSING stuffs ?

Ghostkeeper, 2020-01-14:
We do have plans to be able to add printer definitions to the Marketplace.

We decided to not implement a UI element to import definitions because we believe that the workflow of creating your own printer definition is complex enough to be in the realm of the programmers rather than the users. If we'd add it, people are going to use it and complain it's too hard. So while we'd like to promote sharing print definitions, we'll section it off better.

You can of course still import print profiles for Custom FFF printer. Works in the same way.