Ultimaker/Cura #6113: Publishing a release candidate before the proper update?

JohnEdwa, 2019-08-01:
Going back, I see 4.2.1, 3.5.1, 3.4.1, 3.3.1, 3.2.1, and so on - basically, almost every other release has had bugs that needed to be patched after the update went out that couldn't wait for the next release. 

So why not do a RC, give people (and yourselves) a few days to properly test it (github-only release and maybe a post on Reddit?), and in the case of 4.2 most of the issues that required the emergency 4.2.1 patch could have been caught and the release would have gone over smoother. If it works, proceed with the release. If not, fix, do another RC and repeat.

It would also allow us that watch this github with the "Releases only" setting to notice the RC and be able to give feedback on it.

nallath, 2019-08-02:
We're switching over to a new build system, which should make it pretty damn easy to also release nightly builds. 

Before this, we've had very little interest in builds like this, so I'm hoping that something good can come from all the noise that the creawesome situation created. 

But the success (and failure) of this setup is greatly influenced by the amount of people in the community willing to test it. Even if we have just a handful of people that regularly test these versions would already be enough.

DivingDuck, 2019-08-02:
Guess, the success will be influenced from how you use this instrument. If the Cura development team is able to release small fixes for glitches found during normal use cases within short time in nightly builds, I guess there will be a lot of users that will use and test these builds.

A main problem with Cura is, that there are to much deep changes in combination with, let me say, usual maintenance fixes. In addition the involved users need to wait too long for an update of those small fixes. 
Users like to see fixing small issues within a short time frame and therefore is a automated nightly build a very good instrument. For you too, as you are the able to push official updates with more confidence in a shorter time frame.

Maybe you know about calibre from Kovid Goyal. His project is an good example how interaction between software development and user integration can work really successful.

JohnEdwa, 2019-08-02:
The master build provided by @smartavionics is kind of a nightly build, with the benefit of having his excellent tweaks. 
It's actually what I use myself and in [which I found the first occurrence of the z-hop bug,](https://github.com/smartavionics/Cura/issues/14) but because it wasn't in the 0724 build but was in the 0727/0728 one, and 4.2 was released two days after that, the fix didn't make it into the official 4.2 release.

nallath, 2019-08-02:
Fair enough, but the fact that there are changes in there makes it a bit harder to use it as a "official" nightly of course.

smartavionics, 2019-08-02:
And I wouldn't want it to be considered a "nightly" build and certainly not "official".

Having said that, if there's an upstream bug (or one of my own devising) that can be fixed and a build released then I would normally do that without too much delay.

Ghostkeeper, 2019-08-07:
I don't think that a release candidate will really solve this issue. After all, we did have a beta and the issues were still not caught there. I think that the users that take the effort of testing the beta are not diverse enough to catch all issues.

Having nightly builds will improve things a bit though. It opens up a possibility for people to test whether a fix that we make works for them as well.

JohnEdwa, 2019-08-07:
For this specific bug it would have because it was introduced in your commits after the beta was released. 4.2 Beta was released 2019-07-16, but the 2019-07-24 master build was fine and the 2019-07-27 one had the bug - 4.2 proper was released 2019-07-30.

nallath, 2019-08-08:
This specific situation would have been caught by it yeah. Having an actual RC won't magically fix all the things that can go wrong, but it's another step in ensuring that the chance that something does go wrong is smaller.

DivingDuck, 2019-08-13:
... and it force development having clean (automated) processes to the whole complex chain. This is the real benefit of being able to establish a working nightly build process (even if you decide to use it e.g. only weekly)
I have recognize your company is ISO 9001 certificated. This also is an good point to keep the idea of ISO certification to a next level (but I guess you are already on the way in this area too).