Ultimaker/Cura #6114: Deprecation in UM modules.

Waples, 2019-08-01:
The code fires deprecation warnings.

```/usr/lib/python3.7/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.7/site-packages/UM/Settings/SettingDefinition.py:46: DeprecationWarning: invalid escape sequence \.
  regex_pattern = '(?<!\.|\w|\d)0+(\d+)'
/usr/lib/python3.7/site-packages/cura/Settings/CuraContainerRegistry.py:76: DeprecationWarning: invalid escape sequence \s
```

rburema, 2019-08-08:
Hi @Waples, Thanks for letting us know. We're currently using python 3.5 however, though we might want to be future compatibele.

Ghostkeeper, 2019-08-08:
I've made fixes for these two cases. I'm not seeing these myself though even though I also run Cura through Python 3.7 in my development environment.

closed 2019-08-12 by Ghostkeeper