Ultimaker/Cura #6115: [4.2.1] Unable to remove custom material from list

jake-meister, 2019-08-01:
I was able to create a new material on the Materials list, but I cannot seem to delete it now. The "Remove" button remins inactive and the new material cannot be selected on the list.

fieldOfView, 2019-08-01:
That is probably because it is the currently selected material for a printer in Cura. Please select/activate a different material, and the Remove button should become active.

jake-meister, 2019-08-01:
Fixed it. The culprit is the "Material Settings" Plugin - it seems to "freeze" the Material Preference Pane.

fieldOfView, 2019-08-01:
I am the creator of the Material Settings plugin. Could you elaborate what you mean by "freeze"?

jake-meister, 2019-08-01:
Sure. If you install the Material Settings Plug-in in cura 4.2.1 the materials cannot be selected or edited from within the "manage Materials" menu. I will upload a video to youtube (private access) to illustrate it better... I am running Cura 4.2.1 on a mac (Mojave 10.14.6)

jake-meister, 2019-08-01:
https://youtu.be/6G4alIyKDHI

Liger0, 2019-08-02:
#6137 is explained then.

Ghostkeeper, 2019-08-08:
Closing this as duplicate of https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues/3 . It looks like fieldOfView has made a fix for it in his plug-in :tada:

closed 2019-08-08 by Ghostkeeper