Ultimaker/Cura #6117: Second extruder does not clear first extruder layer when printing

pswiss, 2019-08-01:
**Application version**

Cura 4.2

**Platform**

Windows 10

**Printer**

Ultimaker 3

**Reproduction steps**

Attempted to print using default configuration

**Actual results**

Extruder 1 (PLA) has been extruded into three cylinders, with Extruder 2 (PVA) serving as dissolvable support. PLA has been extruded approximately 2mm higher than pva. When laying down PVA, printer does not lower build plate, but instead simply barges through existing PLA
Video available here:
https://drive.google.com/drive/u/1/folders/1nwNDWvrgStYO30qhOhZ2YESxCCtEOacP 

**Expected results**

I would expect the second extruder to avoid barrelling through the existing print

**Additional information**

G-code, picture, and video available here:
https://drive.google.com/drive/u/1/folders/1nwNDWvrgStYO30qhOhZ2YESxCCtEOacP

rburema, 2019-08-08:
Could you attach the original model this was sliced from? If you zip it, github will accept embedding the file here.

Ghostkeeper, 2019-08-08:
I wonder how you made it print the PLA 2mm higher than the PVA. It should always print from bottom to top.

pswiss, 2019-08-08:
@rburema Model uploaded. I had arranged three of this model on the buildplate in the manner shown in the picture
[WormGear_2.zip](https://github.com/Ultimaker/Cura/files/3482385/WormGear_2.zip)

@Ghostkeeper No idea how it happened. I didn't see anything wrong during the active leveling.

Ghostkeeper, 2019-08-13:
So far it's printing fine.

[image]

Ghostkeeper, 2019-08-13:
[image]

I must say though that due to overextrusion the PLA was slightly higher than the PVA during the second half of the print. The PVA was underextruding and the PLA was overextruding.

pswiss, 2019-08-13:
I guess I'll chalk this up to a one-off issue with offset calibration then

closed 2019-08-14 by Ghostkeeper